brektrou/rtl8821CU #5: 802.11ac in Access Point Mode?

hangim, 2019-05-26:
Hi, I have installed this driver and want to run as `Access Point Mode`, in *5Ghz*.

```
sudo ./create_ap -w 2 --no-virt -c 36 --ieee80211n --ieee80211ac \
  --ht_capab '[HT40+]' wlan0 eth0 AP_name password
```

Found the max bandwidth is only *150Mbps*, *802.11n*.

I have tried a lot of time, but still can't run it at bandwidth *433Mbps*, *802.11ac*.

I have tested, use it connect to a Router, everything is fine, could achieve *433Mbps*.

I am not sure the problem caused by my command, or the driver, or the device limit.

So, I want to get some help with this.

Want to know it could be running **802.11ac** in `Access Point Mode` and achieve *433Mbps*?

brektrou, 2019-05-28:
What if you change `CONFIG_MP_VHT_HW_TX_MODE = n` to `CONFIG_MP_VHT_HW_TX_MODE = y` in `Makefile` and recompile the driver?

I don't have any other 802.11ac device, so I never tested it.

hangim, 2019-05-28:
> What if you change `CONFIG_MP_VHT_HW_TX_MODE = n` to `CONFIG_MP_VHT_HW_TX_MODE = y` in `Makefile` and recompile the driver?
> 
> I don't have any other 802.11ac device, so I never tested it.

Tested, seems the signal is better, but still *150Mbps*, nothing else change

hangim, 2019-05-29:
In windows with the official driver, could achieve *200 Mbps*, seems 1x1 40MHz, 802.11 ac, I can't find any configure to improve it.
Looking forward to some idea about 802.11ac， 1x1 80 Mhz, *443 Mbps* on Linux.

softwarecreations, 2019-06-05:
@hangim I don't know if this will help, but I'm also interested in running 5ghz AP's.
I'm considering buying a RTL8821CU (because RTL8812AU isn't available in my area at the moment)
But I don't want to buy it without knowing that it'll work in AP mode at 5ghz.

@hangim are you saying you got it working at 5ghz at 150mbps? What channel? Please share your hostapd config.

I've got the RTL8812AU and have successfully setup an AP where one of my phones can connect at 400mbps.

Here is my hostapd config for RTL8812AU

```
#working with gordboy 5.2.20.2 driver
#maybe also armbian nanoPi Neo2 4.19.20 driver

interface=tenda-rtl8812au
ssid=test

driver=nl80211

auth_algs=1
wpa=2
wpa_passphrase=testtest
wpa_key_mgmt=WPA-PSK
rsn_pairwise=CCMP

bridge=br0
#beacon_int>550 doesn't show up on iPhone6
beacon_int=500
#SSID not hidden
ignore_broadcast_ssid=0

hw_mode=a
#ch 44 is the highest channel that works with RTL8812AU
channel=44
max_num_sta=128

### IEEE 802.11n
ieee80211n=1
ht_capab=[HT20][HT40+][SHORT-GI-20][SHORT-GI-40][SHORT-GI-80][DSSS_CCK-40]

### IEEE 802.11ac
ieee80211ac=1
#vht_oper_chwidth=1
#vht_capab=[HT40-]
#vht_oper_centr_freq_seg0_idx=46

### WMM
wmm_enabled=1
uapsd_advertisement_enabled=1
wmm_ac_bk_cwmin=4
wmm_ac_bk_cwmax=10
wmm_ac_bk_aifs=7
wmm_ac_bk_txop_limit=0
wmm_ac_bk_acm=0
wmm_ac_be_aifs=3
wmm_ac_be_cwmin=4
wmm_ac_be_cwmax=10
wmm_ac_be_txop_limit=0
wmm_ac_be_acm=0
wmm_ac_vi_aifs=2
wmm_ac_vi_cwmin=3
wmm_ac_vi_cwmax=4
wmm_ac_vi_txop_limit=94
wmm_ac_vi_acm=0
wmm_ac_vo_aifs=2
wmm_ac_vo_cwmin=2
wmm_ac_vo_cwmax=3
wmm_ac_vo_txop_limit=47
wmm_ac_vo_acm=0

### TX queue parameters
tx_queue_data3_aifs=7
tx_queue_data3_cwmin=15
tx_queue_data3_cwmax=1023
tx_queue_data3_burst=0
tx_queue_data2_aifs=3
tx_queue_data2_cwmin=15
tx_queue_data2_cwmax=63
tx_queue_data2_burst=0
tx_queue_data1_aifs=1
tx_queue_data1_cwmin=7
tx_queue_data1_cwmax=15
tx_queue_data1_burst=3.0
tx_queue_data0_aifs=1
tx_queue_data0_cwmin=3
tx_queue_data0_cwmax=7
tx_queue_data0_burst=1.5

### DFS - works but not needed
#country_code=DE
#ieee80211d=1
#ieee80211h=1

```

hangim, 2019-06-05:
@softwarecreations 
I use this shell script to create AP, which could generate hostapd config automatically.
https://github.com/oblique/create_ap

The command I used shared in the first comment.

Channel=36 or 149, nothing difference. 

The `hostapd.conf` should look like
```
beacon_int=100
ssid=ap_name
interface=wlan0
driver=nl80211
channel=36
ctrl_interface=/tmp/create_ap.xxxxxx/hostapd_ctrl
ctrl_interface_group=0
ignore_broadcast_ssid=0
ap_isolate=0
hw_mode=a
ieee80211n=1
ht_capab=[HT40+]
ieee80211ac=1
wmm_enabled=1
wpa=2
wpa_passphrase=password
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP CCMP
rsn_pairwise=CCMP
```

softwarecreations, 2019-06-05:
@hangim thanks for info.
Try my config and see if you get better speed.
Also try different mobile phones.
My Samsung Galaxy S6 Edge+ connected at 400mbps whereas the other phones (like S3 and S5, etc) only 150mbps, same AP, same settings.

hangim, 2019-06-05:
@softwarecreations 
I have two devices, both tested.

In my opinion, the big difference between my config and yours is `ht_capab`.
I have tested your `ht_capab` parameter, nothing different, still *150mbps*.
In fact, I have tested a lot of `ht_capab` combine, but none of them could make me satisfied.

I don't think `WMM` and `TX queue` will affect the result, maybe these will resulted in some performance improved, but I think it's very little and imperceptible. And I don't see these configure in most of configure example.

B1oHazard, 2019-12-21:
To reach 433mbps and 80MHz channel width u need load driver with rtw_vht_enable=2 or find rtw_vht_enable in source code and change from 1 to 2 and recompile driver.
Or u can add 8821cu to /etc/modules and create /etc/modprobe.d/8821cu.conf with content
`options 8821cu rtw_vht_enable=2`
reboot and test hostapd

hostapd.conf:
```
interface=wlan0
driver=nl80211
ssid=YOURSSID

hw_mode=a
ieee80211n=1
ieee80211h=1
ieee80211d=1
country_code=US

channel=36
vht_oper_centr_freq_seg0_idx=42
vht_oper_chwidth=2
ht_capab=[HT40+]

wpa=2
wpa_passphrase=12345678
wpa_key_mgmt=WPA-PSK
rsn_pairwise=CCMP
```

tested with 8811CU cheap dongle, also will work with rtl8812au driver from aircrack-ng repo

hangim, 2019-12-22:
@B1oHazard Thank you very much, I have tested your solution, create `/etc/modprobe.d/8821cu.conf` with content `options 8821cu rtw_vht_enable=2`.
Now, everything works very well, and *443mbps* has reached.

tony1016, 2021-02-09:
> To reach 433mbps and 80MHz channel width u need load driver with rtw_vht_enable=2 or find rtw_vht_enable in source code and change from 1 to 2 and recompile driver.
> Or u can add 8821cu to [...]

But the actual iperf speed only 8mbps,don't know why

bmilde, 2022-01-03:
The jetson nano 2G arm dev board has a USB dongle for wifi. It works with hostapd, but no matter what I had set in hostapd.conf it was stuck at 150mbit (5Ghz 40mhz).

The driver on the jetson nano was called rtl8821cu though. Can confirm that creating a new file /etc/modprobe.d/rtl8821cu.conf with the contents:

    options rtl8821cu rtw_vht_enable=2

finally got me to a reported "Bit Rate:434 Mb/s" in sudo  iwconfig wlan0 and hostapd to use 80 Mhz 5Ghz channels. Iperf3 also gets me a much faster 260mbits now. Woohoo! 

Thanks u/tony1016 for this suggestion!