brektrou/rtl8821CU #9: debian buster support rtl8821CU in firmware

zhenghub, 2019-06-15:
I used this to fix bluetooth problems in ubuntu19. Thanks.
And now Debian buster is already able to support 8821Cu in firmware. Just to execute:
> apt install firmware-realtek
Just to say thanks, and to notice people.

meijkl, 2019-07-28:
Works for me with Debian Buster & Liquorix 5.2 kernel!

Thanks!
Klaus

voltagex, 2019-09-07:
Doesn't work on Raspbian Buster, as the module appears to be missing  (or the kernel is too old)

sabirovrinat85, 2019-09-10:
fresh installed Debian Buster 10.1 , same thing, doesn't work on x86 pc, linux kernel 4.19, cheap dongle from Comfast is recognized as 0bda:c811 Realtek Semiconductor Corp., "make" fails with status 2, in make.log there's something about rtw_debug.c:45:62 error macro DATE might prevent reproducible builds

sabirovrinat85, 2019-09-10:
adding  additional string

EXTRA_CFLAGS += -Wno-error=date-time

into Makefile made compilation process done, so problem is solved (don't forget installing wpasupplicant and wireless-tools if using Debian's netinstall version)

gfiasco, 2019-10-21:
@voltagex did you manage it on raspberry pi?

dyang-7, 2019-12-25:
> @voltagex have you made it work on raspberry pi?

https://github.com/brektrou/rtl8821CU#arm-architecture-tweak-for-this-driver-this-solves-compilation-problem-of-this-driver

```
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 10 (buster)
Release:	10
Codename:	buster
```

```
phy#1
	Interface wlan0
		ifindex 3
		wdev 0x100000001
		addr 20:0b:d0:36:f2:1d
		type managed
		txpower 12.00 dBm
```