brektrou/rtl8821CU #10: Build fail on ARM (Raspberry Pi 2) - Fedora 30

rrafal1337, 2019-06-23:
Hi. Code compiles and works well for x86_64 on Fedora 29/30 but it fails on ARM (Raspberry Pi 2):
```
[Sun Jun 23 17:08:36 2019] usb 1-1.4: new high-speed USB device number 4 using dwc2
[Sun Jun 23 17:08:36 2019] usb 1-1.4: New USB device found, idVendor=0bda, idProduct=c811, bcdDevice= 2.00
[Sun Jun 23 17:08:36 2019] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Sun Jun 23 17:08:36 2019] usb 1-1.4: Product: 802.11ac NIC
[Sun Jun 23 17:08:36 2019] usb 1-1.4: Manufacturer: Realtek
[Sun Jun 23 17:08:36 2019] usb 1-1.4: SerialNumber: 123456
[Sun Jun 23 17:08:36 2019] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[Sun Jun 23 17:08:36 2019] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[Sun Jun 23 17:08:37 2019] 8821cu: loading out-of-tree module taints kernel.
[Sun Jun 23 17:08:37 2019] 8821cu: module verification failed: signature and/or required key missing - tainting kernel
**[Sun Jun 23 17:08:37 2019] 8821cu: Unknown symbol __aeabi_ui2d (err -2)
[Sun Jun 23 17:08:37 2019] 8821cu: Unknown symbol __aeabi_dsub (err -2)
[Sun Jun 23 17:08:37 2019] 8821cu: Unknown symbol __aeabi_fcmple (err -2)
[Sun Jun 23 17:08:37 2019] 8821cu: Unknown symbol __aeabi_ddiv (err -2)
[Sun Jun 23 17:08:37 2019] 8821cu: Unknown symbol __aeabi_fsub (err -2)
[Sun Jun 23 17:08:37 2019] 8821cu: Unknown symbol __aeabi_d2uiz (err -2)
[Sun Jun 23 17:08:37 2019] 8821cu: Unknown symbol __aeabi_d2iz (err -2)
[Sun Jun 23 17:08:37 2019] 8821cu: Unknown symbol __aeabi_d2f (err -2)
[Sun Jun 23 17:08:37 2019] 8821cu: Unknown symbol __aeabi_dcmpge (err -2)
[Sun Jun 23 17:08:37 2019] 8821cu: Unknown symbol __aeabi_dmul (err -2)
[Sun Jun 23 17:08:37 2019] 8821cu: Unknown symbol __aeabi_dcmple (err -2)
[Sun Jun 23 17:08:37 2019] 8821cu: Unknown symbol __aeabi_i2d (err -2)
[Sun Jun 23 17:08:37 2019] 8821cu: Unknown symbol __aeabi_i2f (err -2)
[Sun Jun 23 17:08:37 2019] 8821cu: Unknown symbol __aeabi_f2iz (err -2)
[Sun Jun 23 17:08:37 2019] 8821cu: Unknown symbol __aeabi_dcmplt (err -2)
[Sun Jun 23 17:08:37 2019] 8821cu: Unknown symbol __aeabi_dcmpgt (err -2)
[Sun Jun 23 17:08:37 2019] 8821cu: Unknown symbol __aeabi_dadd (err -2)**

```
Similar driver for different chipset compiles and works ok on ARM: https://github.com/cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959

Easy to reproduce on any latest Fedora. 
sudo arm-image-installer --image=Fedora-Server-armhfp-30-1.2-sda.raw.xz --media=/dev/sdb --target=rpi2

rrafal1337, 2019-06-23:
Ok found solution for that. My previous tactics was bad because tried to get rid of hard-float from Makefile and that caused this error. New approach fixed issue. Now it works on Raspberry Pi:
Please add these additional steps to Readme for ARM architecture:

```
sudo cp /lib/modules/$(uname -r)/build/arch/arm/Makefile /lib/modules/$(uname -r)/build/arch/arm/Makefile.$(date +%Y%m%d%H%M)
sudo sed -i 's/-msoft-float//' /lib/modules/$(uname -r)/build/arch/arm/Makefile
sudo ln -s /lib/modules/$(uname -r)/build/arch/arm /lib/modules/$(uname -r)/build/arch/armv7l
```

AngelsWillRule, 2019-07-03:
> Ok found solution for that. My previous tactics was bad because tried to get rid of hard-float from Makefile and that caused this error. New approach fixed issue. Now it works on Raspberry Pi: [...]

Thank you. It's working perfectly on Raspberry Pi 3+. Don't forget to install bc first: 

`sudo apt-get install bc `

Cheers!

closed 2019-08-27 by rrafal1337