brektrou/rtl8821CU #13: Build error on Armbian 5.93 for Nanopi Neo

edmont, 2019-08-22:
```
root@localhost:~/build/rtl8821CU# uname -a
Linux localhost 4.19.64-sunxi #5.93 SMP Mon Aug 5 14:17:36 CEST 2019 armv7l GNU/Linux
root@localhost:~/build/rtl8821CU# make
make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.19.64-sunxi/build M=/root/build/rtl8821CU  modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.64-sunxi'
  CC [M]  /root/build/rtl8821CU/core/rtw_cmd.o
gcc: error: -mfloat-abi=soft and -mfloat-abi=hard may not be used together
make[2]: *** [scripts/Makefile.build:304: /root/build/rtl8821CU/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1518: _module_/root/build/rtl8821CU] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.19.64-sunxi'
make: *** [Makefile:2182: modules] Error 2
```
Maybe some tweaking needs to be done in the Makefile?

rrafal1337, 2019-08-22:
Hi. Already provided solution for that https://github.com/brektrou/rtl8821CU/issues/10
I can confirm that solves problem.

edmont, 2019-08-22:
Thanks @rrafal1337! Yes, now it's compiling!