brektrou/rtl8821CU #15: Can not work on kernel 5.3-rc5

yutent, 2019-08-23:
```
OS: Linux Mint 19.2 Tina x86_64 
```
It works on kernel 5.2.9, but not 5.3-rc5

smp79, 2019-09-11:
Same here with current 5.3-rc8. OS: LibreELEC

licardoY, 2019-09-16:
works on 5.1.10, but not 5.3rc8.....

breisig, 2019-09-21:
It doesn't work on 5.3.0 final either.

dscottboggs, 2019-09-22:
Is there any more information besides "it doesn't work"?

breisig, 2019-09-23:
Here's some debug message on 5.3.1

[    9.578352] usb 1-1: New USB device found, idVendor=0bda, idProduct=c811, bcdDevice= 2.00
[    9.578353] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    9.578354] usb 1-1: Product: 802.11ac NIC
[    9.578355] usb 1-1: Manufacturer: Realtek
[    9.578355] usb 1-1: SerialNumber: 123456
[    9.596880] 8821cu: no symbol version for module_layout
[    9.596893] 8821cu: loading out-of-tree module taints kernel.
[    9.656145] 8821cu: module verification failed: signature and/or required key missing - tainting kernel
[    9.666426] RTW: module init start
[    9.666428] RTW: rtl8821cu v5.4.1_28754.20180921_COEX20180712-3232
[    9.666428] RTW: build time: Sep 21 2019 14:33:47
[    9.666428] RTW: rtl8821cu BT-Coex version = COEX20180712-3232
[    9.666463] RTW: [HALMAC]11692M
               HALMAC_MAJOR_VER = 1
               HALMAC_PROTOTYPE_VER = 4
               HALMAC_MINOR_VER = 19
               HALMAC_PATCH_VER = 3
[    9.670527] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[   10.402471] RTW: HW EFUSE
[   10.402472] RTW: 0x000: 29 81 00 BC  09 00 31 00  6F 04 A4 10  12 00 30 0B
[   10.402475] RTW: 0x010: 23 23 23 24  24 24 2B 2B  2B 2C 2C 02  FF FF FF FF
[   10.402477] RTW: 0x020: FF FF 2B 2A  28 28 27 27  28 28 29 29  27 29 29 29
[   10.402479] RTW: 0x030: 24 FF FF FF  FF FF D0 FF  FF FF 23 23  23 24 24 24
[   10.402481] RTW: 0x040: 2B 2B 2B 2C  2C 02 FF FF  FF FF FF FF  FF FF FF FF
[   10.402483] RTW: 0x050: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   10.402485] RTW: 0x060: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   10.402487] RTW: 0x070: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   10.402489] RTW: 0x080: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   10.402491] RTW: 0x090: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   10.402493] RTW: 0x0A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   10.402495] RTW: 0x0B0: FF FF FF FF  FF FF FF FF  7F 06 1A 00  FF FF FF FF
[   10.402496] RTW: 0x0C0: FF 22 00 41  00 00 00 00  00 FF 22 FF  FF FF FF FF
[   10.402498] RTW: 0x0D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   10.402500] RTW: 0x0E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   10.402502] RTW: 0x0F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   10.402504] RTW: 0x100: DA 0B 11 C8  C3 46 02 40  A5 EF 4E 92  81 09 03 52
[   10.402506] RTW: 0x110: 65 61 6C 74  65 6B 0E 03  38 30 32 2E  31 31 61 63
[   10.402508] RTW: 0x120: 20 4E 49 43  08 03 31 32  33 34 35 36  FF FF FF FF
[   10.402510] RTW: 0x130: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   10.402512] RTW: 0x140: FF FF FF FF  00 00 31 0F  FF FF FF FF  FF FF FF FF
[   10.402514] RTW: 0x150: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   10.402516] RTW: 0x160: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   10.402518] RTW: 0x170: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   10.402520] RTW: 0x180: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   10.402522] RTW: 0x190: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   10.402524] RTW: 0x1A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   10.402526] RTW: 0x1B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   10.402527] RTW: 0x1C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   10.402529] RTW: 0x1D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   10.402531] RTW: 0x1E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   10.402533] RTW: 0x1F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   10.403149] RTW: hal_com_config_channel_plan chplan:0x7F
[   10.435401] RTW: [HALMAC][ALWAYS]shall R reg twice!!
[   10.444928] RTW: WARN [HALMAC][WARN]H2C/C2H ver is compatible!!
[   10.742752] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[   10.743356] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[   10.743990] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[   10.744600] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[   10.745220] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[   10.745838] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[   10.746491] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[   10.747266] RTW: rtw_regsty_chk_target_tx_power_valid return _FALSE for band:0, path:0, rs:0, t:-1
[   10.753103] ------------[ cut here ]------------
[   10.753123] WARNING: CPU: 7 PID: 5732 at net/wireless/core.c:868 wiphy_register.cold+0x37/0x435
[   10.753123] Modules linked in: pcc_cpufreq(-) acpi_cpufreq(-) xt_multiport ip6t_rpfilter ip6t_REJECT nf_reject_ipv6 ipt_REJECT nf_reject_ipv4 xt_conntrack ebtable_nat ip6table_nat ip6table_mangle ip6table_security ip6table_raw iptable_nat nf_nat iptable_mangle iptable_security iptable_raw nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 8821cu(OE+) ebtable_filter ebtables intel_rapl_msr intel_rapl_common ip6table_filter ip6_tables iptable_filter bpfilter crct10dif_pclmul crc32_pclmul crc32c_intel snd_pcm_oss ghash_clmulni_intel snd_mixer_oss iTCO_wdt aesni_intel iTCO_vendor_support bochs_drm aes_x86_64 uhci_hcd snd_hda_intel crypto_simd drm_vram_helper snd_hda_codec cryptd ttm glue_helper snd_hda_core joydev drm_kms_helper snd_hwdep snd_pcm drm intel_agp snd_timer intel_gtt snd lpc_ich soundcore ehci_pci agpgart ehci_hcd serio_raw i2c_i801 evdev sch_fq_codel ip_tables
[   10.753139] CPU: 7 PID: 5732 Comm: modprobe Tainted: G           OE     5.3.1 #1
[... cut ...]
[   10.806410] RAX: ffffffffffffffda RBX: 00000000013f1ae0 RCX: 00007f8ebbaddb4d
[   10.806411] RDX: 0000000000000000 RSI: 000000000041b090 RDI: 0000000000000003
[   10.806411] RBP: 000000000041b090 R08: 0000000000000000 R09: 0000000000000000
[   10.806412] R10: 0000000000000003 R11: 0000000000000206 R12: 0000000000000000
[   10.806412] R13: 00000000013f2210 R14: 0000000000040000 R15: 00000000013f1ae0
[   10.806413] ---[ end trace ab43c952f683a7e4 ]---
[   10.809475] RTW: WARN free_recv_skb_queue not empty, 8

smp79, 2019-09-24:
[Fix for kernel 5.3](https://github.com/smp79/rtl8821CU/commit/2ec29cfa7fd05d2321aa022c8281b97c1d190665). Tested and working.

breisig, 2019-10-01:
Can we get this merged?

troyfolger, 2019-10-16:
+1 for merging this -- I tested on Fedora 30 with 5.3.5-200.fc30.x86_64 and it works ...

rrafal1337, 2019-10-22:
Same for me, tested on 5.3.6 - Fedora 30:
https://pastebin.com/2r07xtXT
[image]

I don't see new commit for branch master. Do fix exists in some other branch?

brektrou, 2019-10-22:
Fixed: 23c0a81b5795bd6d794c13fbeb76509cef6ad76e db47217e31b95b4a760a58d6c4790631d1da4896

closed 2019-11-01 by brektrou