brektrou/rtl8821CU #19: fix nanopc t4 unrecognized command line option ‘-mhard-float’

georgefang73, 2019-09-18:
platform: nanopc t4 with armbian 5.95
uname -a:
Linux nanopct4 4.4.190-rk3399 #5 SMP Sun Sep 1 23:12:43 CEST 2019 aarch64 GNU/Linux

get the driver to compile successfully by changing the following in Makefile

CONFIG_MP_VHT_HW_TX_MODE = y
to
CONFIG_MP_VHT_HW_TX_MODE = n

georgefang73, 2019-09-18:
the original fix method comes from https://github.com/aircrack-ng/rtl8812au/issues/173

brektrou, 2019-09-18:
Did you have tried this?
https://github.com/brektrou/rtl8821CU/issues/10#issuecomment-504784263

georgefang73, 2019-09-18:
I didn't try this because there are some different things in my board with linux-headers-rk3399. I left this commit just want to show another method to solve the problem during compiling the driver. I'm sorry if my expression make you feel bad