brektrou/rtl8821CU #21: Building on Rasberry PI Issue

bernid, 2019-09-24:
Hi, 
I'm trying to build this driver/module on RPI by DKMS. 
Unfortunately I receive an error:
```
DKMS make.log for rtl8821CU-5.4.1 for kernel 4.19.66-v7+ (armv7l)
Tue 24 Sep 11:07:34 BST 2019
/bin/sh: 1: bc: not found
make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.19.66-v7+/build M=/var/lib/dkms/rtl8821CU/5.4.1/build  modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.66-v7+'
/bin/sh: 1: bc: not found
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_cmd.o
In file included from /var/lib/dkms/rtl8821CU/5.4.1/build/include/drv_types.h:30,
                 from /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_cmd.c:17:
/var/lib/dkms/rtl8821CU/5.4.1/build/include/wifi.h:1032: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 #define IEEE80211_MAX_AMPDU_BUF 0x40
 
In file included from /var/lib/dkms/rtl8821CU/5.4.1/build/include/osdep_service_linux.h:83,
                 from /var/lib/dkms/rtl8821CU/5.4.1/build/include/osdep_service.h:50,
                 from /var/lib/dkms/rtl8821CU/5.4.1/build/include/drv_types.h:27,
                 from /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_cmd.c:17:
./include/linux/ieee80211.h:1442: note: this is the location of the previous definition
 #define IEEE80211_MAX_AMPDU_BUF  0x100
 
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_security.o
In file included from /var/lib/dkms/rtl8821CU/5.4.1/build/include/drv_types.h:30,
                 from /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_security.c:17:
/var/lib/dkms/rtl8821CU/5.4.1/build/include/wifi.h:1032: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 #define IEEE80211_MAX_AMPDU_BUF 0x40
 
In file included from /var/lib/dkms/rtl8821CU/5.4.1/build/include/osdep_service_linux.h:83,
                 from /var/lib/dkms/rtl8821CU/5.4.1/build/include/osdep_service.h:50,
                 from /var/lib/dkms/rtl8821CU/5.4.1/build/include/drv_types.h:27,
                 from /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_security.c:17:
./include/linux/ieee80211.h:1442: note: this is the location of the previous definition
 #define IEEE80211_MAX_AMPDU_BUF  0x100
 
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_debug.o
In file included from /var/lib/dkms/rtl8821CU/5.4.1/build/include/drv_types.h:30,
                 from /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_debug.c:17:
/var/lib/dkms/rtl8821CU/5.4.1/build/include/wifi.h:1032: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 #define IEEE80211_MAX_AMPDU_BUF 0x40
 
In file included from /var/lib/dkms/rtl8821CU/5.4.1/build/include/osdep_service_linux.h:83,
                 from /var/lib/dkms/rtl8821CU/5.4.1/build/include/osdep_service.h:50,
                 from /var/lib/dkms/rtl8821CU/5.4.1/build/include/drv_types.h:27,
                 from /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_debug.c:17:
./include/linux/ieee80211.h:1442: note: this is the location of the previous definition
 #define IEEE80211_MAX_AMPDU_BUF  0x100
 
/var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_debug.c: In function â€dump_drv_versionâ€™:
/var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_debug.c:45:62: error: macro "__DATE__" might prevent reproducible builds [-Werror=date-time]
  RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
                                                              ^
/var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_debug.c:45:62: error: macro "__TIME__" might prevent reproducible builds [-Werror=date-time]
/var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_debug.c:45:62: error: macro "__DATE__" might prevent reproducible builds [-Werror=date-time]
/var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_debug.c:45:62: error: macro "__TIME__" might prevent reproducible builds [-Werror=date-time]
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:304: /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_debug.o] Error 1
make[1]: *** [Makefile:1519: _module_/var/lib/dkms/rtl8821CU/5.4.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.19.66-v7+'
make: *** [Makefile:2198: modules] Error 2
```
Is there any way to fix this? Can I do it by myself e.g editing and changing  Makefile?

rrafal1337, 2019-09-24:
You need to install bc package with your favorite package manager:

_/bin/sh: 1: bc: not found_

bernid, 2019-09-24:
Thank you. 
I looked at the end of log file and not at the  beginning and I missed lack of 'bc' package. 
After installing 'bc' and applying 'ARM architecture tweak' (from the readme file) the build process end successfully. 
I also applied few lines adjustment to Makefile to enable build driver for RPI:
https://github.com/zebulon2/rtl8812au/commit/d80d907e84be6001edbc08f8f0d0b93be1538094
After that everything looks ok. 

I use the newest rasbian /debian buster Linux.
BTW: you can consider to add to Makefile similar adjustment for RPI.

closed 2019-09-24 by bernid

realcarbonneau, 2020-04-18:
Same thing on a standard Debian 10 Buster fresh install all up to date.  But the driver still compiles, installs and works correctly (so far)

```
uname -a
Linux email 4.19.0-8-amd64 #1 SMP Debian 4.19.98-1 (2020-01-26) x86_64 GNU/Linux
```

bc is installed:
```
apt install bc -y
...
bc is already the newest version (1.07.1-2+b1).
```