brektrou/rtl8821CU #23: Multiple RTW RFK-CHK errors.

cjd3089, 2019-10-25:
Hey I'm running a 8821cu based EDUP AC1661 Wi-Fi/Bluetooth combo on Manjaro with kernel 5.3.6-1. Driver build worked fine but I'm experiencing an unstable connection with multiple soft dropouts plus disconnects and failures to connect. I did some searching on the log errors but I still cant find a solution. Any help would be great.
KSystemlog:
25/10/19 11:21 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
25/10/19 11:21 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
25/10/19 11:21 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:21 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:21 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:21 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:21 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:21 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:21 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:22 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
25/10/19 11:22 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
25/10/19 11:22 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:22 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:22 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:22 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:22 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:22 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:22 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:22 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:22 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:23 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
25/10/19 11:23 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
25/10/19 11:23 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
25/10/19 11:23 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:23 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:23 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:23 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:23 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:23 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:23 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
25/10/19 11:23 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
25/10/19 11:23 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
25/10/19 11:23 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:23 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:23 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:23 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:24 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:24 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
25/10/19 11:24 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
25/10/19 11:24 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
25/10/19 11:24 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:24 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:24 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:24 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:24 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:24 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:25 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
25/10/19 11:25 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
25/10/19 11:25 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
25/10/19 11:25 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:25 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:25 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:25 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:25 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:26 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
25/10/19 11:26 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
25/10/19 11:26 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
25/10/19 11:26 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:26 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:26 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:26 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:26 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:26 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:27 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
25/10/19 11:27 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
25/10/19 11:27 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:27 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:27 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
[... 43 lines cut ...]
25/10/19 11:32 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
25/10/19 11:32 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
25/10/19 11:32 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:32 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:32 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:32 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:32 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:32 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:32 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:32 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:32 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
25/10/19 11:32 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
25/10/19 11:32 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:33 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:33 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
[... cut ...]
25/10/19 11:33 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:33 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
25/10/19 11:33 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
25/10/19 11:33 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:33 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch
25/10/19 11:33 AM	kernel	RTW: ERROR [RFK-CHK] RF-K not allowed due to ld_iface not stayin union ch

cjd3089, 2019-10-25:
And here are the initial connection errors....
25/10/19 12:10 PM	wpa_supplicant	wlp3s0f0u7i2: Associated with 00:00:00:00:00:00
25/10/19 12:10 PM	wpa_supplicant	wlp3s0f0u7i2: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
25/10/19 12:10 PM	NetworkManager	<warn>  [1571965821.9910] sup-iface[0x56327f70b200,wlp3s0f0u7i2]: connection disconnected (reason -3)
25/10/19 12:10 PM	NetworkManager	<info>  [1571965821.9911] device (wlp3s0f0u7i2): supplicant interface state: associating -> disconnected
25/10/19 12:10 PM	NetworkManager	<info>  [1571965821.9912] device (p2p-dev-wlp3s0f0u7i2): supplicant management interface state: associating -> disconnected
25/10/19 12:10 PM	NetworkManager	<info>  [1571965821.9915] device (wlp3s0f0u7i2): supplicant interface state: disconnected -> associated
25/10/19 12:10 PM	NetworkManager	<info>  [1571965821.9915] device (p2p-dev-wlp3s0f0u7i2): supplicant management interface state: disconnected -> associated
25/10/19 12:10 PM	wpa_supplicant	wlp3s0f0u7i2: Trying to associate with ec:be:dd:4b:57:76 (SSID='Optus_4B5773_5GHz' freq=5180 MHz)
25/10/19 12:10 PM	kernel	RTW: rtw_set_802_11_connect(wlp3s0f0u7i2)  fw_state=0x00000008
25/10/19 12:10 PM	NetworkManager	<info>  [1571965822.0055] device (wlp3s0f0u7i2): supplicant interface state: associated -> associating
25/10/19 12:10 PM	NetworkManager	<info>  [1571965822.0055] device (p2p-dev-wlp3s0f0u7i2): supplicant management interface state: associated -> associating
25/10/19 12:10 PM	kernel	RTW: start auth
25/10/19 12:10 PM	kernel	RTW: auth success, start assoc
25/10/19 12:10 PM	kernel	RTW: assoc success
25/10/19 12:10 PM	wpa_supplicant	wlp3s0f0u7i2: Associated with ec:be:dd:4b:57:76
25/10/19 12:10 PM	wpa_supplicant	wlp3s0f0u7i2: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
25/10/19 12:10 PM	wpa_supplicant	wlp3s0f0u7i2: CTRL-EVENT-REGDOM-CHANGE init=COUNTRY_IE type=COUNTRY alpha2=AU
25/10/19 12:10 PM	kernel	RTW: recv eapol packet
25/10/19 12:10 PM	NetworkManager	<info>  [1571965826.7141] device (wlp3s0f0u7i2): supplicant interface state: associating -> 4-way handshake
25/10/19 12:10 PM	NetworkManager	<info>  [1571965826.7141] device (p2p-dev-wlp3s0f0u7i2): supplicant management interface state: associating -> 4-way handshake
25/10/19 12:10 PM	kernel	RTW: send eapol packet
25/10/19 12:10 PM	kernel	RTW: recv eapol packet
25/10/19 12:10 PM	kernel	RTW: send eapol packet
25/10/19 12:10 PM	kernel	RTW: recv eapol packet
25/10/19 12:10 PM	kernel	RTW: send eapol packet
25/10/19 12:10 PM	kernel	RTW: ============ STA [ec:be:dd:4b:57:76]  ===================
25/10/19 12:10 PM	kernel	RTW: mac_id : 0
25/10/19 12:10 PM	kernel	RTW: wireless_mode : 0x44
25/10/19 12:10 PM	kernel	RTW: mimo_type : 0
25/10/19 12:10 PM	kernel	RTW: bw_mode : 80MHz, ra_bw_mode : 80MHz
25/10/19 12:10 PM	kernel	RTW: rate_id : 10
25/10/19 12:10 PM	kernel	RTW: rssi : 49 (%), rssi_level : 0
25/10/19 12:10 PM	kernel	RTW: is_support_sgi : Y, is_vht_enable : Y
25/10/19 12:10 PM	kernel	RTW: disable_ra : N, disable_pt : N
25/10/19 12:10 PM	kernel	RTW: is_noisy : N
25/10/19 12:10 PM	kernel	RTW: txrx_state : 0
25/10/19 12:10 PM	kernel	RTW: curr_tx_rate : CCK_1M (L)
25/10/19 12:10 PM	kernel	RTW: curr_tx_bw : 20MHz
25/10/19 12:10 PM	kernel	RTW: curr_retry_ratio : 0
25/10/19 12:10 PM	kernel	RTW: ra_mask : 0x00000000003ffff0

25/10/19 12:10 PM	kernel	RTW: recv eapol packet
25/10/19 12:10 PM	wpa_supplicant	wlp3s0f0u7i2: WPA: Key negotiation completed with ec:be:dd:4b:57:76 [PTK=CCMP GTK=CCMP]
25/10/19 12:10 PM	wpa_supplicant	wlp3s0f0u7i2: CTRL-EVENT-CONNECTED - Connection to ec:be:dd:4b:57:76 completed [id=0 id_str=]
25/10/19 12:10 PM	kernel	RTW: send eapol packet
25/10/19 12:10 PM	kernel	IPv6: ADDRCONF(NETDEV_CHANGE): wlp3s0f0u7i2: link becomes ready
25/10/19 12:10 PM	NetworkManager	<info>  [1571965830.5284] device (wlp3s0f0u7i2): supplicant interface state: 4-way handshake -> completed
25/10/19 12:10 PM	NetworkManager	<info>  [1571965830.5284] device (wlp3s0f0u7i2): Activation: (wifi) Stage 2 of 5 (Device Configure) successful. Connected to wireless network "Optus_4B5773_5GHz"
25/10/19 12:10 PM	NetworkManager	<info>  [1571965830.5285] device (p2p-dev-wlp3s0f0u7i2): supplicant management interface state: 4-way handshake -> completed
25/10/19 12:10 PM	NetworkManager	<info>  [1571965830.5289] device (wlp3s0f0u7i2): state change: config -> ip-config (reason 'none', sys-iface-state: 'managed')
25/10/19 12:10 PM	NetworkManager	<info>  [1571965830.5296] dhcp4 (wlp3s0f0u7i2): activation: beginning transaction (timeout in 45 seconds)
25/10/19 12:10 PM	avahi-daemon	Joining mDNS multicast group on interface wlp3s0f0u7i2.IPv6 with address fe80::5b5d:52d9:bf0b:7816.
25/10/19 12:10 PM	avahi-daemon	New relevant interface wlp3s0f0u7i2.IPv6 for mDNS.
25/10/19 12:10 PM	avahi-daemon	Registering new address record for fe80::5b5d:52d9:bf0b:7816 on wlp3s0f0u7i2.*.
25/10/19 12:10 PM	kdeconnectd	kdeconnect.core: Broadcasting identity packet
25/10/19 12:11 PM	NetworkManager	<warn>  [1571965875.4696] dhcp4 (wlp3s0f0u7i2): request timed out
25/10/19 12:11 PM	NetworkManager	<info>  [1571965875.4697] dhcp4 (wlp3s0f0u7i2): state changed unknown -> timeout
25/10/19 12:11 PM	NetworkManager	<info>  [1571965875.5083] dhcp4 (wlp3s0f0u7i2): canceled DHCP transaction
25/10/19 12:11 PM	NetworkManager	<info>  [1571965875.5083] dhcp4 (wlp3s0f0u7i2): state changed timeout -> done
[... cut ...]
25/10/19 12:11 PM	NetworkManager	<warn>  [1571965917.4326] sup-iface[0x56327f70b200,wlp3s0f0u7i2]: connection disconnected (reason 1)
25/10/19 12:11 PM	NetworkManager	<info>  [1571965917.9381] device (wlp3s0f0u7i2): supplicant interface state: disconnected -> scanning
25/10/19 12:11 PM	NetworkManager	<info>  [1571965917.9382] device (p2p-dev-wlp3s0f0u7i2): supplicant management interface state: disconnected -> scanning
25/10/19 12:12 PM	bluetoothd	Endpoint unregistered: sender=:1.133 path=/MediaEndpoint/A2DPSink/sbc
25/10/19 12:12 PM	bluetoothd	Endpoint unregistered: sender=:1.133 path=/MediaEndpoint/A2DPSource/sbc

brektrou, 2019-11-04:
It seems that **NetworkManager** is in conflict with **wpa_supplicant** on your system. You must turn off **NetworkManager** service if you want to connect with **wpa_supplicant** and vice versa.

slesru, 2020-05-29:
I have the same messages, just installed from current git , ubuntu 20.04:

[Fri May 29 19:15:58 2020] RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
[Fri May 29 19:16:00 2020] RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
[Fri May 29 19:17:30 2020] RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
[Fri May 29 19:17:33 2020] RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
[Fri May 29 19:18:05 2020] RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
[Fri May 29 19:18:07 2020] RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
[Fri May 29 19:18:39 2020] RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
[Fri May 29 19:18:41 2020] RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
[Fri May 29 19:19:13 2020] RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
[Fri May 29 19:19:15 2020] RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
[Fri May 29 19:19:48 2020] RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
[Fri May 29 19:19:50 2020] RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
[Fri May 29 19:20:22 2020] RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
[Fri May 29 19:20:24 2020] RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
[Fri May 29 19:20:54 2020] RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
[Fri May 29 19:20:56 2020] RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
[Fri May 29 19:21:02 2020] RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
[Fri May 29 19:21:04 2020] RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
[Fri May 29 19:21:32 2020] RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
[Fri May 29 19:21:34 2020] RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
[Fri May 29 19:22:01 2020] RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
[Fri May 29 19:22:03 2020] RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
[Fri May 29 19:22:35 2020] RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
[Fri May 29 19:22:37 2020] RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey
[Fri May 29 19:22:39 2020] RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey

Looks like these messages are harmless, but it is not clear why are they appear...

Thank you!

tclovisct, 2020-10-08:
Hi, 

I have the same problems as "kernel: [951.480756] RTW: ERROR [RFK-CHK] RF-K not allowed due to ifaces under site-survey", using the 5.8.0-21-lowlatency kernel, on Ubuntu 20.10. Does anyone have a tip on how to stop these messages?

silgon, 2020-12-07:
same here, any workaround?

slesru, 2020-12-08:
Well, I commented this message out. Everything works just fine.

silgon, 2020-12-08:
So. I ended up fixing the BSSID (so it doesn't scan anymore) which you can see in the configuration below:
[image]
In my case, since it's a remote computer with no X11 access. I use `nmtui`. The configuration looks something like the following:
[image]

srid, 2021-01-23:
@silgon 

> So. I ended up fixing the BSSID (so it doesn't scan anymore) 

What exactly did you do? Clear the BSSID field? What was the value in it before?

silgon, 2021-01-23:
What I mean by fixing is to set the `BSSID` value of my router which has the form of a `MAC` address. And since your computer has now the fixed value of the device it's looking for, it will not look any more for other devices with the same `SSID`. Thus, once you're connected, you're not scanning any more, which is the process that, for some reason, makes the failure. The `BBSID` you'll find it in the configuration of your router. Good luck @srid  ;)

unifreak, 2023-01-08:
Filling `BSSID` doesn't do good for me. But this problem maybe related to your VPN (like v2raya). Try turn if off and see whether the errors still pouring in. I confirmed on my desktop that enable v2raya enable this error. Worth checking.