brektrou/rtl8821CU #28: anyone know how to istall this drive on openwrt?

jinglei207, 2019-11-27:


dausruddin, 2021-03-03:
Try this perhaps?

https://github.com/immortalwrt/immortalwrt/tree/d799ed3ce59c85929facc30a8834f3838c7a95a7/package/ctcgfw/rtl8821cu

AddaxSoft, 2021-03-03:
please report back if it worked!

dausruddin, 2021-03-03:
I actually compiled it into my firmware succesfully, but my usd adapter wouldn't work. Would be great if someone can tell me what packages are essentials to have wifi working on openwrt system, because I might be missing some packages.

tony1016, 2021-03-03:
lsusb -t show driver status

dausruddin, 2021-03-04:
I had my friend tested the firmware. Today I want to rebuild my firmware with the driver mentioned above and it failed to build.

tony1016, 2021-03-05:
> please report back if it worked!

yes,it works on 19.07.7

tony1016, 2021-03-05:
> I had my friend tested the firmware. Today I want to rebuild my firmware with the driver mentioned above and it failed to build.
> I got the same error as this [immortalwrt/immortalwrt#249](https://github.com/immortalwrt/immortalwrt/issues/249)
> 
> Must be something from openwrt repo. At this point, I am thinking to quit as this is my limit of knowledge. My RTL8811CU USB adapter is on its way to my home though XD

I build it with offical 19.07.0 successfully.And it's perfect,THX

tony1016, 2021-03-05:
My build 
https://drive.google.com/file/d/1pV5FvWZODYOcsZuQuvrT1WZPIAwwvEjW/view?usp=sharing

dausruddin, 2021-03-09:
Guys! If you are not able to get wifi working, see this issue https://github.com/immortalwrt/immortalwrt/issues/299.

TL;DR:
- Check kernel log after plugging in the adapter.
- If the adapter [detected as storage](https://github.com/immortalwrt/immortalwrt/issues/299#issuecomment-791137162), install [usb-modeswitch](https://openwrt.org/packages/pkgdata/usb-modeswitch).

It is working for me now. You might also need `wpad` and `iwinfo` as mentioned in link of the issue I attached above.
Give feedback on your progress!

Boos4721, 2022-03-25:
dude, anyone knows about rtl88x2bu on openwrt ?

trongbang2, 2023-10-22:
> My build https://drive.google.com/file/d/1pV5FvWZODYOcsZuQuvrT1WZPIAwwvEjW/view?usp=sharing

Sorry, how to install this driver in openwrt 19.07.
I got the Executing package manager error

Unknown package 'kmod-rtl8821cu'.

Errors

Collected errors:
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-rtl8821cu found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-rtl8821cu.

The opkg install command failed with code 255

Thank you!.