brektrou/rtl8821CU #30: New driver version 5.8.1

brektrou, 2019-12-25:
I got this driver from [here](https://support.dlink.com/ProductInfo.aspx?m=DWA-171)

Clone the driver:
`git clone -b 5.8.1 https://github.com/brektrou/rtl8821CU.git rtl8821CU-5.8.1`

Build with **make**:
`cd rtl8821CU-5.8.1`
`make`
`sudo make install`

Or build with **dkms**:
`sudo cp -rvf rtl8821CU-5.8.1 /usr/src`
`sudo dkms add rtl8821CU/5.8.1`
`sudo dkms install rtl8821CU/5.8.1 -k $(uname -r)`

[Driver info & changelog](https://github.com/brektrou/rtl8821CU/raw/5.8.1/changelog.pdf)

jinglei207, 2019-12-29:
how to autoswitch to wifi when plugged in?

birdofprey, 2020-03-29:
its so big 牛逼

kylroma, 2020-05-12:
Thank you! I've installed the driver in manjaro.

igorferret, 2020-05-22:
hi.... it works fine to navigate, but i cant packet injection.... packet injection not works.... can you help me?

> I got this driver from [here](https://support.dlink.com/ProductInfo.aspx?m=DWA-171)
> 
> Clone the driver:
> `git clone -b 5.8.1 https://github.com/brektrou/rtl8821CU.git rtl8821CU-5.8.1`
> 
> Build & install with **make**:
> `cd rtl8821CU-5.8.1`
> `make`
> `sudo make install`
> 
> Or build & install with **dkms**:
> `sudo cp -rvf rtl8821CU-5.8.1 /usr/src`
> `sudo dkms add rtl8821CU/5.8.1`
> `sudo dkms install rtl8821CU/5.8.1 -k $(uname -r)`
> 
> [Driver info & changelog](https://github.com/brektrou/rtl8821CU/raw/5.8.1/changelog.pdf)

Sirojio, 2020-07-30:
I have this problem: 
# sudo dkms install rtl8821CU/5.8.1 -k $(uname -r)

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' KVER=5.7.0-kali1-amd64....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.7.0-kali1-amd64 (x86_64)
Consult /var/lib/dkms/rtl8821CU/5.8.1/build/make.log for more information.

Kali on VirtualBox
root@kali:~# grep VERSION /etc/os-release
VERSION="2020.3"
VERSION_ID="2020.3"
VERSION_CODENAME="kali-rolling"
root@kali:~# uname -v
#1 SMP Debian 5.7.6-1kali2 (2020-07-01)
root@kali:~# uname -r
5.7.0-kali1-amd64

How can I decide this problem?

Sirojio, 2020-07-30:
> I got this driver from here
> Clone the driver:
> git clone -b 5.8.1 https://github.com/brektrou/rtl8821CU.git rtl8821CU-5.8.1
> Build & install with make:
> cd rtl8821CU-5.8.1
> make
> sudo make install
> Or [...]

Could you help?

devbharat, 2020-08-28:
Hey! Does the driver+firmware+module support packet injection? I can get the monitor mode to work,  but I can't seem to inject packets.

OfflineLubuntu, 2020-10-07:
I'm a bit new to cloning stuff like git (I tried before with 8821au as a premade ZIP but that was wrong since mine is C not A) so I'm going to try and do that with GitHub desktop since my only internet connection is with windows and the whole point of doing the DWA-171 is to get internet on my lubuntu 32bit =/

once I copy and paste the folder (guess never zipped but it's small so who cares) then I guess I can follow steps from there, I hope...

luckylittle, 2020-12-13:
@brektrou I am unable to compile it. Below are the details:
* Red Hat Enterprise Linux release `8.3` (Ootpa)
* Kernel `4.18.0-240.1.1.el8_3.x86_64`
* Error:

```text
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.18.0-240.1.1.el8_3.x86_64/build M=/home/USER/Downloads/WiFi/rtl8821CU-5.8.1  modules
make[1]: Entering directory '/usr/src/kernels/4.18.0-240.1.1.el8_3.x86_64'
  CC [M]  /home/USER/Downloads/WiFi/rtl8821CU-5.8.1/os_dep/linux/rtw_android.o
/home/USER/Downloads/WiFi/rtl8821CU-5.8.1/os_dep/linux/rtw_android.c: In function ‘rtw_android_priv_cmd’:
/home/USER/Downloads/WiFi/rtl8821CU-5.8.1/os_dep/linux/rtw_android.c:669:62: error: macro "access_ok" passed 3 arguments, but takes just 2
  if (!access_ok(VERIFY_READ, priv_cmd.buf, priv_cmd.total_len)) {
                                                              ^
/home/USER/Downloads/WiFi/rtl8821CU-5.8.1/os_dep/linux/rtw_android.c:669:7: error: ‘access_ok’ undeclared (first use in this function)
  if (!access_ok(VERIFY_READ, priv_cmd.buf, priv_cmd.total_len)) {
       ^~~~~~~~~
/home/USER/Downloads/WiFi/rtl8821CU-5.8.1/os_dep/linux/rtw_android.c:669:7: note: each undeclared identifier is reported only once for each function it appears in
make[2]: *** [scripts/Makefile.build:316: /home/USER/Downloads/WiFi/rtl8821CU-5.8.1/os_dep/linux/rtw_android.o] Error 1
make[1]: *** [Makefile:1544: _module_/home/USER/Downloads/WiFi/rtl8821CU-5.8.1] Error 2
make[1]: Leaving directory '/usr/src/kernels/4.18.0-240.1.1.el8_3.x86_64'
make: *** [Makefile:2327: modules] Error 2
```

keil555, 2020-12-13:
It works! Thanks!

J-Rojas, 2021-12-03:
I'm getting this error with kernel 5.13

/home/jrojas/work/github/rtl8821CU-5.8.1/os_dep/linux/recv_linux.c:358:55: error: ‘GRO_DROP’ undeclared (first use in this function)
  358 |    if (rtw_napi_gro_receive(&padapter->napi, pskb) != GRO_DROP)
      |                                                       ^~~~~~~~

iamhere2, 2021-12-17:
@J-Rojas Have the same problem.
Found appropriate patch here: https://github.com/RinCat/RTL88x2BU-Linux-Driver/issues/81
Will try to use it, and tell you about the results.

mozeus14, 2022-08-16:
Building module:
cleaning build area...
'make' KVER=5.18.0-kali5-amd64.........................(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.18.0-kali5-amd64 (x86_64)
Consult /var/lib/dkms/rtl8821CU/5.8.1/build/make.log for more information.