brektrou/rtl8821CU #34: Device not recognized on Debian 10. Driver probably not loaded although it's installed.

AXDOOMER, 2020-01-15:
I can't use my Wi-Fi dongle on Debian 10. I installed  the driver via DKMS, but the new adapter doesn't appear in the NetworkManager applet. The light doesn't bright up on the device like it does on my Ubuntu and OpenSuse system where it works.  Here's some info so you can help:

`uname -a`: 
```
Linux doom 4.19.0-6-amd64 #1 SMP Debian 4.19.67-2+deb10u2 (2019-11-11) x86_64 GNU/Linux
```
`dmesg`: 
```
[  287.400156] usb 2-3: USB disconnect, device number 3
[  289.288446] usb 2-3: new high-speed USB device number 4 using xhci_hcd
[  289.436869] usb 2-3: New USB device found, idVendor=0bda, idProduct=c811, bcdDevice= 2.00
[  289.436874] usb 2-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  289.436876] usb 2-3: Product: 802.11ac NIC
[  289.436878] usb 2-3: Manufacturer: Realtek
[  289.436880] usb 2-3: SerialNumber: 123456
```
`lsusb`: 
```
Bus 002 Device 004: ID 0bda:c811 Realtek Semiconductor Corp. 
```
`sudo dkms status`:
```
broadcom-sta, 6.30.223.271, 4.19.0-6-amd64, x86_64: installed
rtl8821CU, 5.4.1: added
```

`ls /lib/modules/4.19.0-6-amd64/kernel/drivers/net/wireless/realtek`:
```
rtl818x  rtl8xxxu  rtlwifi
```
`/lib/modules/4.19.0-6-amd64/kernel/drivers/net/wireless/realtek# ls -R`:
```
.:
rtl818x  rtl8xxxu  rtlwifi

./rtl818x:
rtl8180  rtl8187

./rtl818x/rtl8180:
rtl818x_pci.ko

./rtl818x/rtl8187:
rtl8187.ko

./rtl8xxxu:
rtl8xxxu.ko

./rtlwifi:
btcoexist  rtl8192c   rtl8192cu  rtl8192ee  rtl8723ae  rtl8723com  rtl_pci.ko  rtlwifi.ko
rtl8188ee  rtl8192ce  rtl8192de  rtl8192se  rtl8723be  rtl8821ae   rtl_usb.ko

./rtlwifi/btcoexist:
btcoexist.ko

./rtlwifi/rtl8188ee:
rtl8188ee.ko

./rtlwifi/rtl8192c:
rtl8192c-common.ko

./rtlwifi/rtl8192ce:
rtl8192ce.ko

./rtlwifi/rtl8192cu:
rtl8192cu.ko

./rtlwifi/rtl8192de:
rtl8192de.ko

./rtlwifi/rtl8192ee:
rtl8192ee.ko

./rtlwifi/rtl8192se:
rtl8192se.ko

./rtlwifi/rtl8723ae:
rtl8723ae.ko

./rtlwifi/rtl8723be:
rtl8723be.ko

./rtlwifi/rtl8723com:
rtl8723-common.ko

./rtlwifi/rtl8821ae:
rtl8821ae.ko
```

brektrou, 2020-01-16:
Your only added the driver to DKMS
`rtl8821CU, 5.4.1: added`

Just install it:
`sudo dkms install rtl8821CU/5.4.1 -k $(uname -r)`

And restart your PC

AXDOOMER, 2020-01-17:
I ran `sudo dkms install rtl8821CU/5.4.1 -k $(uname -r)`. The first time, there was an error. I checked the log file `/var/lib/dkms/rtl8821CU/5.4.1/build/make.log` and it revealed that it failed to make because the program `bc` wasn't installed. Installing `bc` and running the command again successfully installed the driver and I didn't have to restart my computer.

Thank you.

closed 2020-01-17 by AXDOOMER