brektrou/rtl8821CU #36: Missing bluetooth

DraTrav, 2020-01-18:
The wifi part works but no bluetooth adapter is detected even though the chip supports bluetooth

etem, 2020-02-15:
Same issue here on Fedora 30 kernel 5.4.18

etem, 2020-02-23:
I've managed to fix that following [this](https://fedoraproject.org/wiki/How_to_debug_Bluetooth_problems) guide for Fedora.

Basically following sequence of commands needs to be applied
- `usb_modeswitch -KW -v 0bda -p 1a2b` - switching adapter mode from USB storage to network adapter
- `systemctl start bluetooth.service` - starting Bluetooth service
- `hciconfig up` - enabling Bluetooth device if it's in down state

etem, 2020-02-23:
Btw, I'm using version 5.4.1 of the driver.

DraTrav, 2020-02-27:
Perfect! that worked, wish it was in the main guide

closed 2020-02-27 by DraTrav