brektrou/rtl8821CU #37: Cannot set monitor mode.

breisig, 2020-01-25:
Installed the latest rtl8821CU from git today on Linux kernel 5.4.14 using the following usb device.

USB ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC

[    8.912689] 8821cu: no symbol version for module_layout
[    8.912705] 8821cu: loading out-of-tree module taints kernel.
[    9.000937] 8821cu: module verification failed: signature and/or required key missing - tainting kernel
[    9.011339] RTW: module init start
[    9.011341] RTW: rtl8821cu v5.4.1_28754.20180921_COEX20180712-3232
[    9.011341] RTW: build time: Jan 24 2020 21:08:44
[    9.011342] RTW: rtl8821cu BT-Coex version = COEX20180712-3232
[    9.011462] RTW: [HALMAC]11692M
               HALMAC_MAJOR_VER = 1
               HALMAC_PROTOTYPE_VER = 4
               HALMAC_MINOR_VER = 19
               HALMAC_PATCH_VER = 3
[    9.012713] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[    9.167885] RTW: HW EFUSE
[    9.167886] RTW: 0x000: 29 81 00 BC  09 00 31 00  6F 04 A4 10  12 00 30 0B
[    9.167889] RTW: 0x010: 23 23 23 24  24 24 2B 2B  2B 2C 2C 02  FF FF FF FF
[    9.167891] RTW: 0x020: FF FF 2B 2A  28 28 27 27  28 28 29 29  27 29 29 29
[    9.167893] RTW: 0x030: 24 FF FF FF  FF FF D0 FF  FF FF 23 23  23 24 24 24
[    9.167894] RTW: 0x040: 2B 2B 2B 2C  2C 02 FF FF  FF FF FF FF  FF FF FF FF
[    9.167896] RTW: 0x050: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[    9.167898] RTW: 0x060: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[    9.167899] RTW: 0x070: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[    9.167901] RTW: 0x080: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[    9.167903] RTW: 0x090: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[    9.167905] RTW: 0x0A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[    9.167906] RTW: 0x0B0: FF FF FF FF  FF FF FF FF  7F 06 1A 00  FF FF FF FF
[    9.167908] RTW: 0x0C0: FF 22 00 41  00 00 00 00  00 FF 22 FF  FF FF FF FF
[    9.167910] RTW: 0x0D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[    9.167912] RTW: 0x0E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[    9.167913] RTW: 0x0F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[    9.167915] RTW: 0x100: DA 0B 11 C8  C3 46 02 40  A5 EF 4E 92  81 09 03 52
[    9.167917] RTW: 0x110: 65 61 6C 74  65 6B 0E 03  38 30 32 2E  31 31 61 63
[    9.167919] RTW: 0x120: 20 4E 49 43  08 03 31 32  33 34 35 36  FF FF FF FF
[    9.167920] RTW: 0x130: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[    9.167922] RTW: 0x140: FF FF FF FF  00 00 31 0F  FF FF FF FF  FF FF FF FF
[    9.167924] RTW: 0x150: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[    9.167926] RTW: 0x160: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[    9.167927] RTW: 0x170: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[    9.167929] RTW: 0x180: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[    9.167931] RTW: 0x190: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[    9.167932] RTW: 0x1A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[    9.167934] RTW: 0x1B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[    9.167936] RTW: 0x1C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[    9.167938] RTW: 0x1D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[    9.167939] RTW: 0x1E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[    9.167941] RTW: 0x1F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[    9.168106] RTW: hal_com_config_channel_plan chplan:0x7F
[    9.176970] RTW: [HALMAC][ALWAYS]shall R reg twice!!
[    9.178990] RTW: WARN [HALMAC][WARN]H2C/C2H ver is compatible!!
[    9.266895] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[    9.267083] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[    9.267268] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[    9.267440] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[    9.267610] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[    9.267780] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[    9.267950] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[    9.268445] RTW: rtw_regsty_chk_target_tx_power_valid return _FALSE for band:0, path:0, rs:0, t:-1
[    9.270326] RTW: rtw_ndev_init(wlan0) if1 mac_addr=40:a5:ef:4e:92:81
[    9.270661] usbcore: registered new interface driver rtl8821cu
[    9.270662] RTW: module init ret=0
[    9.338068] RTW: [HALMAC][ALWAYS]shall R reg twice!!
[    9.339961] RTW: WARN [HALMAC][WARN]H2C/C2H ver is compatible!!
[   10.184918] RTW: wlan0- hw port(0) mac_addr =40:a5:ef:4e:92:81

I checked the Makefile and  monitor is enabled.
CONFIG_WIFI_MONITOR = y

But when I try to run airodump-ng, it says it cannot set monitor mode
root@server ~# airmon-ng check kill
root@server ~# airmon-ng start wlan0

PHY     Interface       Driver          Chipset

phy0    wlan0           rtl8821cu       Realtek Semiconductor Corp. 802.11ac NIC

Newly created monitor mode interface wlan0mon is *NOT* in monitor mode.
Removing non-monitor wlan0mon interface...

WARNING: unable to start monitor mode, please run "airmon-ng check kill"

Of course I tried 'airmon-ng check kill' but nothing happens.

WingsLikeEagles, 2020-06-16:
As it says in the README, you need to use the following line to enable monitor mode:
**iw dev wlan0 set monitor none**

Bramzor, 2020-11-15:
> iw dev wlan0 set monitor none

iw dev wlan0 set monitor none
command failed: Operation not supported (-95)

fdservices, 2022-03-05:
Got the same error and realised that usb_modeswitch had not worked. Ran it again in a terminal, and the error went away.