brektrou/rtl8821CU #38: NULL pointer dereference on 3.4.113 linux kernel

9600bps, 2020-02-02:
I've cross compiled the driver for armv7 sun8iw7p1 (Allwinner H3, Orange Pi PC), 3.4.113 linux kernel (Linux LibreELEC 3.4.113 #1 SMP PREEMPT Wed Mar 1 00:01:16 CET 2017 armv7l GNU/Linux).

But unfortunately the driver crashed with "Unable to handle kernel NULL pointer dereference at virtual address 00000000". Is 3.x kernel version supported? What information should I add to the issue to help investigate the reason?

I added `-g` to the Makefile and removed `--strip-debug`, but it doesn't help to make traces more informative. A chunk from dmesg:
```
[  258.213035] RTW: module init start
[  258.213053] RTW: rtl8821cu v5.4.1_28754.20180921_COEX20180712-3232
[  258.213063] RTW: build time: Feb  2 2020 11:40:39
[  258.213071] RTW: rtl8821cu BT-Coex version = COEX20180712-3232
[  258.213087] sw_usb_enable_hcd: usbc_num = 2
[  258.213096] [wifi_pm]: set wl_reg_on 1 !
[  258.223115] hci: ERR: sunxi_ehci is already enable, can not enable again
[  258.223130] hci: ERR: sunxi_ohci is already enable, can not enable again
[  258.223381] usbcore: registered new interface driver rtl8821cu
[  258.223395] RTW: module init ret=0
[  283.508176] ehci_irq: highspeed device connect
[  283.770109] usb 4-1: new high-speed USB device number 2 using sunxi-ehci
[  283.923109] RTW: [HALMAC]11692M
[  283.923120] HALMAC_MAJOR_VER = 1
[  283.923128] HALMAC_PROTOTYPE_VER = 4
[  283.923137] HALMAC_MINOR_VER = 19
[  283.923145] HALMAC_PATCH_VER = 3
[  283.925151] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[  284.185832] RTW: HW EFUSE
[  284.185866] RTW: 0x000: 29 81 00 BC  09 00 21 00  6E 04 A4 10  10 00 30 0B  
[  284.185969] RTW: 0x010: FF FF FF FF  FF FF FF FF  FF FF FF 02  FF FF FF FF  
[  284.186070] RTW: 0x020: FF FF 28 28  28 28 28 28  28 28 28 28  28 28 28 28  
[  284.186171] RTW: 0x030: 02 FF FF FF  FF FF EC FF  FF FF 2B 2B  2B 2B 2B 2B  
[  284.186272] RTW: 0x040: 30 30 30 30  30 33 FF FF  FF FF FF FF  FF FF FF FF  
[  284.186373] RTW: 0x050: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  284.186474] RTW: 0x060: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  284.186575] RTW: 0x070: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  284.186676] RTW: 0x080: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  284.186777] RTW: 0x090: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  284.186878] RTW: 0x0A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  284.186979] RTW: 0x0B0: FF FF FF FF  FF FF FF FF  7F 06 20 00  FF FF FF FF  
[  284.187080] RTW: 0x0C0: FF 01 00 41  00 00 00 00  00 FF 22 FF  FF FF FF FF  
[  284.187179] RTW: 0x0D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  284.187280] RTW: 0x0E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  284.187381] RTW: 0x0F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  284.187482] RTW: 0x100: DA 0B 11 C8  E3 46 02 1C  BF CE 84 15  CA 09 03 52  
[  284.187583] RTW: 0x110: 65 61 6C 74  65 6B 0E 03  38 30 32 2E  31 31 61 63  
[  284.187685] RTW: 0x120: 20 4E 49 43  08 03 31 32  33 34 35 36  FF FF FF FF  
[  284.187786] RTW: 0x130: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  284.187887] RTW: 0x140: FF FF FF FF  00 00 31 0F  FF FF FF FF  FF FF FF FF  
[  284.187988] RTW: 0x150: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  284.188089] RTW: 0x160: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  284.188190] RTW: 0x170: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  284.188292] RTW: 0x180: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  284.188393] RTW: 0x190: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  284.188494] RTW: 0x1A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  284.188595] RTW: 0x1B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  284.188795] RTW: 0x1C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  284.188897] RTW: 0x1D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  284.188998] RTW: 0x1E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  284.189099] RTW: 0x1F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  284.189584] RTW: hal_com_config_channel_plan chplan:0x7F
[  284.202703] RTW: [HALMAC][ALWAYS]shall R reg twice!!
[  284.206592] RTW: WARN [HALMAC][WARN]H2C/C2H ver is compatible!!
[  284.321491] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[  284.321864] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[  284.322360] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[  284.322727] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[  284.323101] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[  284.323476] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[  284.323851] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[  284.348957] RTW: rtw_regsty_chk_target_tx_power_valid return _FALSE for band:0, path:0, rs:0, t:-1
[  284.375511] Unable to handle kernel NULL pointer dereference at virtual address 00000000
[  284.375552] pgd = c0004000
[... 112 lines cut ...]
[  284.760530] 9d60: e8bee180 e8a04a48 ef1a9de4 ef1a9d78 c038a3a0 c033fb70 00000001 00000000
[  284.760530] 9d80: 00000000 00000000 00001388 00000000 e8bee140 ee641468 e8bee184 00000001
[  284.760530] 9da0: 00000000 00000004 e8bee184 eebfe250 00000000 ee94fc00 c03807d4 ee641400
[  284.760530] 9dc0: 00000001 c034277c c0bc1ffc 00000008 c0bc182c 00000000 ef1a9dfc ef1a9de8
[  284.760530] 9de0: c03925dc c0389e28 ee641400 c0bc1ffc ef1a9e14 ef1a9e00 c038bc10 c039258c
[  284.760530] 9e00: ee641468 c0cafd34 ef1a9e3c ef1a9e18 c0342640 c038bbe8 c0bc1ffc ee641468
[  284.760530] 9e20: c034277c 00000000 ee95c068 c0bc182c ef1a9e54 ef1a9e40 c03427b4 c034257c
[  284.760530] 9e40: 00000000 ee641468 ef1a9e7c ef1a9e58 c0340c74 c0342788 ef179a78 ef101c3c
[  284.760530] 9e60: ef120400 ee641468 c0bc1990 ee64149c ef1a9e9c ef1a9e80 c034251c c0340bec
[  284.760530] 9e80: ee641468 c0bc1990 ee641468 00000000 ef1a9ebc ef1a9ea0 c0341a4c c034249c
[  284.760530] 9ea0: ee641468 00000000 ee641470 00000000 ef1a9f04 ef1a9ec0 c033ffe8 c0341a20
[  284.760530] 9ec0: 00000007 e8a04280 39383100 3538333a ee94fc00 00000000 ef1a9f04 ee641400
[  284.760530] 9ee0: ee641468 00000001 ee95c000 00000000 ee94fc18 00000000 ef1a9f24 ef1a9f08
[  284.760530] 9f00: c0383860 c033fb70 ee938e38 00000001 ee641400 ee95c000 ef1a9fb4 ef1a9f28
[  284.760530] 9f20: c0384a04 c03836b4 09f43d74 00000000 00000000 0000000a 00000001 ee95c09c
[  284.760530] 9f40: ee938e08 00000002 ee938c20 ee938c00 ee94fc00 ee938e89 ee938c20 00000000
[  284.760530] 9f60: ee95c000 ee938e00 ef047f0c 05010013 00000000 00000000 ef0f39c0 c0049ed8
[  284.760530] 9f80: ef1a9f80 ef1a9f80 00000013 ef047efc 00000000 c0383f8c 00000013 00000000
[  284.760530] 9fa0: 00000000 00000000 ef1a9ff4 ef1a9fb8 c0049644 c0383f98 00000000 00000000
[  284.760530] 9fc0: 00000000 00000000 00000000 00000000 ef1a9fd0 ef1a9fd0 00000000 ef047efc
[  284.760530] 9fe0: c00495b0 c000f300 00000000 ef1a9ff8 c000f300 c00495bc 7b77ffdf bffbbfff
[  284.760530] Code: e4805004 e4806004 e4807004 e4808004 (e480e004) 
[  285.442781] ---[ end trace 65e9928d154e6a5d ]---

```

brektrou, 2020-02-07:
This driver doesn't support Linux version 3

rrafal1337, 2020-02-07:
I can propose Arch Linux ARM for your allwinner board. Already tested on my NanoPi Neo and it works fine even if archlinuxarm doesn't say they support it. All allwinnerboards shares same u-boot-sunxi-with-spl.bin na boot.scr and works on architecture armv7. Here is link: https://archlinuxarm.org/platforms/armv7/allwinner/a20-olinuxino-micro
There is no need to install u-boot package later.

9600bps, 2020-02-08:
Ok, thx guys.

closed 2020-02-08 by 9600bps