brektrou/rtl8821CU #39: version magic problem

zjiekai, 2020-02-14:
```
$ make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.3.0-28-generic/build M=/home/zjk/github/rtl8821CU  modules
make[1]: Entering directory '/usr/src/linux-headers-5.3.0-28-generic'
  Building modules, stage 2.
  MODPOST 1 modules
make[1]: Leaving directory '/usr/src/linux-headers-5.3.0-28-generic'

$ sudo modprobe 8821cu
modprobe: ERROR: could not insert '8821cu': Unknown symbol in module, or unknown parameter (see dmesg)
```

```
$ dmesg
[55260.676417] usbcore: registered new interface driver rtl88xxau
[55276.054904] usb 1-5.3: new high-speed USB device number 8 using xhci_hcd
[55276.155247] usb 1-5.3: New USB device found, idVendor=0bda, idProduct=c811, bcdDevice= 2.00
[55276.155252] usb 1-5.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[55276.155255] usb 1-5.3: Product: 802.11ac NIC
[55276.155258] usb 1-5.3: Manufacturer: Realtek
[55276.155261] usb 1-5.3: SerialNumber: 123456
[55276.201639] 8821cu: version magic '5.0.0-37-generic SMP mod_unload ' should be '5.3.0-28-generic SMP mod_unload '
[55713.343830] usbcore: deregistering interface driver rtl88xxau
[56170.871989] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[56170.872123] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[56170.873184] 8821cu: version magic '5.0.0-37-generic SMP mod_unload ' should be '5.3.0-28-generic SMP mod_unload '
```

zjiekai, 2020-02-14:
Seems I'm having a path problem?

SatanicRaw, 2020-02-16:
You may need to rebuild the kernel with the compiler installed on the system.

zjiekai, 2020-02-16:
`modprobe` probably is using the wrong file.  `sudo insmod ./8821cu.ko` works fine.

SatanicRaw, 2020-03-08:
Most likely, everything works fine, you just use insmod 8821cu instead of modprobe 8821cu. Insmod does not load the modules that 8821cu depends on. As far as I understand, in this case it is enough to execute modprobe cfg80211.

brektrou, 2020-03-08:
Are you just run `sudo insmod ./8821cu.ko` command after you installing or updating kernels?

closed 2020-03-09 by zjiekai