brektrou/rtl8821CU #42: TOTOLINK A650UA with RTL8811CU still not detected, why?

Sayyid01, 2020-02-26:
Hi, I have usb wifi adapter TOTOLINK A650UA, and I was installing your driver, but my pc just detected my usb adapter like this

`Bus 001 Device 024: ID 0bda:2006 Realtek Semiconductor Corp. 
`
The driver was successfully installed, but I don't know why my USB not detected as network interfaces at all.

here is the detail :
from `sudo dkms status
`
`rtl8821CU, 5.4.1, 5.3.0-40-generic, x86_64: installed
`

and from 

> sudo lshw -C network
> 
> *-network                 
>        description: Wireless interface
>        product: QCA9565 / AR9565 Wireless Network Adapter
>        vendor: Qualcomm Atheros
>        physical id: 0
>        bus info: pci@0000:01:00.0
>        logical name: wlp1s0
>        version: 01
>        serial: f0:03:8c:43:c9:ab
>        width: 64 bits
>        clock: 33MHz
>        capabilities: pm msi pciexpress bus_master cap_list rom ethernet physical wireless
>        configuration: broadcast=yes driver=ath9k driverversion=5.3.0-40-generic firmware=N/A ip=192.168.44.7 latency=0 link=yes multicast=yes wireless=IEEE 802.11
>        resources: irq:37 memory:fea00000-fea7ffff memory:fea80000-fea8ffff
> *-network
>        description: Ethernet interface
>        product: RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller
>        vendor: Realtek Semiconductor Co., Ltd.
>        physical id: 0
>        bus info: pci@0000:02:00.0
>        logical name: enp2s0
>        version: 10
>        serial: 2c:4d:54:2b:e3:80
>        capacity: 1Gbit/s
>        width: 64 bits
>        clock: 33MHz
>        capabilities: pm msi pciexpress msix vpd bus_master cap_list ethernet physical tp mii 10bt 10bt-fd 100bt 100bt-fd 1000bt-fd autonegotiation
>        configuration: autonegotiation=on broadcast=yes driver=r8169 firmware=rtl8168g-3_0.0.1 04/23/13 latency=0 link=no multicast=yes port=MII
>        resources: irq:35 ioport:e000(size=256) memory:fe904000-fe904fff memory:fe900000-fe903fff

anhbinhvodanh, 2020-04-30:
HI @Sayyid01 
I have same problem with my TOTOLINK A650UA.
Did you make it work?
Thanks.

SatanicRaw, 2020-05-01:
@Sayyid01 @anhbinhvodanh 

Hey. Try adding the ID of your adapters to the driver.
Clone \ download repository, open file
`os_dep/linux/usb_intf.c`
find section

```
#ifdef CONFIG_RTL8821C
        / * === Realtek demoboard === * /
        {USB_DEVICE_AND_INTERFACE_INFO (USB_VENDER_ID_REALTEK, 0xb82b, 0xff, 0xff, 0xff), .driver_info = RTL8821C}, / * 8821CU * /
        {USB_DEVICE_AND_INTERFACE_INFO (USB_VENDER_ID_REALTEK, 0xb820, 0xff, 0xff, 0xff), .driver_info = RTL8821C}, / * 8821CU * /
        {USB_DEVICE_AND_INTERFACE_INFO (USB_VENDER_ID_REALTEK, 0xC821, 0xff, 0xff, 0xff), .driver_info = RTL8821C}, / * 8821CU * /
        {USB_DEVICE_AND_INTERFACE_INFO (USB_VENDER_ID_REALTEK, 0xC820, 0xff, 0xff, 0xff), .driver_info = RTL8821C}, / * 8821CU * /
        {USB_DEVICE_AND_INTERFACE_INFO (USB_VENDER_ID_REALTEK, 0xC82A, 0xff, 0xff, 0xff), .driver_info = RTL8821C}, / * 8821CU * /
        {USB_DEVICE_AND_INTERFACE_INFO (USB_VENDER_ID_REALTEK, 0xC82B, 0xff, 0xff, 0xff), .driver_info = RTL8821C}, / * 8821CU * /
        {USB_DEVICE_AND_INTERFACE_INFO (USB_VENDER_ID_REALTEK, 0xC811, 0xff, 0xff, 0xff), .driver_info = RTL8821C}, / * 8811CU * /
        {USB_DEVICE_AND_INTERFACE_INFO (USB_VENDER_ID_REALTEK, 0x8811, 0xff, 0xff, 0xff), .driver_info = RTL8821C}, / * 8811CU * /
        / * === Customer ID === * /
        {USB_DEVICE (0x2001, 0x331d), .driver_info = RTL8821C}, / * D-Link - DWA-171C * /
#endif
```

And before `#endif` add a line (in @Sayyid01 case, @anhbinhvodanh your adapter may have a different ID, you can see it through `lsusb`)

`{USB_DEVICE_AND_INTERFACE_INFO (USB_VENDER_ID_REALTEK, 0x2006, 0xff, 0xff, 0xff), .driver_info = RTL8821C},`

And then build the module

anhbinhvodanh, 2020-05-01:
@SatanicRaw 
Yes, You are a genius. It works like a charm. 
Why I can see your answer soon before buying a new adapter :((.

Thank you so much.

Sayyid01, 2020-05-01:
@SatanicRaw I have not tested it yet coz I using windows (for now 😄), but if it works on @anhbinhvodanh device, then it should work for me too.

Thanks for your help, and I will close this issue 😁.

closed 2020-05-01 by Sayyid01