brektrou/rtl8821CU #48: RaspBerry Pi 4B / Raspbian Buster Make fail: Error! Bad return status for module build on kernel: 4.19.97-v7l+ (armv7l)

zambalee, 2020-03-29:
Hardware: Raspberry Pi 4B;
OS: Raspbian Buster;

Process:
1. Install tools:
    `apt install -y gcc linux-header git dkms`
2. Clone driver:
  `git clone https://github.com/brektrou/rtl8821CU.git`
3. Install use dkms-install.sh
`root@raspberrypi:~/rtl8821CU# ./dkms-install.sh
About to run dkms install steps...

Creating symlink /var/lib/dkms/rtl8821CU/5.4.1/source ->
                 /usr/src/rtl8821CU-5.4.1

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' KVER=4.19.97-v7l+...(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.19.97-v7l+ (armv7l)
Consult /var/lib/dkms/rtl8821CU/5.4.1/build/make.log for more information.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' KVER=4.19.97-v7l+...(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.19.97-v7l+ (armv7l)
Consult /var/lib/dkms/rtl8821CU/5.4.1/build/make.log for more information.
Finished running dkms install steps.
`
4. Error log
`root@raspberrypi:~/rtl8821CU# more  /var/lib/dkms/rtl8821CU/5.4.1/build/make.log
DKMS make.log for rtl8821CU-5.4.1 for kernel 4.19.97-v7l+ (armv7l)
Sun 29 Mar 09:50:52 BST 2020
/bin/sh: 1: bc: not found
make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/4.19.97-v7l+/build M=/var/lib/dkms/rtl8821CU/5.4.1/build  modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.97-v7l+'
Makefile:614: arch/armv7l/Makefile: No such file or directory
make[1]: *** No rule to make target 'arch/armv7l/Makefile'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-4.19.97-v7l+'
make: *** [Makefile:2213: modules] Error 2`

rrafal1337, 2020-03-29:
Hi. You forgot to install bc package.
```bash
apt install bc
```

zambalee, 2020-03-29:
Hi Rafał,

Thanks for your help!  But after installed "bc" by using "apt install -y bc", the same error still appears:

**Platform Detail**
- root@raspberrypi:~# cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

- root@raspberrypi:~# cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3
.
.
.
Hardware        : BCM2835
Revision        : b03111
Serial          : 100000000e75a992
Model           : Raspberry Pi 4 Model B Rev 1.1

**Post Output**
=================================================
root@raspberrypi:~/rtl8821CU# ./dkms-install.sh
About to run dkms install steps...
Error! DKMS tree already contains: rtl8821CU-5.4.1
You cannot add the same module/version combo more than once.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' KVER=4.19.97-v7l+...(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.19.97-v7l+ (armv7l)
Consult /var/lib/dkms/rtl8821CU/5.4.1/build/make.log for more information.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' KVER=4.19.97-v7l+...(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.19.97-v7l+ (armv7l)
Consult /var/lib/dkms/rtl8821CU/5.4.1/build/make.log for more information.
Finished running dkms install steps.
=================================================

**/var/lib/dkms/rtl8821CU/5.4.1/build/make.log**
=================================================
root@raspberrypi:~/rtl8821CU# more /var/lib/dkms/rtl8821CU/5.4.1/build/make.log
DKMS make.log for rtl8821CU-5.4.1 for kernel 4.19.97-v7l+ (armv7l)
Sun 29 Mar 16:35:20 BST 2020
make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/4.19.97-v7l+/build M=/var/lib/dk
ms/rtl8821CU/5.4.1/build  modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.97-v7l+'
Makefile:614: arch/armv7l/Makefile: No such file or directory
make[1]: *** No rule to make target 'arch/armv7l/Makefile'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-4.19.97-v7l+'
make: *** [Makefile:2213: modules] Error 2
=================================================

rrafal1337, 2020-03-29:
It seems that installation bc helped and you won't have same error again. This time there error is related to missing directory in kernel sources and you must follow instruction from readme file about avoiding errors during compilation. https://github.com/brektrou/rtl8821CU/#arm-architecture-tweak-for-this-driver-this-solves-compilation-problem-of-this-driver

zambalee, 2020-03-29:
Sorry for that, just missing the whole big picture......  

Thanks for your help! I should read the manual more carefully, now it works fine, thanks again!

Zamba

closed 2020-03-29 by zambalee