brektrou/rtl8821CU #50: Kali Linux 2020.1 errors

jimmyarn, 2020-04-12:
Please help.  I am a very new linux user.  When I attempt to compile the driver for install I get the following error.  What is going wrong here?

ali@kali:~$ sudo su
root@kali:~/build/rtl8821CU# apt-get update
Get:1 http://kali.download/kali kali-rolling InRelease [30.5 kB]
Get:2 http://kali.download/kali kali-rolling/contrib Sources [62.0 kB]
Get:3 http://kali.download/kali kali-rolling/main Sources [12.8 MB]
Get:4 http://kali.download/kali kali-rolling/non-free Sources [126 kB]    
Get:5 http://kali.download/kali kali-rolling/main amd64 Packages [16.3 MB]
Get:6 http://kali.download/kali kali-rolling/contrib amd64 Packages [99.6 kB]
Get:7 http://kali.download/kali kali-rolling/non-free amd64 Packages [193 kB]
Fetched 29.5 MB in 16s (1,794 kB/s)                                       
Reading package lists... Done
root@kali:/home/kali# mkdir -p ~/build
root@kali:/home/kali# cd ~/build
root@kali:~/build# git clone https://github.com/brektrou/rtl8821CU.git
Cloning into 'rtl8821CU'...
remote: Enumerating objects: 63, done.
remote: Counting objects: 100% (63/63), done.
remote: Compressing objects: 100% (62/62), done.
remote: Total 1403 (delta 23), reused 9 (delta 1), pack-reused 1340
Receiving objects: 100% (1403/1403), 5.57 MiB | 2.99 MiB/s, done.
Resolving deltas: 100% (796/796), done.
root@kali:~/build# cd ~/build/rtl8821CU
root@kali:~/build/rtl8821CU# make
/bin/sh: 1: bc: not found
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.4.0-kali2-amd64/build M=/root/build/rtl8821CU  modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-kali2-amd64'
/bin/sh: 1: bc: not found
  CC [M]  /root/build/rtl8821CU/core/rtw_cmd.o
In file included from /root/build/rtl8821CU/include/drv_types.h:30,
                 from /root/build/rtl8821CU/core/rtw_cmd.c:17:
/root/build/rtl8821CU/include/wifi.h:1032: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 1032 | #define IEEE80211_MAX_AMPDU_BUF 0x40
      | 
In file included from /root/build/rtl8821CU/include/osdep_service_linux.h:83,
                 from /root/build/rtl8821CU/include/osdep_service.h:50,
                 from /root/build/rtl8821CU/include/drv_types.h:27,
                 from /root/build/rtl8821CU/core/rtw_cmd.c:17:
/usr/src/linux-headers-5.4.0-kali2-common/include/linux/ieee80211.h:1451: note: this is the location of the previous definition
 1451 | #define IEEE80211_MAX_AMPDU_BUF  0x100
      | 
  CC [M]  /root/build/rtl8821CU/core/rtw_security.o
In file included from /root/build/rtl8821CU/include/drv_types.h:30,
                 from /root/build/rtl8821CU/core/rtw_security.c:17:
/root/build/rtl8821CU/include/wifi.h:1032: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 1032 | #define IEEE80211_MAX_AMPDU_BUF 0x40
      | 
In file included from /root/build/rtl8821CU/include/osdep_service_linux.h:83,
                 from /root/build/rtl8821CU/include/osdep_service.h:50,
                 from /root/build/rtl8821CU/include/drv_types.h:27,
                 from /root/build/rtl8821CU/core/rtw_security.c:17:
/usr/src/linux-headers-5.4.0-kali2-common/include/linux/ieee80211.h:1451: note: this is the location of the previous definition
 1451 | #define IEEE80211_MAX_AMPDU_BUF  0x100
      | 
  CC [M]  /root/build/rtl8821CU/core/rtw_debug.o
In file included from /root/build/rtl8821CU/include/drv_types.h:30,
                 from /root/build/rtl8821CU/core/rtw_debug.c:17:
/root/build/rtl8821CU/include/wifi.h:1032: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 1032 | #define IEEE80211_MAX_AMPDU_BUF 0x40
      | 
In file included from /root/build/rtl8821CU/include/osdep_service_linux.h:83,
                 from /root/build/rtl8821CU/include/osdep_service.h:50,
                 from /root/build/rtl8821CU/include/drv_types.h:27,
                 from /root/build/rtl8821CU/core/rtw_debug.c:17:
/usr/src/linux-headers-5.4.0-kali2-common/include/linux/ieee80211.h:1451: note: this is the location of the previous definition
 1451 | #define IEEE80211_MAX_AMPDU_BUF  0x100
      | 
/root/build/rtl8821CU/core/rtw_debug.c: In function ‘dump_drv_version’:
/root/build/rtl8821CU/core/rtw_debug.c:45:62: error: macro "__DATE__" might prevent reproducible builds [-Werror=date-time]
   45 |  RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
      |                                                              ^
In file included from /root/build/rtl8821CU/include/drv_types.h:60,
                 from /root/build/rtl8821CU/core/rtw_debug.c:17:
/root/build/rtl8821CU/core/rtw_debug.c:45:44: error: macro "__TIME__" might prevent reproducible builds [-Werror=date-time]
   45 |  RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
      |                                            ^~~~~~~~
/root/build/rtl8821CU/include/rtw_debug.h:114:34: note: in definition of macro ‘RTW_PRINT’
  114 |    _dbgdump(DRIVER_PREFIX fmt, ##arg);\
      |                                  ^~~
/root/build/rtl8821CU/core/rtw_debug.c:45:2: note: in expansion of macro ‘RTW_PRINT_SEL’
   45 |  RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
      |  ^~~~~~~~~~~~~
/root/build/rtl8821CU/core/rtw_debug.c:45:54: error: macro "__DATE__" might prevent reproducible builds [-Werror=date-time]
   45 |  RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
      |                                                      ^~~~~~~~
/root/build/rtl8821CU/include/rtw_debug.h:114:34: note: in definition of macro ‘RTW_PRINT’
  114 |    _dbgdump(DRIVER_PREFIX fmt, ##arg);\
      |                                  ^~~
/root/build/rtl8821CU/core/rtw_debug.c:45:2: note: in expansion of macro ‘RTW_PRINT_SEL’
   45 |  RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
      |  ^~~~~~~~~~~~~
/root/build/rtl8821CU/core/rtw_debug.c:45:44: error: macro "__TIME__" might prevent reproducible builds [-Werror=date-time]
   45 |  RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
      |                                            ^~~~~~~~
/root/build/rtl8821CU/include/rtw_debug.h:224:25: note: in definition of macro ‘RTW_PRINT_SEL’
  224 |    _seqdump(sel, fmt, ##arg) /*rtw_warn_on(1)*/; \
      |                         ^~~
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-headers-5.4.0-kali2-common/scripts/Makefile.build:271: /root/build/rtl8821CU/core/rtw_debug.o] Error 1
make[2]: *** [/usr/src/linux-headers-5.4.0-kali2-common/Makefile:1665: /root/build/rtl8821CU] Error 2
make[1]: *** [/usr/src/linux-headers-5.4.0-kali2-common/Makefile:179: sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-kali2-amd64'
make: *** [Makefile:2213: modules] Error 2

SatanicRaw, 2020-04-12:
Hey. Install `bc`, run `make RT_CFLAGS = "- Wno-error = date-time"` instead of `make`.
And yes, Kali Linux is a specialized distribution for checking network security; it is not suitable for beginners and permanent work.

jimmyarn, 2020-04-12:
Thank you, installing bc straightened everything out.

closed 2020-04-12 by jimmyarn