brektrou/rtl8821CU #51: rPi Zero

mocallins, 2020-04-13:
Should this driver work on a Pi Zero (first edition) ?

The make runs, but has a couple warnings.
I fibnally did get to comp8ile and load, after seeing the additional changes at the bottom of the README.md  :(.
But still didn't work.

Retried  again today. but now doesn't load, but it wasn't the "exec ofrmat error"

I notice a discrepncy with the ARCH, mine shows  armv6l  (that's six el, not (1, one)

mocallins, 2020-04-14:
Hmmm i think the latest compile worked correctly, had to reinstal the kernel headers.

pi: lsusb 
Bus 001 Device 005: ID 0c45:8001 Microdia
Bus 001 Device 004: ID 148f:7601 Ralink Technology, Corp. MT7601U Wireless Adapter
Bus 001 Device 007: ID **_0bda:b711_** Realtek Semiconductor Corp.
Bus 001 Device 002: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pi:> lsmod | grep 80211
mac80211              712704  1 mt7601u
cfg80211              643072  3 mac80211,mt7601u,**_8821cu_**
rfkill                 28672  3 cfg80211
pi:>

dachshund-digital, 2020-10-31:
Make sure you edit the Makefile to select the right chip set and processor type.  PiZero is only ARM v6, but I did happen to get this driver to work on ARM v7, RPi 3 B.