brektrou/rtl8821CU #52: Activation of network connection failed

ismaeelakram, 2020-04-22:
I am able to detect wifi connections and enter the password. However, whenever I try to connect, it waits for a minute or two until telling me "Activation of network connection failed."

SatanicRaw, 2020-05-01:
Try changing the distribution to Manjaro \ Ubuntu \ Arch etc. Kali Linux is a specific distribution that is not suitable for daily use.

ismaeelakram, 2020-05-03:
> Try changing the distribution to Manjaro \ Ubuntu \ Arch etc. Kali Linux is a specific distribution that is not suitable for daily use.

I need Kali Linux for its pentesting tools.

avently, 2020-05-31:
@SatanicRaw I have the same behavior on Manjaro. It's really painful to use the driver because it connects after multiple tries only. Sometimes it takes 3-5 minutes. Other adapters with different driver work well (15-20 seconds to connect after plugging in).

SatanicRaw, 2020-06-13:
Logs are needed here, without them it is not clear what the problem is. I have an adapter on the same chip and it works without problems with this driver.

AugustoNapuri, 2020-07-03:
Im having the same issue, cannot connect with the correct password in Kali linux.

multikirin, 2022-02-23:
This driver seems to be hit-or-miss depending on the distribution. On PearOS, it only works 1% of the time. Meanwhile, on Elementary OS, it always works, even though both are ubuntu-based.