brektrou/rtl8821CU #53: DKMS install error on Kali Linux

waaynee, 2020-04-23:
```
kali@kali:~$ git clone https://github.com/brektrou/rtl8821CU
Cloning into 'rtl8821CU'...
remote: Enumerating objects: 63, done.
remote: Counting objects: 100% (63/63), done.
remote: Compressing objects: 100% (62/62), done.
remote: Total 1403 (delta 23), reused 9 (delta 1), pack-reused 1340
Receiving objects: 100% (1403/1403), 5.57 MiB | 5.07 MiB/s, done.
Resolving deltas: 100% (796/796), done.
kali@kali:~$ cd rtl8821CU/
kali@kali:~/rtl8821CU$ sudo ./dkms-install.sh 
About to run dkms install steps...
Error! DKMS tree already contains: rtl8821CU-5.4.1
You cannot add the same module/version combo more than once.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' KVER=5.5.0-kali1-amd64....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.5.0-kali1-amd64 (x86_64)
Consult /var/lib/dkms/rtl8821CU/5.4.1/build/make.log for more information.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' KVER=5.5.0-kali1-amd64....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.5.0-kali1-amd64 (x86_64)
Consult /var/lib/dkms/rtl8821CU/5.4.1/build/make.log for more information.
Finished running dkms install steps.
```

sabirovrinat85, 2020-04-23:
You don't need  to do "make" as dkms-install will suffice
And "bc: not found" states clearly what is missing also...

waaynee, 2020-04-23:
> You don't need to do "make" as dkms-install will suffice
I wanted to show the outcome of the second method obviously

> And "bc: not found" states clearly what is missing also...
Unfortunately i did not know that this package is required. Note in readme would be good for timesaving

SatanicRaw, 2020-05-01:
> Unfortunately i did not know that this package is required. Note in readme would be good for timesaving

If you use Kali Linux, this should not be a problem for you, an error with bc affects only users of this distribution.

waaynee, 2020-05-02:
> If you use Kali Linux, this should not be a problem for you, an error with bc affects only users of this distribution.

Seems like it is

closed 2020-05-02 by waaynee