brektrou/rtl8821CU #56: Compile failed with ubuntu 18.04, kernel 5.3.0-51

CliffHan, 2020-05-15:
Seems that RHEL_RELEASE_CODE macro missing. Here's logs:

/home/cliff/Workspace/rtl8821CU/include/../os_dep/linux/ioctl_cfg80211.h:372:94: error: missing binary operator before token "("
 X_VERSION_CODE < KERNEL_VERSION(4, 2, 0) && RHEL_RELEASE_CODE < RHEL_RELEASE_VERSION(7,2))
                                                                                     ^
/home/cliff/Workspace/rtl8821CU/include/../os_dep/linux/ioctl_cfg80211.h:394:146: error: missing binary operator before token "("
 X_VERSION_CODE < KERNEL_VERSION(4, 7, 0) && RHEL_RELEASE_CODE < RHEL_RELEASE_VERSION(7,2))

CliffHan, 2020-05-15:
I copied followed part into ioctl_cfg80211.h and it got compiled

#ifndef RHEL_RELEASE_VERSION
#define RHEL_RELEASE_VERSION(a,b) (((a) << 8) + (b))
#endif

#ifndef RHEL_RELEASE_CODE
#define RHEL_RELEASE_CODE 0
#endif

hkbu-kennycheng, 2020-05-15:
I could confirm reverting commit 6049ce17f36bd7fd84665832cb6489e30e46ec2d would solve the problem.

breisig, 2020-05-15:
This is also not compiling with me either since the Centos/RHEL patches were applied to git.

closed 2020-05-30 by CliffHan