brektrou/rtl8821CU #60: Hardware floating point build error ARMv8

SyntrophicConsortium, 2020-05-19:
I have an ARMv8 device (Libre La Frite) I'm trying to build this on and I'm getting the hardware floating point error.

> unrecognized command line option ‘-mfloat-abi=hard’

I don't think the suggestions in the Readme are applicable here (the option that sed is supposed to remove doesn't exist in the Makefile for arm64 in /lib/modules/$(uname -r)/arch/. My understanding is that ARMv8 supports hardware floating point, but I'll admit I don't know much about this. 

I have `CONFIG_PLATFORM_ARM64_RPI = y`, but is that right? Is there some additional changes needed in the Makefile?

`uname -ir`:
> 4.19.57+ aarch64

Build error:

> make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/4.19.57+/build M=/home/andy/rtl8821CU-master  modules
> make[1]: Entering directory '/usr/src/linux-headers-4.19.57+'
>   CC [M]  /home/andy/rtl8821CU-master/core/rtw_cmd.o
> gcc: error: unrecognized command line option ‘-mfloat-abi=hard’
> scripts/Makefile.build:303: recipe for target '/home/andy/rtl8821CU-master/core/rtw_cmd.o' failed
> make[2]: *** [/home/andy/rtl8821CU-master/core/rtw_cmd.o] Error 1
> Makefile:1517: recipe for target '_module_/home/andy/rtl8821CU-master' failed
> make[1]: *** [_module_/home/andy/rtl8821CU-master] Error 2
> make[1]: Leaving directory '/usr/src/linux-headers-4.19.57+'
> Makefile:2213: recipe for target 'modules' failed
> make: *** [modules] Error 2

smp79, 2020-05-24:
Set CONFIG_MP_VHT_HW_TX_MODE = n

SyntrophicConsortium, 2020-05-27:
Still can't build it, but that fixed the `-mfloat-abi=hard` error so I will close.

closed 2020-05-27 by SyntrophicConsortium