brektrou/rtl8821CU #61: Unable to install on kali

vvkvas, 2020-05-27:
Please suggest on below:

> root@Vkali:/home/v/bu/rtl8821CU# ./dkms-install.sh 
> About to run dkms install steps...
> 
> Creating symlink /var/lib/dkms/rtl8821CU/5.4.1/source ->
>                  /usr/src/rtl8821CU-5.4.1
> 
> DKMS: add completed.
> 
> Kernel preparation unnecessary for this kernel.  Skipping...
> 
> Building module:
> cleaning build area...
> 'make' KVER=5.6.0-kali1-amd64....(bad exit status: 2)
> Error! Bad return status for module build on kernel: 5.6.0-kali1-amd64 (x86_64)
> Consult /var/lib/dkms/rtl8821CU/5.4.1/build/make.log for more information.
> 
> Kernel preparation unnecessary for this kernel.  Skipping...
> 
> Building module:
> cleaning build area...
> 'make' KVER=5.6.0-kali1-amd64....(bad exit status: 2)
> Error! Bad return status for module build on kernel: 5.6.0-kali1-amd64 (x86_64)
> Consult /var/lib/dkms/rtl8821CU/5.4.1/build/make.log for more information.
> Finished running dkms install steps.

**Also without DKMS:**

> root@Vkali:/home/v2/c/rtl8821CU# make
> /bin/sh: 1: bc: not found
> make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.6.0-kali1-amd64/build M=/home/v2/c/rtl8821CU  modules
> make[1]: Entering directory '/usr/src/linux-headers-5.6.0-kali1-amd64'
> /bin/sh: 1: bc: not found
>   CC [M]  /home/v2/c/rtl8821CU/core/rtw_cmd.o
> In file included from /home/v2/c/rtl8821CU/include/drv_types.h:30,
>                  from /home/v2/c/rtl8821CU/core/rtw_cmd.c:17:
> /home/v2/c/rtl8821CU/include/wifi.h:1032: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
>  1032 | #define IEEE80211_MAX_AMPDU_BUF 0x40
>       | 
> In file included from /home/v2/c/rtl8821CU/include/osdep_service_linux.h:83,
>                  from /home/v2/c/rtl8821CU/include/osdep_service.h:50,
>                  from /home/v2/c/rtl8821CU/include/drv_types.h:27,
>                  from /home/v2/c/rtl8821CU/core/rtw_cmd.c:17:
> /usr/src/linux-headers-5.6.0-kali1-common/include/linux/ieee80211.h:1451: note: this is the location of the previous definition
>  1451 | #define IEEE80211_MAX_AMPDU_BUF  0x100
>       | 
>   CC [M]  /home/v2/c/rtl8821CU/core/rtw_security.o
> In file included from /home/v2/c/rtl8821CU/include/drv_types.h:30,
>                  from /home/v2/c/rtl8821CU/core/rtw_security.c:17:
> /home/v2/c/rtl8821CU/include/wifi.h:1032: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
>  1032 | #define IEEE80211_MAX_AMPDU_BUF 0x40
>       | 
> In file included from /home/v2/c/rtl8821CU/include/osdep_service_linux.h:83,
>                  from /home/v2/c/rtl8821CU/include/osdep_service.h:50,
>                  from /home/v2/c/rtl8821CU/include/drv_types.h:27,
>                  from /home/v2/c/rtl8821CU/core/rtw_security.c:17:
> /usr/src/linux-headers-5.6.0-kali1-common/include/linux/ieee80211.h:1451: note: this is the location of the previous definition
>  1451 | #define IEEE80211_MAX_AMPDU_BUF  0x100
>       | 
>   CC [M]  /home/v2/c/rtl8821CU/core/rtw_debug.o
> In file included from /home/v2/c/rtl8821CU/include/drv_types.h:30,
>                  from /home/v2/c/rtl8821CU/core/rtw_debug.c:17:
> /home/v2/c/rtl8821CU/include/wifi.h:1032: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
>  1032 | #define IEEE80211_MAX_AMPDU_BUF 0x40
>       | 
> In file included from /home/v2/c/rtl8821CU/include/osdep_service_linux.h:83,
>                  from /home/v2/c/rtl8821CU/include/osdep_service.h:50,
>                  from /home/v2/c/rtl8821CU/include/drv_types.h:27,
>                  from /home/v2/c/rtl8821CU/core/rtw_debug.c:17:
> /usr/src/linux-headers-5.6.0-kali1-common/include/linux/ieee80211.h:1451: note: this is the location of the previous definition
>  1451 | #define IEEE80211_MAX_AMPDU_BUF  0x100
>       | 
> /home/v2/c/rtl8821CU/core/rtw_debug.c: In function ‘dump_drv_version’:
> /home/v2/c/rtl8821CU/core/rtw_debug.c:45:62: error: macro "__DATE__" might prevent reproducible builds [-Werror=date-time]
>    45 |  RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
>       |                                                              ^
> In file included from /home/v/c/rtl8821CU/include/drv_types.h:60,
>                  from /home/v/c/rtl8821CU/core/rtw_debug.c:17:
> /home/v/c/rtl8821CU/core/rtw_debug.c:45:44: error: macro "__TIME__" might prevent reproducible builds [-Werror=date-time]
>    45 |  RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
>       |                                            ^~~~~~~~
> /home/v/c/rtl8821CU/include/rtw_debug.h:114:34: note: in definition of macro ‘RTW_PRINT’
>   114 |    _dbgdump(DRIVER_PREFIX fmt, ##arg);\
>       |                                  ^~~
> /home/v/c/rtl8821CU/core/rtw_debug.c:45:2: note: in expansion of macro ‘RTW_PRINT_SEL’
>    45 |  RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
>       |  ^~~~~~~~~~~~~
> /home/v/c/rtl8821CU/core/rtw_debug.c:45:54: error: macro "__DATE__" might prevent reproducible builds [-Werror=date-time]
>    45 |  RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
>       |                                                      ^~~~~~~~
> /home/v/c/rtl8821CU/include/rtw_debug.h:114:34: note: in definition of macro ‘RTW_PRINT’
>   114 |    _dbgdump(DRIVER_PREFIX fmt, ##arg);\
>       |                                  ^~~
> /home/v/c/rtl8821CU/core/rtw_debug.c:45:2: note: in expansion of macro ‘RTW_PRINT_SEL’
>    45 |  RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
>       |  ^~~~~~~~~~~~~
> /home/v/c/rtl8821CU/core/rtw_debug.c:45:44: error: macro "__TIME__" might prevent reproducible builds [-Werror=date-time]
>    45 |  RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
>       |                                            ^~~~~~~~
> /home/v/c/rtl8821CU/include/rtw_debug.h:224:25: note: in definition of macro ‘RTW_PRINT_SEL’
>   224 |    _seqdump(sel, fmt, ##arg) /*rtw_warn_on(1)*/; \
>       |                         ^~~
> cc1: some warnings being treated as errors
> make[3]: *** [/usr/src/linux-headers-5.6.0-kali1-common/scripts/Makefile.build:273: /home/v/c/rtl8821CU/core/rtw_debug.o] Error 1
> make[2]: *** [/usr/src/linux-headers-5.6.0-kali1-common/Makefile:1696: /home/v/c/rtl8821CU] Error 2
> make[1]: *** [/usr/src/linux-headers-5.6.0-kali1-common/Makefile:180: sub-make] Error 2
> make[1]: Leaving directory '/usr/src/linux-headers-5.6.0-kali1-amd64'
> make: *** [Makefile:2213: modules] Error 2
> root@Vkali:/home/v/c/rtl8821CU# 
> 
>

rrafal1337, 2020-05-27:
Just install bc package:
```
/bin/sh: 1: bc: not found
````

vvkvas, 2020-05-27:
Yes, thank you it worked after:

sudo apt-get install build-essential -y
sudo apt-get install bc -y
sudo apt-get install unzip git -y

closed 2020-05-27 by vvkvas

InesCardinot, 2021-01-16:
Building for 5.8.0-38-generic
Building for architecture x86_64
Building initial module for 5.8.0-38-generic
ERROR: Cannot create report: [Errno 17] File exists: '/var/crash/rtl8188fu-dkms.0.crash'
Error! Bad return status for module build on kernel: 5.8.0-38-generic (x86_64)
Consult /var/lib/dkms/rtl8188fu/1.0.3/build/make.log for more information.
dpkg: erro ao processar o pacote rtl8188fu-dkms (--configure):
 o subprocesso instalado, do pacote rtl8188fu-dkms, o script post-installation retornou erro do status de saída 10
Erros foram encontrados durante o processamento de:
 rtl8188fu-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)

(O meu esta dando erro - ubuntu)