brektrou/rtl8821CU #64: can't open wifi after successfully installed driver

monchin, 2020-06-10:
Hello, I installed driver on ubuntu 20.04, but still can't open wifi. Below is my process:

First, I followed `README.md` and installed driver successfully, and I can make sure `8821cu.ko` file present on `/lib/modules/5.4.0-37-generic/kernel/drivers/net/wireless/realtek/rtl8821cu`.
I typed `sudo modprobe 8211cu` and returned nothing.
Besides, I saw `5.4.0-26-generic` and `5.4.0-37-generic` in `/lib/modules/`, and I don't know whether it matters.

Second, I rebooted and couldn't open wifi, so I typed `lsusb` and got:
```
Bus 001 Device 007: ID 0bda:c811 Realtek Semiconductor Corp. 
```
Then I typed `sudo usb_modeswitch -KW -v 0bda -p c811` but failed:
```
Access device 007 on bus 001
Get the current device configuration ...
Current configuration number is 1
Use interface number 0
 with class 255
Error: can't use storage command in MessageContent with interface 0; interface class is 255, expected 8. Abort
```
I don't really know how to solve it.

Third, some more information:
`dmesg`:
```
[    2.481967] usb 1-9.1: New USB device found, idVendor=0bda, idProduct=c811, bcdDevice= 2.00
[    2.481971] usb 1-9.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.481973] usb 1-9.1: Product: 802.11ac NIC
[    2.481975] usb 1-9.1: Manufacturer: Realtek
[    2.481977] usb 1-9.1: SerialNumber: 123456
```

`iwconfig`:
```
wlx30b49eeb9d26  unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=0/100  Signal level=0 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

ens1      no wireless extensions.

wlp4s0    IEEE 802.11  ESSID:off/any  
          Mode:Managed  Access Point: Not-Associated   Tx-Power=off   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:on
          
enp5s0    no wireless extensions.

enp0s31f6  no wireless extensions.

enp8s0    no wireless extensions.

lo        no wireless extensions.
```

I edited the file `/lib/udev/rules.d/40-usb_modeswitch.rules` by adding the following line:
```
ATTRS{idVendor}=="0bda", ATTRS{idProduct}=="c811", RUN+="usb_modeswitch '%b/%k'"
```

`usb_modeswitch.log`:
```
Use global config file: /etc/usb_modeswitch.conf

Raw args from udev: 1-9.1

Bus ID for device not given by udev.
 Trying to determine it from kernel name (1-9.1) ...
Use top device dir /sys/bus/usb/devices/1-9.1

USB dir exists: /sys/bus/usb/devices/1-9.1

SCSI dir exists: /sys/bus/usb/devices/1-9.1
Warning: SCSI attribute "vendor" not readable.
Warning: SCSI attribute "model" not readable.
Warning: SCSI attribute "rev" not readable.
Check class of first interface ...
 No access to interface 0. Exit
```
Any advice would be appreciated. Thank you!

SatanicRaw, 2020-06-13:
Hi, modprobe does not display any messages if the module loaded normally.
`usb_modeswitch` and `udev` rules are not needed in this case.
Judging by the logs, the module was build for the kernel 5.4.0-37, check if this particular kernel is loaded using `uname -a`
Judging by the source code of the module, your chipset is supported

SinvalVJunior, 2020-06-29:
Have the same problem on Ubuntu 20.04. It worked for a while, then a restarted my system and it stopped.

closed 2024-11-29 by monchin