brektrou/rtl8821CU #73: usb_modeswitch is returning error

kulkarniravi, 2020-08-02:
Hi,

I am on Ubuntu 20.04. I am following your instructions in README below, and here is the error I encountered:

```$ sudo usb_modeswitch -KW -v 0bda -p b812
[sudo] password for ravi: 
Take all parameters from the command line

 * usb_modeswitch: handle USB devices with multiple modes
 * Version 2.5.2 (C) Josua Dietze 2017
 * Based on libusb1/libusbx

 ! PLEASE REPORT NEW CONFIGURATIONS !

DefaultVendor=  0x0bda
DefaultProduct= 0xb812

StandardEject=1

Look for default devices ...
  found USB ID 1d6b:0003
  found USB ID 1d6b:0002
  found USB ID 1d6b:0003
  found USB ID 0b05:18f3
  found USB ID 062a:5918
  found USB ID 1d6b:0002
  found USB ID 1d6b:0003
  found USB ID 058f:6366
  found USB ID 0bda:b812
   vendor ID matched
   product ID matched
  found USB ID 1d6b:0002
 Found devices in default mode (1)
Access device 004 on bus 001
Get the current device configuration ...
Current configuration number is 1
Use interface number 0
 with class 255
Error: can't use storage command in MessageContent with interface 0; interface class is 255, expected 8. Abort
```
Any help will be much appreciated.

qiaoy99, 2020-08-30:
same problem.  I tried both on ubuntu 18.04 and 20.04.  Same error message.

yufalcon, 2020-09-04:
> Hi,
>
> I am on Ubuntu 20.04. I am following your instructions in README below, and here is the error I encountered:
>
> ```
> $ sudo usb_modeswitch -KW -v 0bda -p b812
> [sudo] password for ravi:
> Take all [...]

same bro

AlanMauricioC, 2020-09-07:
Same here, any solution?
```
Error: can't use storage command in MessageContent with interface 0; interface class is 255, expected 8. Abort
```
I can use it but usually with troubles

S.O: Manjaro
chipset: RTL8811CU

RussH, 2020-10-21:
Similar - usb mdoeswitch didn't work, i went ahead and built the driver, and installed it - wifi sprang into life

kulkravi, 2020-10-22:
I cloned, rebuilt the driver and installed. Still didn't work.

Prikalel, 2020-11-01:
Same problemo

wdvictor, 2020-11-20:
Hi there, people. I was experiencing the same problem (the exact same problem), and I was capable to solve following these instructions: 
https://askubuntu.com/questions/1162974/wireless-usb-adapter-0bdac811-realtek-semiconductor-corp

follows:

sudo apt update
sudo apt install build-essential git dkms
git clone https://github.com/brektrou/rtl8821CU.git
cd rtl8821CU
chmod +x dkms-install.sh
sudo ./dkms-install.sh

sudo modprobe 8821cu

I expect to help because it's cost my entire night, and this solves my problem.

serginhosis, 2020-12-25:
@wdvictor
I was used the same instructions to install in my machine, but the adapter is not recognized

careliinfinitii, 2021-02-24:
Still same issue on Pi4 B+ with Pi Os

Take all parameters from the command line

 * usb_modeswitch: handle USB devices with multiple modes
 * Version 2.5.2 (C) Josua Dietze 2017
 * Based on libusb1/libusbx

 ! PLEASE REPORT NEW CONFIGURATIONS !

DefaultVendor=  0x0bda
DefaultProduct= 0xf179

StandardEject=1

Look for default devices ...
  found USB ID 1d6b:0003
  found USB ID 0bda:f179
   vendor ID matched
   product ID matched
  found USB ID 1a86:e5e3
  found USB ID 2109:3431
  found USB ID 1d6b:0002
 Found devices in default mode (1)
Access device 004 on bus 001
Get the current device configuration ...
Current configuration number is 1
Use interface number 0
 with class 255
Error: can't use storage command in MessageContent with interface 0; interface class is 255, expected 8. Abort

ijakparov, 2021-07-30:
That's work for me https://github.com/morrownr/8821cu

niauah, 2022-04-11:
> That's work for me https://github.com/morrownr/8821cu

@ijakparov This worked for me, thanks!

Sub-read, 2022-05-16:
@wdvictor that worked for me, thanks a lot.

paulomcnally, 2022-11-20:
> sudo ./dkms-install.sh

```
paulomcnally@raspberrypi:~/rtl8821CU $ sudo ./dkms-install.sh
About to run dkms install steps...

Creating symlink /var/lib/dkms/rtl8821CU/5.4.1/source ->
                 /usr/src/rtl8821CU-5.4.1

DKMS: add completed.
Error! echo
Your kernel headers for kernel 5.15.61+ cannot be found at
/lib/modules/5.15.61+/build or /lib/modules/5.15.61+/source.
You can use the --kernelsourcedir option to tell DKMS where it's located.
Error! echo
Your kernel headers for kernel 5.15.61+ cannot be found at
/lib/modules/5.15.61+/build or /lib/modules/5.15.61+/source.
You can use the --kernelsourcedir option to tell DKMS where it's located.
Finished running dkms install steps.
```

potassium-chloride, 2024-12-07:
I found this issue via search and I have same problem. I've checked dmesg and found strange logs:
```
[1263959.000170] usb 1-7: new high-speed USB device number 56 using xhci_hcd
[1263959.148584] usb 1-7: New USB device found, idVendor=0bda, idProduct=1a2b, bcdDevice= 2.00
[1263959.148587] usb 1-7: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[1263959.148588] usb 1-7: Product: DISK
[1263959.148590] usb 1-7: Manufacturer: Realtek
[1263959.149515] usb-storage 1-7:1.0: USB Mass Storage device detected
[1263959.149754] scsi host3: usb-storage 1-7:1.0
[1263960.154565] scsi 3:0:0:0: CD-ROM            Realtek  Driver Storage   1.00 PQ: 0 ANSI: 0 CCS
[1263960.159691] sr 3:0:0:0: [sr0] scsi3-mmc drive: 0x/0x caddy
[1263960.208044] sr 3:0:0:0: Attached scsi CD-ROM sr0
[1263960.208194] sr 3:0:0:0: Attached scsi generic sg2 type 5
[1263960.246323] usb 1-7: USB disconnect, device number 56
[1263960.632120] usb 1-7: new high-speed USB device number 57 using xhci_hcd
[1263960.780318] usb 1-7: New USB device found, idVendor=0bda, idProduct=c811, bcdDevice= 2.00
[1263960.780321] usb 1-7: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[1263960.780323] usb 1-7: Product: 802.11ac NIC
[1263960.780324] usb 1-7: Manufacturer: Realtek
[1263960.780325] usb 1-7: SerialNumber: 123456
[1263960.782775] option 1-7:1.0: GSM modem (1-port) converter detected
[1263960.782871] usb 1-7: GSM modem (1-port) converter now attached to ttyUSB0
```
Probably, my USB device try to do some smart things (like CD-ROM with Windows driver) but this behavior breaks Linux driver