brektrou/rtl8821CU #74: Fails to compile on linux 5.8

ws-gregm, 2020-08-04:
Compiles fine on linux 5.6.x, however on the newly released linux 5.8

```
gmaglio@oxcart:~/build/rtl8821CU$ make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.8.0-050800-lowlatency/build M=/home/gmaglio/build/rtl8821CU  modules
make[1]: Entering directory '/usr/src/linux-headers-5.8.0-050800-lowlatency'
  CC [M]  /home/gmaglio/build/rtl8821CU/core/rtw_cmd.o
In file included from /home/gmaglio/build/rtl8821CU/include/drv_types.h:30:0,
                 from /home/gmaglio/build/rtl8821CU/core/rtw_cmd.c:17:
/home/gmaglio/build/rtl8821CU/include/wifi.h:1032:0: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 #define IEEE80211_MAX_AMPDU_BUF 0x40
 
In file included from /home/gmaglio/build/rtl8821CU/include/osdep_service_linux.h:83:0,
                 from /home/gmaglio/build/rtl8821CU/include/osdep_service.h:50,
                 from /home/gmaglio/build/rtl8821CU/include/drv_types.h:27,
                 from /home/gmaglio/build/rtl8821CU/core/rtw_cmd.c:17:
./include/linux/ieee80211.h:1600:0: note: this is the location of the previous definition
 #define IEEE80211_MAX_AMPDU_BUF  0x100
 
In file included from /home/gmaglio/build/rtl8821CU/include/drv_types.h:78:0,
                 from /home/gmaglio/build/rtl8821CU/core/rtw_cmd.c:17:
/home/gmaglio/build/rtl8821CU/include/rtw_security.h:252:8: error: redefinition of ‘struct sha256_state’
 struct sha256_state {
        ^~~~~~~~~~~~
In file included from ./include/linux/filter.h:23:0,
                 from ./include/net/sock.h:59,
                 from ./include/linux/tcp.h:19,
                 from ./include/linux/ipv6.h:88,
                 from ./include/net/addrconf.h:50,
                 from /home/gmaglio/build/rtl8821CU/include/osdep_service_linux.h:48,
                 from /home/gmaglio/build/rtl8821CU/include/osdep_service.h:50,
                 from /home/gmaglio/build/rtl8821CU/include/drv_types.h:27,
                 from /home/gmaglio/build/rtl8821CU/core/rtw_cmd.c:17:
./include/crypto/sha.h:84:8: note: originally defined here
 struct sha256_state {
        ^~~~~~~~~~~~
scripts/Makefile.build:280: recipe for target '/home/gmaglio/build/rtl8821CU/core/rtw_cmd.o' failed
make[2]: *** [/home/gmaglio/build/rtl8821CU/core/rtw_cmd.o] Error 1
Makefile:1756: recipe for target '/home/gmaglio/build/rtl8821CU' failed
make[1]: *** [/home/gmaglio/build/rtl8821CU] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.8.0-050800-lowlatency'
Makefile:2213: recipe for target 'modules' failed
make: *** [modules] Error 2
gmaglio@oxcart:~/build/rtl8821CU$ uname -a
Linux oxcart 5.8.0-050800-lowlatency #202008022230 SMP PREEMPT Sun Aug 2 22:36:20 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

IlgazC, 2020-08-07:
Same here on Manjaro Linux - Testing. Now, 5.8.x is suggested too (at least on this distro)

prathmesh01, 2020-08-14:
Got error on EndeavourOS on kernel 5.8.1-arch1-1

DKMS make.log for rtl8821cu-5.4.18193b04bf for kernel 5.8.1-arch1-1 (x86_64)
Friday 14 August 2020 10:11:01 AM IST
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.8.1-arch1-1/build M=/var/lib/dkms/rtl8821cu/5.4.18193b04bf/build  modules
make[1]: Entering directory '/usr/lib/modules/5.8.1-arch1-1/build'
  CC [M]  /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/core/rtw_io.o
In file included from /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/include/drv_types.h:30,
                 from /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/core/rtw_security.c:17:
/var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/include/wifi.h:1032: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 1032 | #define IEEE80211_MAX_AMPDU_BUF 0x40
      | 
In file included from /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/include/drv_types.h:30,
                 from /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/core/rtw_io.c:47:
/var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/include/wifi.h:1032: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 1032 | #define IEEE80211_MAX_AMPDU_BUF 0x40
      | 
In file included from /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/include/drv_types.h:30,
                 from /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/core/rtw_debug.c:17:
/var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/include/wifi.h:1032: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 1032 | #define IEEE80211_MAX_AMPDU_BUF 0x40
      | 
In file included from /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/include/osdep_service_linux.h:83,
                 from /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/include/osdep_service.h:50,
                 from /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/include/drv_types.h:27,
                 from /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/core/rtw_security.c:17:
./include/linux/ieee80211.h:1600: note: this is the location of the previous definition
 1600 | #define IEEE80211_MAX_AMPDU_BUF  0x100
      | 
In file included from /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/include/osdep_service_linux.h:83,
                 from /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/include/osdep_service.h:50,
                 from /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/include/drv_types.h:27,
                 from /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/core/rtw_io.c:47:
./include/linux/ieee80211.h:1600: note: this is the location of the previous definition
 1600 | #define IEEE80211_MAX_AMPDU_BUF  0x100
      | 
In file included from /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/include/osdep_service_linux.h:83,
                 from /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/include/osdep_service.h:50,
                 from /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/include/drv_types.h:27,
                 from /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/core/rtw_debug.c:17:
./include/linux/ieee80211.h:1600: note: this is the location of the previous definition
 1600 | #define IEEE80211_MAX_AMPDU_BUF  0x100
      | 
In file included from /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/include/drv_types.h:30,
                 from /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/core/rtw_cmd.c:17:
/var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/include/wifi.h:1032: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 1032 | #define IEEE80211_MAX_AMPDU_BUF 0x40
      | 
In file included from /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/include/osdep_service_linux.h:83,
                 from /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/include/osdep_service.h:50,
                 from /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/include/drv_types.h:27,
                 from /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/core/rtw_cmd.c:17:
./include/linux/ieee80211.h:1600: note: this is the location of the previous definition
 1600 | #define IEEE80211_MAX_AMPDU_BUF  0x100
      | 
In file included from /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/include/drv_types.h:78,
                 from /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/core/rtw_security.c:17:
/var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/include/rtw_security.h:252:8: error: redefinition of ‘struct sha256_state’
  252 | struct sha256_state {
      |        ^~~~~~~~~~~~
In file included from /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/include/drv_types.h:78,
                 from /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/core/rtw_io.c:47:
/var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/include/rtw_security.h:252:8: error: redefinition of ‘struct sha256_state’
  252 | struct sha256_state {
      |        ^~~~~~~~~~~~
In file included from /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/include/drv_types.h:78,
                 from /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/core/rtw_debug.c:17:
[... 39 lines cut ...]
In file included from /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/include/drv_types.h:78,
                 from /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/core/rtw_cmd.c:17:
/var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/include/rtw_security.h:252:8: error: redefinition of ‘struct sha256_state’
  252 | struct sha256_state {
      |        ^~~~~~~~~~~~
In file included from ./include/linux/filter.h:23,
                 from ./include/net/sock.h:59,
[... cut ...]
make[2]: *** [scripts/Makefile.build:281: /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/core/rtw_io.o] Error 1
make[2]: *** [scripts/Makefile.build:281: /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/core/rtw_cmd.o] Error 1
make[2]: *** [scripts/Makefile.build:281: /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build/core/rtw_debug.o] Error 1
make[1]: *** [Makefile:1756: /var/lib/dkms/rtl8821cu/5.4.18193b04bf/build] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.8.1-arch1-1/build'
make: *** [Makefile:2213: modules] Error 2

IlgazC, 2020-08-14:
I started to pan to move back to 5.4 LTS series because of this. It is Realtek's fault nobody else.

mbrysa, 2020-08-15:
After you fix the struct name conflict, there's also a problem with frame registration, because the API changed (https://github.com/torvalds/linux/commit/6cd536fe62ef58d7c4eac2da07ab0ed7fd19010d). AFAICT `cfg80211_rtw_mgmt_frame_register()` does nothing (https://github.com/mbrysa/rtl8821CU/blob/1697653e7c1aedf35ac1ef53e401afca11ebf7ac/os_dep/linux/ioctl_cfg80211.c#L7173), so I just disabled it.

This compiles and seems to work (I'm writing this while using it on Linux 5.8.1 :-) ): https://github.com/mbrysa/rtl8821CU/commit/1697653e7c1aedf35ac1ef53e401afca11ebf7ac

IlgazC, 2020-08-15:
It works fine on Manjaro 5.8.1 too. Thank you!

prathmesh01, 2020-08-17:
Thank you @mbrysa, It works fine with Endeavour OS 5.8.1-arch1-1!

athoik, 2020-08-18:
There is a patch available for kernel 5.8 here: https://github.com/oe-alliance/oe-alliance-core/blob/088f76a09b6ba189ede5947e2601527e8b313da7/meta-oe/recipes-connectivity/realtek/rt8821cu/add-5.8-support.patch

It seems to handle properly the changes required for frame registration!

PR #76 created with above patch. Hope it helps.

yufalcon, 2020-09-07:
> There is a patch available for kernel 5.8 here: [...]

thx u bro but my rtl8821cu always dc and dead why this happen?

closed 2020-12-18 by brektrou