brektrou/rtl8821CU #79: Support for raw packet injection on RTL8821CU

devbharat, 2020-08-28:
Does the driver+firmware+module support packet injection? I can get the monitor mode to work, but I can't seem to inject packets.

kimocoder, 2020-12-27:
I may add support if @brektrou want's ut 🥇

tester009, 2021-01-30:
> I may add support if @brektrou want's ut 1st_place_medal

Packet injection is urgently needed in RTL8821CU driver.
This chip is widely used and still can't be used in security testing, etc.
Wish, packet injection feature will be added soon!

magwangkh, 2021-02-19:
Great job on the drivers..Am running kali3 5.4.0.. Managed to install drivers now is working. i have a comfast cf759bf rtl8821cu... Havent tried connecting to wifi yet but monitor mode is working using iw dev wlan0 set monitor none.... But the thing is its not showing wlan0mon just wlan0 but in iwconfig it states mode is monitor... Now my question is whats the code to disable monitor mode and is there anyway to packet inject using rtl8821cu. Thanks waiting for anyones reply

magwangkh, 2021-02-19:
Sorry guys just ignore the diable monitor mode question i figured found iw wlan0 set type managed works fine... I realized my dongle read by iwconfig as unassociated... Anyone can help

hankie148, 2021-03-22:
> Does the driver+firmware+module support packet injection? I can get the monitor mode to work, but I can't seem to inject packets.

Yeah I also came across this problem. It is a shame because reaver and bully and many more applications will not work properly without packet injection. I'll hope they bring out a new update soon!

henriquedesousa, 2021-04-20:
I've managed to put my 8821cu card in monitor mode, and it also injects packets, however on mdk3 it only does around 15 packets/second, way less than my onboard card that does more than 1000 packets. 

Any idea of a configuration flag to help increase the rate?