brektrou/rtl8821CU #82: Continuous connection/disconnection after update to 5.4.0-47-generic #51-Ubuntu

eloraiby, 2020-09-09:
Continuous disconnections with the following log after kernel update to  5.4.0-47-generic #51-Ubuntu:
```
[ 2284.812262] RTW: ============ STA [XXXXXXXXX]  ===================
[ 2284.812264] RTW: mac_id : 0
[ 2284.812264] RTW: wireless_mode : 0x0b
[ 2284.812264] RTW: mimo_type : 0
[ 2284.812265] RTW: bw_mode : 20MHz, ra_bw_mode : 20MHz
[ 2284.812266] RTW: rate_id : 3
[ 2284.812267] RTW: rssi : -1 (%), rssi_level : 0
[ 2284.812267] RTW: is_support_sgi : Y, is_vht_enable : N
[ 2284.812268] RTW: disable_ra : N, disable_pt : N
[ 2284.812268] RTW: is_noisy : N
[ 2284.812268] RTW: txrx_state : 0
[ 2284.812269] RTW: curr_tx_rate : CCK_1M (L)
[ 2284.812269] RTW: curr_tx_bw : 20MHz
[ 2284.812270] RTW: curr_retry_ratio : 0
[ 2284.812270] RTW: ra_mask : 0x00000000000fffff

[ 2284.823635] RTW: recv eapol packet
[ 2284.823869] RTW: send eapol packet
[ 2284.840394] RTW: recv eapol packet
[ 2284.840743] RTW: send eapol packet
[ 2284.840801] RTW: set pairwise key camid:4, addr:XXXXXXXXX, kid:0, type:AES
[ 2284.840917] IPv6: ADDRCONF(NETDEV_CHANGE): wlx1XXXXXXXXX: link becomes ready
[ 2284.842683] RTW: set group key camid:5, addr:XXXXXXXXX, kid:2, type:AES
[ 2665.696556] usb 3-2.4.1: USB disconnect, device number 8
[ 2665.707014] RTW: rtw_ndev_uninit(wlxXXXXXXXXX) if1
[ 2665.830832] RTW: rtw_dev_unload: driver not in IPS
[ 2665.830839] RTW: ERROR _halmac_reg_write_8: I/O FAIL!
[ 2665.830840] RTW: ERROR _halmac_reg_write_8: I/O FAIL!
[ 2665.830927] RTW: WARN free_recv_skb_queue not empty, 6
[ 2666.078785] usb 3-2.4.1: new high-speed USB device number 9 using xhci_hcd
[ 2666.179202] usb 3-2.4.1: New USB device found, idVendor=0bda, idProduct=c811, bcdDevice= 2.00
[ 2666.179205] usb 3-2.4.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2666.179206] usb 3-2.4.1: Product: 802.11ac NIC
[ 2666.179207] usb 3-2.4.1: Manufacturer: Realtek
[ 2666.179208] usb 3-2.4.1: SerialNumber: 123456
[ 2666.239059] RTW: [HALMAC]11692M
               HALMAC_MAJOR_VER = 1
               HALMAC_PROTOTYPE_VER = 4
               HALMAC_MINOR_VER = 19
               HALMAC_PATCH_VER = 3
[ 2666.239540] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[ 2666.322181] RTW: HW EFUSE
[ 2666.322182] RTW: 0x000: 29 81 00 BC  09 00 21 00  6E 04 A4 10  10 00 30 0B  
[ 2666.322187] RTW: 0x010: FF FF FF FF  FF FF FF FF  FF FF FF 02  FF FF FF FF  
[ 2666.322191] RTW: 0x020: FF FF 28 28  28 28 28 28  28 28 28 28  28 28 28 28  
[ 2666.322195] RTW: 0x030: 02 FF FF FF  FF FF EC FF  FF FF 2B 2B  2B 2B 2B 2B  
[ 2666.322198] RTW: 0x040: 30 30 30 30  30 33 FF FF  FF FF FF FF  FF FF FF FF  
[ 2666.322202] RTW: 0x050: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2666.322206] RTW: 0x060: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2666.322209] RTW: 0x070: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2666.322213] RTW: 0x080: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2666.322216] RTW: 0x090: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2666.322220] RTW: 0x0A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2666.322224] RTW: 0x0B0: FF FF FF FF  FF FF FF FF  7F 06 20 00  FF FF FF FF  
[ 2666.322227] RTW: 0x0C0: FF 01 00 41  00 00 00 00  00 FF 22 FF  FF FF FF FF  
[ 2666.322231] RTW: 0x0D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2666.322235] RTW: 0x0E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2666.322238] RTW: 0x0F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2666.322242] RTW: 0x100: DA 0B 11 C8  E3 46 02 XXXXXXXXX 09 03 52  
[ 2666.322245] RTW: 0x110: 65 61 6C 74  65 6B 0E 03  38 30 32 2E  31 31 61 63  
[ 2666.322249] RTW: 0x120: 20 4E 49 43  08 03 31 32  33 34 35 36  FF FF FF FF  
[ 2666.322253] RTW: 0x130: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2666.322256] RTW: 0x140: FF FF FF FF  00 00 31 0F  FF FF FF FF  FF FF FF FF  
[ 2666.322260] RTW: 0x150: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2666.322264] RTW: 0x160: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2666.322267] RTW: 0x170: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2666.322271] RTW: 0x180: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2666.322274] RTW: 0x190: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2666.322278] RTW: 0x1A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2666.322281] RTW: 0x1B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2666.322285] RTW: 0x1C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2666.322289] RTW: 0x1D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2666.322292] RTW: 0x1E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2666.322296] RTW: 0x1F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2666.322372] RTW: hal_com_config_channel_plan chplan:0x7F
[ 2666.327650] RTW: [HALMAC][ALWAYS]shall R reg twice!!
[ 2666.328814] RTW: WARN [HALMAC][WARN]H2C/C2H ver is compatible!!
[ 2666.376052] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[ 2666.376131] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[ 2666.376211] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[ 2666.376291] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[ 2666.376371] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[ 2666.376454] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[ 2666.376534] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[ 2666.376860] RTW: rtw_regsty_chk_target_tx_power_valid return _FALSE for band:0, path:0, rs:0, t:-1
[ 2666.377520] RTW: rtw_ndev_init(wlan0) if1 mac_addr=XXXXXXXXX
[ 2666.419726] rtl8821cu 3-2.4.1:1.0 wlx1XXXXXXXXX: renamed from wlan0
[ 2666.463847] RTW: [HALMAC][ALWAYS]shall R reg twice!!
[ 2666.464980] RTW: WARN [HALMAC][WARN]H2C/C2H ver is compatible!!
[ 2667.119466] RTW: wlx1XXXXXXXXX- hw port(0) mac_addr =1c:bf:ce:87:a8:cc
[ 2671.116300] RTW: rtw_set_802_11_connect(wlx1XXXXXXXXX)  fw_state=0x00000008
[ 2671.459439] RTW: start auth
[ 2671.463686] RTW: auth success, start assoc
[ 2671.467221] RTW: assoc success
[ 2671.468962] RTW: ============ STA [XXXXXXXXX]  ===================
[ 2671.468964] RTW: mac_id : 0
[ 2671.468965] RTW: wireless_mode : 0x0b
[ 2671.468965] RTW: mimo_type : 0
[ 2671.468967] RTW: bw_mode : 20MHz, ra_bw_mode : 20MHz
[ 2671.468968] RTW: rate_id : 3
[ 2671.468969] RTW: rssi : -1 (%), rssi_level : 0
[ 2671.468969] RTW: is_support_sgi : Y, is_vht_enable : N
[ 2671.468970] RTW: disable_ra : N, disable_pt : N
[ 2671.468970] RTW: is_noisy : N
[ 2671.468971] RTW: txrx_state : 0
[ 2671.468972] RTW: curr_tx_rate : CCK_1M (L)
[ 2671.468973] RTW: curr_tx_bw : 20MHz
[ 2671.468973] RTW: curr_retry_ratio : 0
[ 2671.468974] RTW: ra_mask : 0x00000000000fffff

[ 2671.480192] RTW: recv eapol packet
[ 2671.480441] RTW: send eapol packet
[ 2671.497065] RTW: recv eapol packet
[ 2671.497159] RTW: send eapol packet
[ 2671.497239] RTW: set pairwise key camid:4, addr:XXXXXXXXX, kid:0, type:AES
[ 2671.497343] IPv6: ADDRCONF(NETDEV_CHANGE): wlx1XXXXXXXXX: link becomes ready
[ 2671.498804] RTW: set group key camid:5, addr:XXXXXXXXX, kid:2, type:AES
```
Any idea ?

yufalcon, 2020-09-09:
Same bro i use parrot sec os

lexxwork, 2020-12-01:
caught the same
===================================
Dec  1 14:31:49 lxxn kernel: [19486.345191] RTW: send eapol packet
>>>>>>>>>>>>>Dec  1 14:41:42 lxxn kernel: [20078.921531] usb 3-2: USB disconnect, device number 11<<<<<<<<<
Dec  1 14:41:42 lxxn kernel: [20078.960766] RTW: rtw_ndev_uninit(wlx002e2d9d80d9) if1
Dec  1 14:41:42 lxxn kernel: [20079.060647] RTW: rtw_dev_unload: driver not in IPS
Dec  1 14:41:42 lxxn kernel: [20079.060655] RTW: ERROR _halmac_reg_write_8: I/O FAIL!
Dec  1 14:41:42 lxxn kernel: [20079.060656] RTW: ERROR _halmac_reg_write_8: I/O FAIL!
Dec  1 14:41:42 lxxn kernel: [20079.060755] RTW: WARN free_recv_skb_queue not empty, 10
Dec  1 14:41:42 lxxn kernel: [20079.336694] usb 3-2: new high-speed USB device number 12 using xhci_hcd
Dec  1 14:41:42 lxxn kernel: [20079.485311] usb 3-2: New USB device found, idVendor=0bda, idProduct=c811, bcdDevice= 2.00
Dec  1 14:41:42 lxxn kernel: [20079.485317] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Dec  1 14:41:42 lxxn kernel: [20079.485320] usb 3-2: Product: 802.11ac NIC
Dec  1 14:41:42 lxxn kernel: [20079.485323] usb 3-2: Manufacturer: Realtek
Dec  1 14:41:42 lxxn kernel: [20079.485325] usb 3-2: SerialNumber: 123456
Dec  1 14:41:42 lxxn kernel: [20079.488594] RTW: [HALMAC]11692M
Dec  1 14:41:42 lxxn kernel: [20079.488594] HALMAC_MAJOR_VER = 1
Dec  1 14:41:42 lxxn kernel: [20079.488594] HALMAC_PROTOTYPE_VER = 4
Dec  1 14:41:42 lxxn kernel: [20079.488594] HALMAC_MINOR_VER = 19
Dec  1 14:41:42 lxxn kernel: [20079.488594] HALMAC_PATCH_VER = 3
Dec  1 14:41:42 lxxn kernel: [20079.489219] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
Dec  1 14:41:43 lxxn kernel: [20079.571527] RTW: HW EFUSE
Dec  1 14:41:43 lxxn kernel: [20079.571531] RTW: 0x000: 29 81 00 BC  09 00 21 00....
===================================
- linux version: 5.4.0-54-generic
- device version: RTL8811

- /etc/modprobe.d/8821cu.conf
options 8821cu rtw_wireless_mode=84 #only 5ghz

- may happen  1-3 times a day 

- The dongle was recently bought. Is it broken?