brektrou/rtl8821CU #83: EDUP EP-AC1661 error: "Empty EFUSE with unknown REF type" - which REF type to use for BT coexistence?

airdrummingfool, 2020-09-11:
Hello,
I'm creating this issue in case others run into this problem, but also I'm curious if anyone has an explanation/suggestion on how to proceed.

I have successfully compiled the driver (versions 5.4.1 and 5.8.1) on an RPi 3 (Buster 5.4.51). When I connect an EDUP EP-AC1661 there is an error shown in `dmesg`, and it does not work.

`dmesg` output:
```
[40760.025616] usb 1-1.4: new high-speed USB device number 5 using dwc_otg
[40760.156657] usb 1-1.4: New USB device found, idVendor=0bda, idProduct=c820, bcdDevice= 2.00
[40760.156673] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[40760.156683] usb 1-1.4: Product: 802.11ac NIC
[40760.156692] usb 1-1.4: Manufacturer: Realtek
[40760.156702] usb 1-1.4: SerialNumber: FF
[40760.158816] RTW: [HALMAC]11692M
               HALMAC_MAJOR_VER = 1
               HALMAC_PROTOTYPE_VER = 6
               HALMAC_MINOR_VER = 5
               HALMAC_PATCH_VER = 5
[40760.160609] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[40760.391042] RTW: HW EFUSE
[40760.391053] RTW: 0x000: 29 81 00 BC  09 00 21 00  6E 04 A4 34  10 00 30 0B
[40760.391099] RTW: 0x010: 1E 1F 20 20  20 20 26 27  28 28 28 FF  FF FF FF FF
[40760.391141] RTW: 0x020: FF FF 29 2A  2A 2A 2C 2C  2C 2C 2C 2C  2A 2B 2B 2B
[40760.391183] RTW: 0x030: 00 FF FF FF  FF FF FF FF  FF FF 1E 1F  20 20 20 20
[40760.391227] RTW: 0x040: 26 27 28 28  28 02 FF FF  FF FF FF FF  FF FF FF FF
[40760.391269] RTW: 0x050: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[40760.391311] RTW: 0x060: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[40760.391354] RTW: 0x070: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[40760.391396] RTW: 0x080: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[40760.391438] RTW: 0x090: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[40760.391481] RTW: 0x0A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[40760.391523] RTW: 0x0B0: FF FF FF FF  FF FF FF FF  7F 04 20 00  FF FF FF FF
[40760.391565] RTW: 0x0C0: FF 21 00 00  00 00 00 00  00 26 FF FF  FF FF FF FF
[40760.391607] RTW: 0x0D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[40760.391650] RTW: 0x0E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[40760.391692] RTW: 0x0F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[40760.391734] RTW: 0x100: DA 0B 20 C8  E3 46 FF E8  4E 06 86 BC  45 09 03 52
[40760.391776] RTW: 0x110: 65 61 6C 74  65 6B 0E 03  38 30 32 2E  31 31 61 63
[40760.391818] RTW: 0x120: 20 4E 49 43  04 03 46 46  FF FF FF FF  FF FF FF FF
[40760.391860] RTW: 0x130: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[40760.391901] RTW: 0x140: FF FF FF FF  00 00 31 0F  FF FF FF FF  FF FF FF FF
[40760.391944] RTW: 0x150: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[40760.391986] RTW: 0x160: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[40760.392028] RTW: 0x170: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[40760.392070] RTW: 0x180: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[40760.392112] RTW: 0x190: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[40760.392154] RTW: 0x1A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[40760.392196] RTW: 0x1B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[40760.392238] RTW: 0x1C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[40760.392280] RTW: 0x1D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[40760.392322] RTW: 0x1E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[40760.392364] RTW: 0x1F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[40760.392583] RTW: hal_com_config_channel_plan chplan:0x7F
[40760.392587] RTW: ERROR

               Empty EFUSE with unknown REF type!!

[40760.392590] RTW: ERROR please program efuse or specify correct RFE type.
[40760.392593] RTW: ERROR cmd: insmod rtl8821cx.ko rtw_RFE_type=<rfe_type>
```

The error appears to come from this line: 
https://github.com/brektrou/rtl8821CU/blob/45a8b4393e3281b969822c81bd93bdb731d58472/hal/rtl8821c/rtl8821c_ops.c#L410

The lines right after the error suggests manually providing a value for `rtw_RFE_type` when loading the module. Based on `rtl8821c_ops.c` there appear to be 8 options:
```
		Type 0 - (2-Ant, DPDT), (2G_WLG, iPA, iLNA, iSW), (5G, iPA, iLNA, iSW)
		Type 1 - (1-Ant, SPDT@Ant1), (2G_WLG, iPA, iLNA, iSW), (5G, iPA, iLNA, iSW)
		Type 2 -(1-Ant, SPDT@Ant1) , (2G_BTG, iPA, iLNA, iSW), (5G, iPA, iLNA, iSW)
		Type 3 - (1-Ant, DPDT@Ant2), (2G_WLG, iPA, iLNA, iSW), (5G, iPA, iLNA, iSW)
		Type 4 - (1-Ant, DPDT@Ant2), (2G_BTG, iPA, iLNA, iSW), (5G, iPA, iLNA, iSW)
		Type 5 - (2-Ant), (2G_WLG, iPA, iLNA, iSW), (5G, iPA, iLNA, iSW)
		Type 6 - (2-Ant), (2G_WLG, iPA, iLNA, iSW), (5G, iPA, iLNA, iSW)
		Type 7 - (1-Ant), (2G_BTG, iPA, iLNA, iSW), (5G, iPA, iLNA, iSW)
```

If I unload the module (`sudo rmmod 8821cu.ko`) and then re-load it specifying any one of the above types, it seems to work fine:
`sudo insmod /lib/modules/$(uname -r)/kernel/drivers/net/wireless/realtek/rtl8821cu/8821cu.ko rtw_RFE_type=7`

The wifi works just as well if I put `rtw_RFE_type=0` or `rtw_RFE_type=1`, etc.

Digging a bit deeper, it appears that the RFE Type is used to determine antenna isolation and the BT coexistence mechanism. Is there a way to know which is correct for my wifi dongle, or which is optimal in any given situation?

airdrummingfool, 2020-09-12:
Update: I bought a second EDUP EP-AC1661 just to see what would happen. Turns out that the EFUSE is a bit different in this one, and a valid `rfe_type` can be read (it's located at 0xCA in the EFUSE, per `EEPROM_RFE_OPTION_8821C`).

Full EFUSE of a valid chip:
```
RTW: HW EFUSE
RTW: 0x000: 29 81 00 BC  09 00 21 00  6E 04 A4 34  10 00 30 0B
RTW: 0x010: 1D 1D 1E 1F  1F 1F 25 25  26 27 27 FF  FF FF FF FF
RTW: 0x020: FF FF 28 28  28 28 2A 29  29 28 28 27  24 26 26 26
RTW: 0x030: 00 FF FF FF  FF FF FF FF  FF FF 1D 1D  1E 1F 1F 1F
RTW: 0x040: 25 25 26 27  27 02 FF FF  FF FF FF FF  FF FF FF FF
RTW: 0x050: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
RTW: 0x060: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
RTW: 0x070: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
RTW: 0x080: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
RTW: 0x090: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
RTW: 0x0A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
RTW: 0x0B0: FF FF FF FF  FF FF FF FF  7F 0A 1F 00  FF FF FF FF
RTW: 0x0C0: FF 21 00 00  00 00 00 00  00 FF 26 FF  FF FF FF FF
RTW: 0x0D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
RTW: 0x0E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
RTW: 0x0F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
RTW: 0x100: DA 0B 20 C8  E3 46 FF E8  4E 06 86 AA  8F 09 03 52
RTW: 0x110: 65 61 6C 74  65 6B 0E 03  38 30 32 2E  31 31 61 63
RTW: 0x120: 20 4E 49 43  04 03 46 46  FF FF FF FF  FF FF FF FF
RTW: 0x130: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
RTW: 0x140: FF FF FF FF  00 00 31 0F  FF FF FF FF  FF FF FF FF
RTW: 0x150: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
RTW: 0x160: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
RTW: 0x170: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
RTW: 0x180: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
RTW: 0x190: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
RTW: 0x1A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
RTW: 0x1B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
RTW: 0x1C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
RTW: 0x1D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
RTW: 0x1E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
RTW: 0x1F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
```

thus `rfe_type`: `0x26` (`38` decimal).

Some other random debug output:
```
RTW: EEPROM rfe_type=0x26
RTW: EEPROM PAType_2G is 0x0, ExternalPA_2G = 0
RTW: EEPROM PAType_5G is 0x0, external_pa_5g = 0
RTW: EEPROM LNAType_2G is 0x0, ExternalLNA_2G = 0
RTW: EEPROM LNAType_5G is 0x0, external_lna_5g = 0
RTW: EEPROM TypeGPA = 0x0
RTW: EEPROM TypeAPA = 0x0
RTW: EEPROM TypeGLNA = 0x0
RTW: EEPROM TypeALNA = 0x0
```

[speculation]
A bit more digging and it appears that this value is assigned to `rfe_type_expand` in `phydm.c`, and then referenced in `phydm+hal_api8821c.c`, where a value of `0x26` indicates that the switch (that controls whether the antenna is being used for wifi or BT?) does not exist. Furthermore, `rfe_type_expand` is shifted right 3 to get `rfe_type`, and `0x26` shifted right 3 is `4`. That corresponds to the Type 4 option listed above... so that's my guess as to what type this module is ¯\_(ツ)_/¯ 
`Type 4 - (1-Ant, DPDT@Ant2), (2G_BTG, iPA, iLNA, iSW), (5G, iPA, iLNA, iSW)`
[/speculation]

If I wanted to continue using the errant dongle, I'd use the following code to load the driver (though I still don't think it makes much of a difference what `rtw_RFE_type` value is provided, as long as it's hex):
```
sudo insmod /lib/modules/$(uname -r)/kernel/drivers/net/wireless/realtek/rtl8821cu/8821cu.ko rtw_RFE_type=0x26
```
... but it's broken, so I'm sending it back.

Thanks for attending my Ted Talk.

closed 2020-09-12 by airdrummingfool

firepainting, 2020-10-31:
Hi, I just got the exact adapter today (EDUP EP-AC1661) and I can't get it to work. I'm just a regular user, not a coder of sorts, and I don't understand a thing you said above :)

So my question is: Will the adapter not work? Is that a conclusion after trying two of them? 

I'm on Linux Mint 19.3. I made a [post on the Mint forums about it](https://forums.linuxmint.com/viewtopic.php?f=90&t=334017) if you want to check it out.

Because I would return this too if it won't work.

Thanks.

airdrummingfool, 2020-10-31:
@firepainting the first adapter I bought did not work out of the box - and I decided to send it back - due to a problem with the contents of the EFUSE. I could have made it work by manually specifying the `rtw_RFE_type` when loading the driver.

The second adapter I purchased worked fine with this driver, without any changes or needing to manually load the driver.

firepainting, 2020-10-31:
Thank you, I appreciate it. 

If I understand it, two adapters of this exact model could be "different" that one might work and the other might not (out of the box), with this driver. I guess I should talk to the vendor, then.

luckylittle, 2020-12-09:
@airdrummingfool @firepainting 

I'm on a brand new Red Hat Enterprise Linux release `8.3 (Ootpa)`, kernel `4.18.0-240.el8.x86_64` and tried what you suggested:

`echo "options 8821cu rtw_RFE_type=0x26" | sudo tee /etc/modprobe.d/8821cu.conf`

To no avail. This is what happens:

```text
[ 2805.561291] RTW: module init start
[ 2805.561293] RTW: rtl8821cu v5.4.1_28754.20180921_COEX20180712-3232
[ 2805.561293] RTW: build time: Dec  9 2020 21:44:46
[ 2805.561294] RTW: rtl8821cu BT-Coex version = COEX20180712-3232
[ 2805.561348] RTW: [HALMAC]11692M
               HALMAC_MAJOR_VER = 1
               HALMAC_PROTOTYPE_VER = 4
               HALMAC_MINOR_VER = 19
               HALMAC_PATCH_VER = 3
[ 2805.561829] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[ 2805.644256] RTW: HW EFUSE
[ 2805.644257] RTW: 0x000: 29 81 00 BC  09 00 21 00  6E 04 A4 34  10 00 30 0B  
[ 2805.644261] RTW: 0x010: 1C 1D 1E 20  20 20 24 25  26 28 28 FF  FF FF FF FF  
[ 2805.644263] RTW: 0x020: FF FF 26 28  28 28 2A 2A  2A 2A 2A 2A  25 26 26 26  
[ 2805.644266] RTW: 0x030: 00 FF FF FF  FF FF FF FF  FF FF 1C 1D  1E 20 20 20  
[ 2805.644268] RTW: 0x040: 24 25 26 28  28 02 FF FF  FF FF FF FF  FF FF FF FF  
[ 2805.644271] RTW: 0x050: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2805.644273] RTW: 0x060: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2805.644276] RTW: 0x070: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2805.644278] RTW: 0x080: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2805.644281] RTW: 0x090: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2805.644283] RTW: 0x0A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2805.644286] RTW: 0x0B0: FF FF FF FF  FF FF FF FF  7F 0A 1F 00  FF FF FF FF  
[ 2805.644288] RTW: 0x0C0: FF 21 00 00  00 00 00 00  00 26 FF FF  FF FF FF FF  
[ 2805.644291] RTW: 0x0D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2805.644293] RTW: 0x0E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2805.644296] RTW: 0x0F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2805.644298] RTW: 0x100: DA 0B 20 C8  E3 46 FF E8  4E 06 88 EF  52 09 03 52  
[ 2805.644301] RTW: 0x110: 65 61 6C 74  65 6B 0E 03  38 30 32 2E  31 31 61 63  
[ 2805.644304] RTW: 0x120: 20 4E 49 43  04 03 46 46  FF FF FF FF  FF FF FF FF  
[ 2805.644306] RTW: 0x130: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2805.644309] RTW: 0x140: FF FF FF FF  00 00 31 0F  FF FF FF FF  FF FF FF FF  
[ 2805.644311] RTW: 0x150: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2805.644314] RTW: 0x160: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2805.644316] RTW: 0x170: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2805.644319] RTW: 0x180: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2805.644321] RTW: 0x190: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2805.644324] RTW: 0x1A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2805.644326] RTW: 0x1B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2805.644329] RTW: 0x1C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2805.644331] RTW: 0x1D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2805.644334] RTW: 0x1E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2805.644336] RTW: 0x1F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2805.644416] RTW: hal_com_config_channel_plan chplan:0x7F
[ 2805.649616] RTW: [HALMAC][ALWAYS]shall R reg twice!!
[ 2805.651503] RTW: WARN [HALMAC][WARN]H2C/C2H ver is compatible!!
[ 2805.699239] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[ 2805.699318] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[ 2805.699398] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[ 2805.699478] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[ 2805.699558] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[ 2805.699638] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[ 2805.699718] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[ 2805.699972] RTW: rtw_regsty_chk_target_tx_power_valid return _FALSE for band:0, path:0, rs:0, t:-1
[ 2805.700519] WARNING: CPU: 3 PID: 18695 at net/wireless/core.c:867 wiphy_register+0x68a/0x940 [cfg80211]
[ 2805.700520] Modules linked in: 8821cu(OE+) cfg80211 nls_utf8 isofs uas usb_storage hidp uinput rfcomm xt_CHECKSUM ipt_MASQUERADE xt_conntrack ipt_REJECT nf_nat_tftp nft_objref nf_conntrack_tftp nft_counter tun bridge stp llc nft_fib_inet nft_fib_ipv4 nft_fib_ipv6 nft_fib nft_reject_inet nf_reject_ipv4 nf_reject_ipv6 nft_reject nft_ct nf_tables_set nft_chain_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6_tables nft_compat ip_set nf_tables nfnetlink cmac bnep sunrpc intel_rapl_msr intel_rapl_common x86_pkg_temp_thermal intel_powerclamp coretemp snd_hda_codec_realtek kvm_intel snd_hda_codec_generic btusb ledtrig_audio snd_hda_codec_hdmi iTCO_wdt btrtl mei_wdt wmi_bmof kvm iTCO_vendor_support btbcm btintel snd_hda_intel bluetooth snd_intel_dspcfg snd_hda_codec irqbypass intel_cstate snd_hda_core ecdh_generic snd_hwdep snd_seq intel_uncore rfkill snd_seq_device i2c_i801 pcspkr intel_rapl_perf snd_pcm lpc_ich snd_timer mei_me snd mei soundcore wmi ip_tables xfs libcrc32c dm_crypt
[ 2805.700542]  sd_mod sg i915 cec intel_gtt i2c_algo_bit drm_kms_helper crct10dif_pclmul crc32_pclmul crc32c_intel syscopyarea ahci sysfillrect sysimgblt fb_sys_fops libahci drm ghash_clmulni_intel libata e1000e video dm_mirror dm_region_hash dm_log dm_mod fuse [last unloaded: cfg80211]
[... cut ...]
/home/USER/Downloads/EP-AC1661/WiFi/rtl8821CU-5.8.1/os_dep/linux/rtw_android.c:669:7: note: each undeclared identifier is reported only once for each function it appears in
make[2]: *** [scripts/Makefile.build:316: /home/USER/Downloads/EP-AC1661/WiFi/rtl8821CU-5.8.1/os_dep/linux/rtw_android.o] Error 1
make[1]: *** [Makefile:1544: _module_/home/USER/Downloads/EP-AC1661/WiFi/rtl8821CU-5.8.1] Error 2
make[1]: Leaving directory '/usr/src/kernels/4.18.0-240.el8.x86_64'
make: *** [Makefile:2327: modules] Error 2
```

Any ideas what to do next?