brektrou/rtl8821CU #85: gcc: error: -mfloat-abi=soft and -mfloat-abi=hard may not be used together

EndlessEden, 2020-09-16:
OS: Archlinux(arm) 
Hardware: Armv7(RaspberryPi 2)

make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/5.4.65-1-ARCH/build M=/var/lib/dkms/rtl8821cu/5.4.18345a8b43/build  modules
make[1]: Entering directory '/usr/lib/modules/5.4.65-1-ARCH/build'
  CC [M]  /var/lib/dkms/rtl8821cu/5.4.18345a8b43/build/core/rtw_cmd.o
gcc: error: -mfloat-abi=soft and -mfloat-abi=hard may not be used together

i dont see this call anywhere, and switching system CFLAGS to explicitly include -mfloat-abi=soft, instead of -mfloat-abi=hard has no effect.

cgarnold, 2020-09-18:
Helo You need to follow this instruction to avoid this problem: https://github.com/brektrou/rtl8821CU#arm-architecture-tweak-for-this-driver-this-solves-compilation-problem-of-this-driver By default is -mfloat-abi=soft but this driver requires -mfloat-abi=hard and you can't use both at same time.

kcozens, 2020-12-23:
In addition to the information in the previous command I commented out the line "EXTRA_CFLAGS += -Wno-implicit-fallthrough" as the gcc v5.4.0 compiler doesn't recognized that option.