brektrou/rtl8821CU #92: ceil() macro in core/rtw_mp.c is mathematically incorrect

pprindeville, 2020-11-05:
I'm looking at this sequence:

```
#define CEILING_POS(X) ((X - (int)(X)) > 0 ? (int)(X + 1) : (int)(X))
#define CEILING_NEG(X) ((X - (int)(X)) < 0 ? (int)(X - 1) : (int)(X))
#define ceil(X) (((X) > 0) ? CEILING_POS(X) : CEILING_NEG(X))
```

and the `CEILING_NEG()` function stands out as being incorrect.  The correct definition of `ceil()` as defined by:

> _the least integer greater than or equal to 'x'_

would require this definition instead:

```
#define CEILING_NEG(X) ((int)(X))
```

i.e. for any x < 0, the ceiling would therefore be x - fract(x).

More broadly, why not have:

```
#Include <math.h>

#define ceil(X) ((int)(ceilf(X)))
```

pprindeville, 2020-11-05:
I've not looked at where this is used to see what the impact of the inaccuracy would be.

champignoom, 2021-03-03:
Being part of driver code, I think the variable is tolerable to have either an imperfect name or a slightly different semantics since it's used only here, and the standard library is certainly too much of a luxury.