brektrou/rtl8821CU #100: Driver doesn't build on kernel 5.10.0-0.rc3.68.fc34.x86_64

patbarron, 2020-11-12:
The rtl8821CU driver doesn't build on kernel version 5.10.0-0.rc3.68.fc34.x86_64 (Fedora Rawhide) because the get_fs() and set_fs() APIs have been removed from the kernel.  These kernel APIs are called from several places in the driver.

core/rtw_btcoex.c
core/rtw_wlan_util.c
os_dep/linux/os_intfs.c
os_dep/linux/osdep_service.c

The use of these APIs has been discouraged for some time, and they were finally removed in kernel 5.10-rc1.

IlgazC, 2020-11-22:
Same here on Manjaro kernel 5.10.rc4 . So, either these API calls get removed or we are stuck on a LTS kernel eh?

Sauerlandlinux, 2020-11-25:
Maybe this patch will help:

`diff -U 3 -H -d -r -N -- a/core/rtw_btcoex.c b/core/rtw_btcoex.c
--- a/core/rtw_btcoex.c	2019-07-20 14:46:28.000000000 +0200
+++ b/core/rtw_btcoex.c	2020-11-24 20:44:47.763115595 +0100
@@ -1444,8 +1444,10 @@
 {
 	u8 error;
 	struct msghdr	udpmsg;
+#ifdef set_fs
 	mm_segment_t	oldfs;
-	struct iovec	iov;
+#endif
+    struct iovec	iov;
 	struct bt_coex_info *pcoex_info = &padapter->coex_info;
 
 	/* RTW_INFO("%s: msg:%s, force:%s\n", __func__, msg, force == _TRUE?"TRUE":"FALSE"); */
@@ -1474,16 +1476,20 @@
 	udpmsg.msg_control	= NULL;
 	udpmsg.msg_controllen = 0;
 	udpmsg.msg_flags	= MSG_DONTWAIT | MSG_NOSIGNAL;
-	oldfs = get_fs();
+#ifdef set_fs
+    oldfs = get_fs();
 	set_fs(KERNEL_DS);
-
+#endif
+    
 #if (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 1, 0))
 	error = sock_sendmsg(pcoex_info->udpsock, &udpmsg);
 #else
 	error = sock_sendmsg(pcoex_info->udpsock, &udpmsg, msg_size);
 #endif
+#ifdef set_fs
 	set_fs(oldfs);
-	if (error < 0) {
+#endif
+    if (error < 0) {
 		RTW_INFO("Error when sendimg msg, error:%d\n", error);
 		return _FAIL;
 	} else
diff -U 3 -H -d -r -N -- a/core/rtw_wlan_util.c b/core/rtw_wlan_util.c
--- a/core/rtw_wlan_util.c	2019-07-20 14:46:28.000000000 +0200
+++ b/core/rtw_wlan_util.c	2020-11-24 20:47:46.411052727 +0100
@@ -4449,7 +4449,9 @@
 
 	int i = 0;
 	struct file *fp;
+#ifdef set_fs
 	mm_segment_t fs;
+#endif
 	loff_t pos = 0;
 	u8 *source = NULL;
 	long len = 0;
@@ -4486,9 +4488,11 @@
 		return 0;
 	}
 
+#ifdef set_fs
 	fs = get_fs();
 	set_fs(KERNEL_DS);
-
+#endif
+    
 	source = rtw_zmalloc(2048);
 
 	if (source != NULL) {
@@ -4497,7 +4501,9 @@
 		rtw_mfree(source, 2048);
 	}
 
+#ifdef set_fs
 	set_fs(fs);
+#endif
 	filp_close(fp, NULL);
 
 	RTW_INFO("-%s-\n", __func__);
diff -U 3 -H -d -r -N -- a/os_dep/linux/os_intfs.c b/os_dep/linux/os_intfs.c
--- a/os_dep/linux/os_intfs.c	2019-07-20 14:46:28.000000000 +0200
+++ b/os_dep/linux/os_intfs.c	2020-11-24 20:54:02.690498868 +0100
@@ -3933,8 +3933,10 @@
 	struct msghdr msg;
 	struct iovec iov;
 	struct sockaddr_nl nladdr;
+#ifdef set_fs
 	mm_segment_t oldfs;
-	char *pg;
+#endif
+    char *pg;
 	int size = 0;
 
 	err = sock_create(AF_NETLINK, SOCK_DGRAM, NETLINK_ROUTE, &sock);
@@ -3972,15 +3974,19 @@
 	msg.msg_controllen = 0;
 	msg.msg_flags = MSG_DONTWAIT;
 
-	oldfs = get_fs();
+#ifdef set_fs
+    oldfs = get_fs();
 	set_fs(KERNEL_DS);
+#endif
 #if (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 1, 0))
 	err = sock_sendmsg(sock, &msg);
 #else
 	err = sock_sendmsg(sock, &msg, sizeof(req));
 #endif
+#ifdef set_fs
 	set_fs(oldfs);
-
+#endif
+    
 	if (err < 0)
 		goto out_sock;
 
@@ -4004,14 +4010,18 @@
 		iov_iter_init(&msg.msg_iter, READ, &iov, 1, PAGE_SIZE);
 #endif
 
+    #ifdef set_fs
 		oldfs = get_fs();
 		set_fs(KERNEL_DS);
+#endif
 #if (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 7, 0))
 		err = sock_recvmsg(sock, &msg, MSG_DONTWAIT);
 #else
 		err = sock_recvmsg(sock, &msg, PAGE_SIZE, MSG_DONTWAIT);
 #endif
+#ifdef set_fs
 		set_fs(oldfs);
+#endif
 
 		if (err < 0)
 			goto out_sock_pg;
@@ -4082,14 +4092,18 @@
 		msg.msg_controllen = 0;
 		msg.msg_flags = MSG_DONTWAIT;
 
+#ifdef set_fs
 		oldfs = get_fs();
 		set_fs(KERNEL_DS);
+#endif
 #if (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 1, 0))
 		err = sock_sendmsg(sock, &msg);
 #else
 		err = sock_sendmsg(sock, &msg, sizeof(req));
 #endif
+#ifdef set_fs
 		set_fs(oldfs);
+#endif
 
 		if (err > 0)
 			goto restart;
diff -U 3 -H -d -r -N -- a/os_dep/osdep_service.c b/os_dep/osdep_service.c
--- a/os_dep/osdep_service.c	2019-07-20 14:46:28.000000000 +0200
+++ b/os_dep/osdep_service.c	2020-11-24 21:23:46.321973595 +0100
@@ -2193,14 +2193,18 @@
 {
 	struct file *fp;
 	int ret = 0;
+#ifdef set_fs
 	mm_segment_t oldfs;
+#endif
 	char buf;
 
 	fp = filp_open(path, O_RDONLY, 0);
 	if (IS_ERR(fp))
 		ret = PTR_ERR(fp);
 	else {
+#ifdef set_fs
 		oldfs = get_fs();
+#endif
 
 		if (1 != readFile(fp, &buf, 1))
 			ret = PTR_ERR(fp);
@@ -2213,7 +2217,9 @@
 			#endif
 		}
 
+#ifdef set_fs
 		set_fs(oldfs);
+#endif
 		filp_close(fp, NULL);
 	}
 	return ret;
@@ -2229,7 +2235,9 @@
 static int retriveFromFile(const char *path, u8 *buf, u32 sz)
 {
 	int ret = -1;
+#ifdef set_fs
 	mm_segment_t oldfs;
+#endif
 	struct file *fp;
 
 	if (path && buf) {
@@ -2237,9 +2245,13 @@
 		if (0 == ret) {
 			RTW_INFO("%s openFile path:%s fp=%p\n", __FUNCTION__, path , fp);
 
+#ifdef set_fs
 			oldfs = get_fs();
 			ret = readFile(fp, buf, sz);
 			set_fs(oldfs);
+#else
+			ret = readFile(fp, buf, sz);
+#endif
 			closeFile(fp);
 
 			RTW_INFO("%s readFile, ret:%d\n", __FUNCTION__, ret);
@@ -2263,7 +2275,9 @@
 static int storeToFile(const char *path, u8 *buf, u32 sz)
 {
 	int ret = 0;
+#ifdef set_fs
 	mm_segment_t oldfs;
+#endif
 	struct file *fp;
 
 	if (path && buf) {
@@ -2271,9 +2285,13 @@
 		if (0 == ret) {
 			RTW_INFO("%s openFile path:%s fp=%p\n", __FUNCTION__, path , fp);
 
+#ifdef set_fs
 			oldfs = get_fs();
 			ret = writeFile(fp, buf, sz);
 			set_fs(oldfs);
+#else
+			ret = readFile(fp, buf, sz);
+#endif
 			closeFile(fp);
 
 			RTW_INFO("%s writeFile, ret:%d\n", __FUNCTION__, ret);
`

patbarron, 2020-11-30:
It looks like this proposed patch just ifdef's out the calls to set_fs() and get_fs(), without replacing them with anything - does that actually work?  If so, why bother using these APIs in the first place?

Sauerlandlinux, 2020-12-02:
Change **#ifdef set_fs** to **#if LINUX_VERSION_CODE < KERNEL_VERSION(5, 10, 0)**

But the old patch has also worked.

d00rsfan, 2020-12-15:
5.10.1 is still error:
```
os_dep/osdep_service.c:2240:12: error: incompatible types when assigning to type ‘mm_segment_t’ from type ‘int’
 2240 |    oldfs = get_fs();
      |            ^~~~~~
```

Sauerlandlinux, 2020-12-15:
Here it builds against kernel 5.10.1:
https://build.opensuse.org/build/home:Sauerland/Kernel_HEAD/x86_64/rtl8821cu/_log
https://build.opensuse.org/package/live_build_log/home:Sauerland/rtl8821cu/Kernel_stable/x86_64

So maybe a bug in your patch?

brektrou, 2020-12-18:
Fixed

closed 2020-12-18 by brektrou