brektrou/rtl8821CU #102: Make on

Jonathon52306, 2020-11-15:
`  Building modules, stage 2.
  MODPOST 1 modules
ERROR: "usb_register_driver" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "cfg80211_mgmt_tx_status" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "wiphy_apply_custom_regulatory" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "ieee80211_frequency_to_channel" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "cfg80211_rx_mgmt" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "__cfg80211_alloc_reply_skb" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "usb_submit_urb" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "usb_free_urb" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "cfg80211_new_sta" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "cfg80211_inform_bss_frame_data" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "ieee80211_get_channel" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "usb_kill_urb" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "cfg80211_unlink_bss" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "usb_alloc_coherent" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "cfg80211_connect_done" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "cfg80211_put_bss" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "cfg80211_roamed" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "cfg80211_scan_done" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "usb_get_dev" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "cfg80211_ibss_joined" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "usb_deregister" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "cfg80211_michael_mic_failure" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "cfg80211_disconnected" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "__cfg80211_send_event_skb" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "cfg80211_get_bss" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "wiphy_new_nm" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "wiphy_free" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "cfg80211_ready_on_channel" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "usb_put_dev" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "cfg80211_ch_switch_notify" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "usb_control_msg" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "usb_alloc_urb" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "usb_reset_device" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "usb_autopm_get_interface" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "wiphy_unregister" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "cfg80211_del_sta_sinfo" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "cfg80211_vendor_cmd_reply" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "usb_autopm_put_interface" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "cfg80211_remain_on_channel_expired" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "wiphy_register" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "__cfg80211_alloc_event_skb" [/root/build/rtl8821CU/8821cu.ko] undefined!
ERROR: "usb_free_coherent" [/root/build/rtl8821CU/8821cu.ko] undefined!
make[2]: *** [scripts/Makefile.modpost:94：__modpost] 错误 1
make[1]: *** [Makefile:1609：modules] 错误 2
make[1]: 离开目录“/root/桌面/linux-5.4”
make: *** [Makefile:2214：modules] 错误 2
`

wangziyannb, 2021-02-17:
Did you finish this MODPOST error?
I meet the same problem :(

wangziyannb, 2021-02-19:
I've fixed this problem: )
"Undefined" simply means you didn't choose to enable cfg80211 and mac80211(this one may not be necessary) when making the kernel.

Find the kernel src, then:

make menuconfig (openwrt is 'make kernel_menuconfig', in the top floor)
Find Networking support/Wireless
Enable cfg80211 and mac80211 options
'make' to recompile the kernel

when kernel is successfully compiled, you can compile the driver again.

简单说就是编译内核的时候没有带上cfg80211，导致给内核编译驱动的时候找不到这个的链接（内核应该提供的）
我的情况是openwrt编译的时候，内核默认不带cfg80211，在编译完成内核后，以kmod的形式安装kmod-cfg80211。所以我这边的kernel实际上没法满足编译这个驱动的要求，过不了modpost。在内核文件夹下make menuconfig或者在openwrt文件夹下make kernel_menuconfig，去加上cfg80211和mac80211(这个不一定要，保险我全加了），重新编译内核再编译驱动就可以得出ko文件了

Jonathon52306, 2021-02-27:
My problem solved with your solution.
Thank you very much.
谢谢，我已经成功使用你的方法解决了这个问题

closed 2021-02-27 by Jonathon52306

koalabearguo, 2022-06-07:
虽然这issue已经关闭了，但我找到了不用重新编译内核的办法，在这里分享一下。
首先需要先编译cfg80211，编译完 之后，环境中会产生mac80211.symvers文件（./build_dir/target-x86_64_musl/linux-x86_64/symvers/mac80211.symvers），把这个文件追加到./build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.188/Module.symvers这个文件尾部，然后再编译rtl8821cu就可以了。不用重新编译内核，只要告诉MODPOST，cfg80211符号表哪里去就可以了。

lieberk2-hd, 2022-11-29:
Anyone know of a way to remove cfg80211 support without modifying kernel?  I don't have the complete kernel source for my distro