brektrou/rtl8821CU #105: Bluetooth failure: Failed to load rtl_bt/rtl8821a_config.bin

cg3dland, 2020-11-25:
The following messages are printed out with the rtl8821cu driver 8821cu.ko when the Nano boots:

bluetooth hci0: Direct firmware load for rtl_bt/rtl8821a_config.bin failed with error -2
bluetooth hci0: Falling back to user helper
Bluetooth: hci0: Failed to load rtl_bt/rtl8821a_config.bin
Bluetooth: hci0 command 0xfc20 tx timeout
Bluetooth: hci0: download fw command failed (-110)

I could not find the rtl8821a_config.bin from the rtl8821cu driver source tree, where could I find this file?

namansharma18899, 2021-06-14:
I'am facing the same Issue, please update the thread cause I'am unable to use my Bluetooth devices currently at this moment cause of this issue.

cg3dland, 2021-06-15:
You could try to find and download btrtl_8821c.tar.bz2, build it, and copy the new driver ko file and bin files into right directories.

Wangzhe00, 2021-09-28:
> You could try to find and download btrtl_8821c.tar.bz2, build it, and copy the new driver ko file and bin files into right directories.

Is there a link to download `btrtl_8821c.tar.bz2` ?