brektrou/rtl8821CU #107: Call trace RIP: 0010:wiphy_register+0x68a/0x940 [cfg80211]

luckylittle, 2020-12-13:
* Red Hat Enterprise Linux release `8.3` (Ootpa)
* Kernel `4.18.0-240.1.1.el8_3.x86_64`
* Device: `EDUP EP-AC1661 2-in-1 WiFi+BT`
* Driver: rtl8821cu v`5.8.1_34171.20190628_COEX20190509-4141` with [this little change](https://github.com/brektrou/rtl8821CU/issues/30#issuecomment-743923510)
* Compiles with no reported errors

1. 
```bash
sudo modprobe 8821cu # complains about missing `rtw_RFE_type`
```

```text
[ 9457.043130] RTW: module init start
[ 9457.043132] RTW: rtl8821cu v5.8.1_34171.20190628_COEX20190509-4141
[ 9457.043132] RTW: build time: Dec 12 2020 23:15:59
[ 9457.043133] RTW: rtl8821cu BT-Coex version = COEX20190509-4141
[ 9457.043282] RTW: [HALMAC]11692M
               HALMAC_MAJOR_VER = 1
               HALMAC_PROTOTYPE_VER = 6
               HALMAC_MINOR_VER = 5
               HALMAC_PATCH_VER = 5
[ 9457.043751] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[ 9457.126749] RTW: HW EFUSE
[ 9457.126751] RTW: 0x000: 29 81 00 BC  09 00 21 00  6E 04 A4 34  10 00 30 0B  
[ 9457.126754] RTW: 0x010: 1C 1D 1E 20  20 20 24 25  26 28 28 FF  FF FF FF FF  
[ 9457.126757] RTW: 0x020: FF FF 26 28  28 28 2A 2A  2A 2A 2A 2A  25 26 26 26  
[ 9457.126760] RTW: 0x030: 00 FF FF FF  FF FF FF FF  FF FF 1C 1D  1E 20 20 20  
[ 9457.126762] RTW: 0x040: 24 25 26 28  28 02 FF FF  FF FF FF FF  FF FF FF FF  
[ 9457.126765] RTW: 0x050: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 9457.126768] RTW: 0x060: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 9457.126770] RTW: 0x070: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 9457.126773] RTW: 0x080: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 9457.126775] RTW: 0x090: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 9457.126778] RTW: 0x0A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 9457.126781] RTW: 0x0B0: FF FF FF FF  FF FF FF FF  7F 0A 1F 00  FF FF FF FF  
[ 9457.126783] RTW: 0x0C0: FF 21 00 00  00 00 00 00  00 26 FF FF  FF FF FF FF  
[ 9457.126786] RTW: 0x0D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 9457.126788] RTW: 0x0E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 9457.126791] RTW: 0x0F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 9457.126794] RTW: 0x100: DA 0B 20 C8  E3 46 FF E8  4E 06 88 EF  52 09 03 52  
[ 9457.126796] RTW: 0x110: 65 61 6C 74  65 6B 0E 03  38 30 32 2E  31 31 61 63  
[ 9457.126799] RTW: 0x120: 20 4E 49 43  04 03 46 46  FF FF FF FF  FF FF FF FF  
[ 9457.126802] RTW: 0x130: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 9457.126804] RTW: 0x140: FF FF FF FF  00 00 31 0F  FF FF FF FF  FF FF FF FF  
[ 9457.126807] RTW: 0x150: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 9457.126809] RTW: 0x160: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 9457.126812] RTW: 0x170: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 9457.126815] RTW: 0x180: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 9457.126817] RTW: 0x190: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 9457.126820] RTW: 0x1A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 9457.126822] RTW: 0x1B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 9457.126825] RTW: 0x1C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 9457.126828] RTW: 0x1D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 9457.126830] RTW: 0x1E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 9457.126833] RTW: 0x1F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 9457.126906] RTW: hal_com_config_channel_plan chplan:0x7F
[ 9457.126906] RTW: ERROR 
               Empty EFUSE with unknown REF type!!
[ 9457.126907] RTW: ERROR please program efuse or specify correct RFE type.
[ 9457.126907] RTW: ERROR cmd: insmod rtl8821cx.ko rtw_RFE_type=<rfe_type>
[ 9457.126954] usbcore: registered new interface driver rtl8821cu
[ 9457.126955] RTW: module init ret=0
```

2. As suggested in #83 , value of rtw_RFE_type was provided
```bash
sudo modprobe -r 8821cu
sudo modprobe 8821cu rtw_RFE_type=7 # produces call trace
```

[... 201 lines cut ...]
[14701.502808]  usb_register_driver+0x7a/0x130
[14701.502809]  ? 0xffffffffc13fa000
[14701.502838]  rtw_drv_entry+0xc0/0x1000 [8821cu]
[14701.502841]  do_one_initcall+0x46/0x1c3
[14701.502845]  ? _cond_resched+0x15/0x30
[14701.502847]  ? kmem_cache_alloc_trace+0x140/0x1c0
[14701.502850]  do_init_module+0x5a/0x220
[14701.502852]  load_module+0x14b4/0x17e0
[14701.502855]  ? __do_sys_finit_module+0xa8/0x110
[14701.502856]  __do_sys_finit_module+0xa8/0x110
[14701.502858]  do_syscall_64+0x5b/0x1a0
[14701.502860]  entry_SYSCALL_64_after_hwframe+0x65/0xca
[14701.502861] RIP: 0033:0x7f6f4bc0c78d
[14701.502863] Code: 00 c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d cb 56 2c 00 f7 d8 64 89 01 48
[14701.502863] RSP: 002b:00007ffd96e30e88 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[14701.502864] RAX: ffffffffffffffda RBX: 00005566b88a8c50 RCX: 00007f6f4bc0c78d
[14701.502865] RDX: 0000000000000000 RSI: 00005566b88a9240 RDI: 0000000000000004
[14701.502865] RBP: 00005566b88a9240 R08: 0000000000000000 R09: 0000000000000000
[14701.502866] R10: 0000000000000004 R11: 0000000000000246 R12: 0000000000000000
[14701.502866] R13: 00005566b88a8b30 R14: 0000000000040000 R15: 0000000000000000
[14701.502868] ---[ end trace ab0e10acefdcbe0f ]---
[14701.502966] RTW: WARN free_recv_skb_queue not empty, 8
[14701.503016] usbcore: registered new interface driver rtl8821cu
[14701.503027] RTW: module init ret=0
```

brektrou, 2020-12-18:
Fixed for version 5.8.1
https://github.com/brektrou/rtl8821CU/commit/878da0557a66be55177559f415198c22ec85bc56
https://github.com/brektrou/rtl8821CU/commit/03a238a62a525c9ddf3bb3c0bbe7031be93122b0

closed 2020-12-18 by brektrou

luckylittle, 2020-12-19:
@brektrou It is unfortunately still not working on Kernel `4.18.0-240.1.1.el8_3.x86_64, x86_64`. I can however confirm, that it works without any errors on `5.9.14-1.el8.elrepo.x86_64, x86_64`, RHEL v8.3.

edrozenberg, 2021-01-01:
I see similar-looking `dmesg` output on my MiSTER FPGA system running `Linux mst1 4.19.0-socfpga-r1 #9 SMP Tue Sep 8 22:13:28 CST 2020 armv7l GNU/Linux`

Built with `make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- KSRC=../Linux-Kernel_MiSTer modules`

And I made the following changes to the `Makefile`:
```
...
CONFIG_MP_VHT_HW_TX_MODE = n
...
EXTRA_CFLAGS += -mfloat-abi=soft
```

My `dmesg` output:
```
[Fri Jan  1 19:50:02 2021] RTW: HW EFUSE
[Fri Jan  1 19:50:02 2021] RTW: 0x000: 29 81 00 BC  09 00 21 00  6E 04 A4 10  10 00 30 0B  
[Fri Jan  1 19:50:02 2021] RTW: 0x010: FF FF FF FF  FF FF FF FF  FF FF FF 02  FF FF FF FF  
[Fri Jan  1 19:50:02 2021] RTW: 0x020: FF FF 23 23  23 23 28 28  28 28 28 28  28 28 28 28  
[Fri Jan  1 19:50:02 2021] RTW: 0x030: 02 FF FF FF  FF FF EC FF  FF FF 2B 2B  2B 2B 2B 2B  
[Fri Jan  1 19:50:02 2021] RTW: 0x040: 30 30 30 30  30 33 FF FF  FF FF FF FF  FF FF FF FF  
[Fri Jan  1 19:50:02 2021] RTW: 0x050: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[Fri Jan  1 19:50:02 2021] RTW: 0x060: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[Fri Jan  1 19:50:02 2021] RTW: 0x070: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[Fri Jan  1 19:50:02 2021] RTW: 0x080: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[Fri Jan  1 19:50:02 2021] RTW: 0x090: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[Fri Jan  1 19:50:02 2021] RTW: 0x0A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[Fri Jan  1 19:50:02 2021] RTW: 0x0B0: FF FF FF FF  FF FF FF FF  7F 06 20 00  FF FF FF FF  
[Fri Jan  1 19:50:02 2021] RTW: 0x0C0: FF 01 00 41  00 00 00 00  00 FF 22 FF  FF FF FF FF  
[Fri Jan  1 19:50:02 2021] RTW: 0x0D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[Fri Jan  1 19:50:02 2021] RTW: 0x0E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[Fri Jan  1 19:50:02 2021] RTW: 0x0F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[Fri Jan  1 19:50:02 2021] RTW: 0x100: DA 0B 11 C8  E3 46 02 1C  BF CE E4 20  F4 09 03 52  
[Fri Jan  1 19:50:02 2021] RTW: 0x110: 65 61 6C 74  65 6B 0E 03  38 30 32 2E  31 31 61 63  
[Fri Jan  1 19:50:02 2021] RTW: 0x120: 20 4E 49 43  08 03 31 32  33 34 35 36  FF FF FF FF  
[Fri Jan  1 19:50:02 2021] RTW: 0x130: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[Fri Jan  1 19:50:02 2021] RTW: 0x140: FF FF FF FF  00 00 31 0F  FF FF FF FF  FF FF FF FF  
[Fri Jan  1 19:50:02 2021] RTW: 0x150: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[Fri Jan  1 19:50:02 2021] RTW: 0x160: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[Fri Jan  1 19:50:02 2021] RTW: 0x170: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[Fri Jan  1 19:50:02 2021] RTW: 0x180: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[Fri Jan  1 19:50:02 2021] RTW: 0x190: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[Fri Jan  1 19:50:02 2021] RTW: 0x1A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[Fri Jan  1 19:50:02 2021] RTW: 0x1B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[Fri Jan  1 19:50:02 2021] RTW: 0x1C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[Fri Jan  1 19:50:02 2021] RTW: 0x1D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[Fri Jan  1 19:50:02 2021] RTW: 0x1E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[Fri Jan  1 19:50:02 2021] RTW: 0x1F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[Fri Jan  1 19:50:02 2021] RTW: hal_com_config_channel_plan chplan:0x7F
[Fri Jan  1 19:50:02 2021] RTW: [HALMAC][ALWAYS]shall R reg twice!!
[Fri Jan  1 19:50:02 2021] RTW: WARN [HALMAC][WARN]H2C/C2H ver is compatible!!
[Fri Jan  1 19:50:03 2021] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[Fri Jan  1 19:50:03 2021] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[Fri Jan  1 19:50:03 2021] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[Fri Jan  1 19:50:03 2021] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[Fri Jan  1 19:50:03 2021] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[Fri Jan  1 19:50:03 2021] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[Fri Jan  1 19:50:03 2021] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[Fri Jan  1 19:50:03 2021] RTW: rtw_regsty_chk_target_tx_power_valid return _FALSE for band:0, path:0, rs:0, t:-1
[Fri Jan  1 19:50:03 2021] RTW: rtw_ndev_init(wlan0) if1 mac_addr=1c:bf:ce:e4:20:f4
[Fri Jan  1 19:50:03 2021] usbcore: registered new interface driver rtl8821cu
[Fri Jan  1 19:50:03 2021] RTW: module init ret=0
[Fri Jan  1 19:50:03 2021] EXT4-fs (loop0): recovery complete
[Fri Jan  1 19:50:03 2021] EXT4-fs (loop0): mounted filesystem with ordered data mode. Opts: (null)
[Fri Jan  1 19:50:04 2021] Micrel KSZ9031 Gigabit PHY stmmac-0:01: attached PHY driver [Micrel KSZ9031 Gigabit PHY] (mii_bus:phy_addr=stmmac-0:01, irq=POLL)
[Fri Jan  1 19:50:04 2021] socfpga-dwmac ff702000.ethernet eth0: No Safety Features support found
[Fri Jan  1 19:50:04 2021] socfpga-dwmac ff702000.ethernet eth0: IEEE 1588-2008 Advanced Timestamp supported
[Fri Jan  1 19:50:04 2021] socfpga-dwmac ff702000.ethernet eth0: registered PTP clock
[Fri Jan  1 19:50:04 2021] RTW: [HALMAC][ALWAYS]shall R reg twice!!
[Fri Jan  1 19:50:04 2021] RTW: WARN [HALMAC][WARN]H2C/C2H ver is compatible!!
...
...
[Fri Jan  1 19:50:09 2021] RTW: mac_id : 0
[Fri Jan  1 19:50:09 2021] RTW: wireless_mode : 0x44
[Fri Jan  1 19:50:09 2021] RTW: mimo_type : 0
[Fri Jan  1 19:50:09 2021] RTW: bw_mode : 80MHz, ra_bw_mode : 80MHz
[Fri Jan  1 19:50:09 2021] RTW: rate_id : 10
[... cut ...]
[Fri Jan  1 19:50:09 2021] RTW: is_noisy : N
[Fri Jan  1 19:50:09 2021] RTW: txrx_state : 0
[Fri Jan  1 19:50:09 2021] RTW: curr_tx_rate : CCK_1M (L)
[Fri Jan  1 19:50:09 2021] RTW: curr_tx_bw : 20MHz
[Fri Jan  1 19:50:09 2021] RTW: curr_retry_ratio : 0
[Fri Jan  1 19:50:09 2021] RTW: ra_mask : 0x00000000003ffff0

[Fri Jan  1 19:50:09 2021] RTW: recv eapol packet
[Fri Jan  1 19:50:09 2021] RTW: send eapol packet
[Fri Jan  1 19:50:09 2021] RTW: recv eapol packet
[Fri Jan  1 19:50:09 2021] RTW: send eapol packet
```