brektrou/rtl8821CU #111: Build Error on Manjaro Linux with Kernel 5.10.2-2-MANJARO

prathmesh01, 2020-12-23:
Error might be same for any Arch Linux based OS with Kernel 5.10.x

```
DKMS make.log for rtl8821CU-5.4.1 for kernel 5.10.2-2-MANJARO (x86_64)
Wednesday 23 December 2020 11:51:54 AM IST
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.10.2-2-MANJARO/build M=/var/lib/dkms/rtl8821CU/5.4.1/build  modules
make[1]: Entering directory '/usr/lib/modules/5.10.2-2-MANJARO/build'
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_cmd.o
In file included from /var/lib/dkms/rtl8821CU/5.4.1/build/include/drv_types.h:30,
                 from /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_cmd.c:17:
/var/lib/dkms/rtl8821CU/5.4.1/build/include/wifi.h:1032: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 1032 | #define IEEE80211_MAX_AMPDU_BUF 0x40
      | 
In file included from /var/lib/dkms/rtl8821CU/5.4.1/build/include/osdep_service_linux.h:83,
                 from /var/lib/dkms/rtl8821CU/5.4.1/build/include/osdep_service.h:50,
                 from /var/lib/dkms/rtl8821CU/5.4.1/build/include/drv_types.h:27,
                 from /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_cmd.c:17:
./include/linux/ieee80211.h:1656: note: this is the location of the previous definition
 1656 | #define IEEE80211_MAX_AMPDU_BUF  0x100
      | 
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_security.o
In file included from /var/lib/dkms/rtl8821CU/5.4.1/build/include/drv_types.h:30,
                 from /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_security.c:17:
/var/lib/dkms/rtl8821CU/5.4.1/build/include/wifi.h:1032: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 1032 | #define IEEE80211_MAX_AMPDU_BUF 0x40
      | 
In file included from /var/lib/dkms/rtl8821CU/5.4.1/build/include/osdep_service_linux.h:83,
                 from /var/lib/dkms/rtl8821CU/5.4.1/build/include/osdep_service.h:50,
                 from /var/lib/dkms/rtl8821CU/5.4.1/build/include/drv_types.h:27,
                 from /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_security.c:17:
./include/linux/ieee80211.h:1656: note: this is the location of the previous definition
 1656 | #define IEEE80211_MAX_AMPDU_BUF  0x100
      | 
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_debug.o
In file included from /var/lib/dkms/rtl8821CU/5.4.1/build/include/drv_types.h:30,
                 from /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_debug.c:17:
/var/lib/dkms/rtl8821CU/5.4.1/build/include/wifi.h:1032: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 1032 | #define IEEE80211_MAX_AMPDU_BUF 0x40
      | 
In file included from /var/lib/dkms/rtl8821CU/5.4.1/build/include/osdep_service_linux.h:83,
                 from /var/lib/dkms/rtl8821CU/5.4.1/build/include/osdep_service.h:50,
                 from /var/lib/dkms/rtl8821CU/5.4.1/build/include/drv_types.h:27,
                 from /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_debug.c:17:
./include/linux/ieee80211.h:1656: note: this is the location of the previous definition
 1656 | #define IEEE80211_MAX_AMPDU_BUF  0x100
      | 
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_io.o
In file included from /var/lib/dkms/rtl8821CU/5.4.1/build/include/drv_types.h:30,
                 from /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_io.c:47:
/var/lib/dkms/rtl8821CU/5.4.1/build/include/wifi.h:1032: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 1032 | #define IEEE80211_MAX_AMPDU_BUF 0x40
      | 
In file included from /var/lib/dkms/rtl8821CU/5.4.1/build/include/osdep_service_linux.h:83,
                 from /var/lib/dkms/rtl8821CU/5.4.1/build/include/osdep_service.h:50,
                 from /var/lib/dkms/rtl8821CU/5.4.1/build/include/drv_types.h:27,
                 from /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_io.c:47:
./include/linux/ieee80211.h:1656: note: this is the location of the previous definition
 1656 | #define IEEE80211_MAX_AMPDU_BUF  0x100
      | 
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_ioctl_query.o
In file included from /var/lib/dkms/rtl8821CU/5.4.1/build/include/drv_types.h:30,
                 from /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_ioctl_query.c:17:
/var/lib/dkms/rtl8821CU/5.4.1/build/include/wifi.h:1032: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 1032 | #define IEEE80211_MAX_AMPDU_BUF 0x40
      | 
In file included from /var/lib/dkms/rtl8821CU/5.4.1/build/include/osdep_service_linux.h:83,
                 from /var/lib/dkms/rtl8821CU/5.4.1/build/include/osdep_service.h:50,
                 from /var/lib/dkms/rtl8821CU/5.4.1/build/include/drv_types.h:27,
                 from /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_ioctl_query.c:17:
./include/linux/ieee80211.h:1656: note: this is the location of the previous definition
[... 364 lines cut ...]
/var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/osdep_service.c: In function ‘isFileReadable’:
/var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/osdep_service.c:2203:11: error: implicit declaration of function ‘get_fs’; did you mean ‘get_sa’? [-Werror=implicit-function-declaration]
 2203 |   oldfs = get_fs();
      |           ^~~~~~
      |           get_sa
/var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/osdep_service.c:2203:11: error: incompatible types when assigning to type ‘mm_segment_t’ from type ‘int’
/var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/osdep_service.c:2216:3: error: implicit declaration of function ‘set_fs’; did you mean ‘sget_fc’? [-Werror=implicit-function-declaration]
 2216 |   set_fs(oldfs);
      |   ^~~~~~
      |   sget_fc
/var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/osdep_service.c: In function ‘retriveFromFile’:
/var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/osdep_service.c:2240:12: error: incompatible types when assigning to type ‘mm_segment_t’ from type ‘int’
 2240 |    oldfs = get_fs();
      |            ^~~~~~
/var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/osdep_service.c: In function ‘storeToFile’:
/var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/osdep_service.c:2274:12: error: incompatible types when assigning to type ‘mm_segment_t’ from type ‘int’
 2274 |    oldfs = get_fs();
      |            ^~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:279: /var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/osdep_service.o] Error 1
make[1]: *** [Makefile:1805: /var/lib/dkms/rtl8821CU/5.4.1/build] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.10.2-2-MANJARO/build'
make: *** [Makefile:2213: modules] Error 2

```

brektrou, 2020-12-24:
Fixed e00c6646716a1e9c98f8b2a39b24e06487632727 f10b0b86703e6380f70082820dbd7704b8c94d7f 397a8182cfb077f24712463595f0f0efcff6871d 6df86a017f3199b003b703fc81cd249361d12c3b

prathmesh01, 2021-01-04:
Thank you, build is now working and the driver is working fine.

closed 2021-01-04 by prathmesh01