brektrou/rtl8821CU #113: Problem running on CentOS 8

VinzenzBildstein, 2020-12-27:
I'm trying to get my Wifi adapter to work under CentOS 8.3. I copied this repository onto the machine, installed dkms, and successfully installed the module. But when I connect the USB Wifi adapter or reboot the computer, it still fails to recognize the adapter. It shows up in lsusb as 0bda:c811, which is supposed to be supported by this module, right?
This is the output I get from dmesg during booting:
```
[   13.883718] 8821cu: loading out-of-tree module taints kernel.
[   13.884562] 8821cu: module verification failed: signature and/or required key missing - tainting kernel
[   13.904631] RTW: module init start
[   13.904632] RTW: rtl8821cu v5.4.1_28754.20180921_COEX20180712-3232
[   13.904632] RTW: build time: Dec 27 2020 12:26:27
[   13.904633] RTW: rtl8821cu BT-Coex version = COEX20180712-3232
[   13.904686] RTW: [HALMAC]11692M
               HALMAC_MAJOR_VER = 1
               HALMAC_PROTOTYPE_VER = 4
               HALMAC_MINOR_VER = 19
               HALMAC_PATCH_VER = 3
[   13.907493] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[   14.179466] RTW: HW EFUSE
[   14.179468] RTW: 0x000: 29 81 00 BC  09 00 31 00  6F 04 A4 10  12 00 30 0B
[   14.179472] RTW: 0x010: FF FF FF FF  FF FF FF FF  FF FF FF 02  FF FF FF FF
[   14.179475] RTW: 0x020: FF FF 28 28  29 2A 2A 2E  2D 2D 30 31  2F 2C 2C 2C
[   14.179477] RTW: 0x030: 12 FF FF FF  FF FF 1C FF  FF FF 28 28  2F 2F 2F 2F
[   14.179480] RTW: 0x040: 33 33 33 33  30 01 FF FF  FF FF FF FF  FF FF FF FF
[   14.179483] RTW: 0x050: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   14.179485] RTW: 0x060: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   14.179488] RTW: 0x070: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   14.179491] RTW: 0x080: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   14.179494] RTW: 0x090: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   14.179496] RTW: 0x0A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   14.179499] RTW: 0x0B0: FF FF FF FF  FF FF FF FF  7F 33 20 00  FF FF FF FF
[   14.179502] RTW: 0x0C0: FF 01 00 41  00 00 00 00  00 FF 22 FF  FF FF FF FF
[   14.179505] RTW: 0x0D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   14.179507] RTW: 0x0E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   14.179510] RTW: 0x0F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   14.179513] RTW: 0x100: DA 0B 11 C8  E3 46 02 1C  BF CE E2 5C  75 09 03 52
[   14.179516] RTW: 0x110: 65 61 6C 74  65 6B 0E 03  38 30 32 2E  31 31 61 63
[   14.179518] RTW: 0x120: 20 4E 49 43  08 03 31 32  33 34 35 36  FF FF FF FF
[   14.179521] RTW: 0x130: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   14.179524] RTW: 0x140: FF FF FF FF  00 00 31 0F  FF FF FF FF  FF FF FF FF
[   14.179526] RTW: 0x150: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   14.179529] RTW: 0x160: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   14.179532] RTW: 0x170: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   14.179535] RTW: 0x180: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   14.179537] RTW: 0x190: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   14.179540] RTW: 0x1A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   14.179543] RTW: 0x1B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   14.179546] RTW: 0x1C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   14.179548] RTW: 0x1D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   14.179551] RTW: 0x1E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   14.179554] RTW: 0x1F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   14.179831] RTW: hal_com_config_channel_plan chplan:0x7F
[   14.192749] RTW: [HALMAC][ALWAYS]shall R reg twice!!
[   14.196333] RTW: WARN [HALMAC][WARN]H2C/C2H ver is compatible!!
[   14.305596] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[   14.305834] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[   14.306130] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[   14.306380] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend 
[   14.306594] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[   14.306833] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[   14.307152] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[   14.307610] RTW: rtw_regsty_chk_target_tx_power_valid return _FALSE for band:0, path:0, rs:0, t:-1
[   14.309211] WARNING: CPU: 7 PID: 767 at net/wireless/core.c:867 wiphy_register+0x68a/0x940 [cfg80211]
[   14.309212] Modules linked in: 8821cu(OE+) dell_wmi snd_hda_codec_realtek sparse_keymap snd_hda_codec_generic video ledtrig_audio intel_powerclamp snd_hda_intel snd_intel_dspcfg snd_hda_codec coretemp cfg80211 snd_hda_core snd_hwdep dell_smbios snd_seq snd_seq_device kvm_intel dell_wmi_descriptor snd_pcm gpio_ich rfkill kvm snd_timer snd ppdev soundcore irqbypass wmi_bmof iTCO_wdt iTCO_vendor_support wmi intel_cstate intel_uncore dcdbas pcspkr parport_pc i2c_i801 i7core_edac parport acpi_cpufreq lpc_ich ip_tables xfs libcrc32c sr_mod cdrom sd_mod sg uas usb_storage radeon i2c_algo_bit drm_kms_helper ahci syscopyarea libahci sysfillrect sysimgblt fb_sys_fops ttm libata crc32c_intel serio_raw drm e1000e dm_mirror dm_region_hash dm_log dm_mod fuse
[   14.309231] CPU: 7 PID: 767 Comm: systemd-udevd Tainted: G           OE    --------- -  - 4.18.0-240.el8.x86_64 #1
[   14.309232] Hardware name: Dell Inc. OptiPlex 980                 /0D441T, BIOS A05 11/03/2010
[   14.309241] RIP: 0010:wiphy_register+0x68a/0x940 [cfg80211]
[   14.309242] Code: 78 20 00 74 25 83 ea 01 48 8d 14 52 48 c1 e2 04 48 01 c2 48 83 78 10 00 74 1c 48 39 c2 74 5a 48 83 c0 30 48 83 78 20 00 75 e9 <0f> 0b b8 ea ff ff ff e9 8a fa ff ff 48 83 78 18 00 75 dd 0f 0b b8
[   14.309243] RSP: 0018:ffffbbe0017239e0 EFLAGS: 00010246
[   14.309244] RAX: ffffffffc0c30180 RBX: ffff9f6f07c514a0 RCX: 0000000000000000
[   14.309244] RDX: 0000000000000002 RSI: ffff9f700d06a058 RDI: 0000000000000001
[   14.309245] RBP: ffffbbe001723a70 R08: 0000000000000004 R09: 0000000000000003
[   14.309245] R10: ffff9f700700f000 R11: ffff9f700d939bc4 R12: ffffbbe00161c4f0
[... 74 lines cut ...]
[   14.309804]  usb_register_driver+0x7a/0x130
[   14.309805]  ? 0xffffffffc0d5b000
[   14.309827]  rtw_drv_entry+0xc0/0x1000 [8821cu]
[   14.309828]  do_one_initcall+0x46/0x1c3
[   14.309830]  ? _cond_resched+0x15/0x30
[   14.309831]  ? kmem_cache_alloc_trace+0x140/0x1c0
[   14.309832]  do_init_module+0x5a/0x220
[   14.309833]  load_module+0x14b4/0x17e0
[   14.309835]  ? __do_sys_finit_module+0xa8/0x110
[   14.309836]  __do_sys_finit_module+0xa8/0x110
[   14.309838]  do_syscall_64+0x5b/0x1a0
[   14.309839]  entry_SYSCALL_64_after_hwframe+0x65/0xca
[   14.309840] RIP: 0033:0x7f9e76cb778d
[   14.309841] Code: 00 c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d cb 56 2c 00 f7 d8 64 89 01 48
[   14.309841] RSP: 002b:00007ffe0e661d48 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[   14.309842] RAX: ffffffffffffffda RBX: 000055dd0d357050 RCX: 00007f9e76cb778d
[   14.309843] RDX: 0000000000000000 RSI: 00007f9e7782980d RDI: 0000000000000010
[   14.309843] RBP: 00007f9e7782980d R08: 0000000000000000 R09: 00007ffe0e661e70
[   14.309844] R10: 0000000000000010 R11: 0000000000000246 R12: 0000000000000000
[   14.309844] R13: 000055dd0d393780 R14: 0000000000020000 R15: 0000000000000000
[   14.309845] ---[ end trace bd8b00e913737b3c ]---
[   14.319234] WARNING: CPU: 7 PID: 767 at /root/rtl8821CU/os_dep/linux/os_intfs.c:3105 rtw_os_ndevs_register+0xaf/0xf6 [8821cu]
```

Is there any way to make this adapter work?

soruk42, 2021-03-29:
I have exactly the same problem, using the master branch as of today's date.

alexisfrjp, 2021-10-03:
Same problem.