brektrou/rtl8821CU #120: usb_mode switch don`t works

Nazarovv676, 2021-01-27:
sudo usb_modeswitch -KW -v 0bda -p c811 
Take all parameters from the command line

 * usb_modeswitch: handle USB devices with multiple modes
 * Version 2.6.1 (C) Josua Dietze 2017
 * Based on libusb1/libusbx

 ! PLEASE REPORT NEW CONFIGURATIONS !

DefaultVendor=  0x0bda
DefaultProduct= 0xc811

StandardEject=1

Look for default devices ...
  found USB ID 1d6b:0003
  found USB ID 0bda:c811
   vendor ID matched
   product ID matched
  found USB ID 05c6:9024
  found USB ID 09da:2268
  found USB ID 0d8c:0012
  found USB ID 1a40:0101
  found USB ID 1d6b:0002
  found USB ID 1d6b:0003
  found USB ID 1bcf:08a0
  found USB ID 1d6b:0002
 Found devices in default mode (1)
Access device 003 on bus 003
Get the current device configuration ...
Current configuration number is 1
Use interface number 0
 with class 255
Error: can't use storage command in MessageContent with interface 0; interface class is 255, expected 8. Abort

martin-wanderer, 2021-03-15:
Have the same issue with tenda U9

ajay340, 2021-03-27:
Same issue here on a rpi 3.
```
sudo usb_modeswitch -KW -v 0bda -p c811
Take all parameters from the command line

 * usb_modeswitch: handle USB devices with multiple modes
 * Version 2.5.2 (C) Josua Dietze 2017
 * Based on libusb1/libusbx

 ! PLEASE REPORT NEW CONFIGURATIONS !

DefaultVendor=  0x0bda
DefaultProduct= 0xc811

StandardEject=1

Look for default devices ...
  found USB ID 0bda:c811
   vendor ID matched
   product ID matched
  found USB ID 0424:ec00
  found USB ID 0424:9514
  found USB ID 1d6b:0002
 Found devices in default mode (1)
Access device 006 on bus 001
Get the current device configuration ...
Current configuration number is 1
Use interface number 0
 with class 255
Error: can't use storage command in MessageContent with interface 0; interface class is 255, expected 8. Abort
```

ajay340, 2021-03-29:
Hey guys, I found another repository which solved my issue. https://github.com/fastoe/RTL8811CU_for_Raspbian

manoj153, 2021-09-16:
Same here can't mode switch 
```
manoj@manoj-MS-7C52:~$ lsusb 
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 004: ID 2717:ff88 Xiaomi Inc. Mi/Redmi series (RNDIS + ADB)
Bus 003 Device 002: ID 0bda:1a2b Realtek Semiconductor Corp. DISK
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 046d:c31d Logitech, Inc. Media Keyboard K200
Bus 001 Device 002: ID 046d:c247 Logitech, Inc. G100S Optical Gaming Mouse
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
manoj@manoj-MS-7C52:~$ sudo usb_modeswitch -KW -v 0bda -p 1a2b
Take all parameters from the command line

 * usb_modeswitch: handle USB devices with multiple modes
 * Version 2.5.2 (C) Josua Dietze 2017
 * Based on libusb1/libusbx

 ! PLEASE REPORT NEW CONFIGURATIONS !

DefaultVendor=  0x0bda
DefaultProduct= 0x1a2b

StandardEject=1

Look for default devices ...
  found USB ID 1d6b:0003
  found USB ID 2717:ff88
  found USB ID 0bda:1a2b
   vendor ID matched
   product ID matched
  found USB ID 1d6b:0002
  found USB ID 1d6b:0003
  found USB ID 046d:c31d
  found USB ID 046d:c247
  found USB ID 1d6b:0002
 Found devices in default mode (1)
Access device 002 on bus 003
Get the current device configuration ...
Current configuration number is 1
Use interface number 0
 with class 8
Use endpoints 0x0b (out) and 0x8a (in)

USB description data (for identification)
-------------------------
Manufacturer: Realtek
     Product: DISK
  Serial No.: not provided
-------------------------
Sending standard EJECT sequence
Looking for active drivers ...
Set up interface 0
Use endpoint 0x0b for message sending ...
Trying to send message 1 to endpoint 0x0b ...
 Sending the message returned error -7. Try to continue
Read the response to message 1 (CSW) ...
 Response reading failed (error -8)
 Device is gone, skip any further commands
-> Run lsusb to note any changes. Bye!
```

oreme2, 2023-04-05:
> Same here can't mode switch
>
> ```
> manoj@manoj-MS-7C52:~$ lsusb
> Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
> Bus 003 Device 004: ID 2717:ff88 Xiaomi Inc. Mi/Redmi series (RNDIS + [...]

same error logs. have you solved it?

magbone, 2023-04-10:
changing other usb interfaces would help me solve the problem