brektrou/rtl8821CU #133: Failed to install on nvidia jetson nano

rootadminWalker, 2021-05-12:
Hello everybody, I'm using nvidia jetson nano and I want to install this driver, but I've encountered the following error:
command: sudo ./dkms-install.sh 
```
About to run dkms install steps...
Error! DKMS tree already contains: rtl8821CU-5.4.1
You cannot add the same module/version combo more than once.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' KVER=4.9.140-tegra...(bad exit status: 2)
ERROR (dkms apport): binary package for rtl8821CU: 5.4.1 not found
Error! Bad return status for module build on kernel: 4.9.140-tegra (aarch64)
Consult /var/lib/dkms/rtl8821CU/5.4.1/build/make.log for more information.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' KVER=4.9.140-tegra...(bad exit status: 2)
ERROR (dkms apport): binary package for rtl8821CU: 5.4.1 not found
Error! Bad return status for module build on kernel: 4.9.140-tegra (aarch64)
Consult /var/lib/dkms/rtl8821CU/5.4.1/build/make.log for more information.
Finished running dkms install steps.

```
I've checked another issue, which another user encountered the problem while he forgotten to install bc.
I'm using:
```
Nvidia's Ubuntu 18.04.4 LTS development kit
bc 1.07.1
linux kernel 4.9.140-tegra aarch64

```
How can I fix this?

rootadminWalker, 2021-05-24:
Is there anybody who has a solution for this? I'm in urgent right now, please🙏🙏🙏

igorpecovnik, 2021-07-04:
Here https://www.armbian.com/jetson-nano/ it should just work out of the box. (kernel 5.10.y)