brektrou/rtl8821CU #137: "FP instruction issued in kernel mode with FP unit disabled" on armv7h

Moarc, 2021-06-24:
Arch Linux ARM on an unsupported board, kernel 5.11.2-1-ARCH, arch ARMv7 (armv7l).
Upon loading the module, it throws the error
```[ 2453.884446] BUG: FP instruction issued in kernel mode with FP unit disabled
[ 2453.891457] FPEXC == 0x00000000
[ 2453.894621] Internal error: Oops - undefined instruction: 0 [#1] PREEMPT SMP ARM
```
and dumps a bit of memory into the kernel log.

My CPU is supposed to have VFP/NEON.

I would need some instructions in order to troubleshoot, for example some other module that also uses floating point instructions, but I certainly can spare the time to provide anything you would need.

Other people have mentioned this very issue in [TI forums](https://e2e.ti.com/support/processors-group/processors/f/processors-forum/820527/linux-am3352-rtl8821cu-usb-wifi-fp-instruction-issued-in-kernel-mode-with-fp-unit-disabled-error), but without detail.

rrafal1337, 2021-06-24:
Hi. Did you tried to perform this workaround? https://github.com/brektrou/rtl8821CU#arm-architecture-tweak-for-this-driver-this-solves-compilation-problem-of-this-driver
It has been called aarch32 in readme but it is for armv7l. Just check those commands. Removing soft float form kernel hearders helped me to compile driver with hard float and thanks to that driver stopped to crash.

Moarc, 2021-06-24:
I actually built the module this way before even encountering any issue, since it was applied by default in the AUR package - so the error appears after having applied this workaround.

Moarc, 2021-06-29:
Sorry, it seems this was an issue with hardware floating point support being disabled in the default Arch Linux ARM armv7l kernel. The driver works perfectly fine on Armbian, with a kernel that does use floating-point instructions.

closed 2021-06-29 by Moarc