brektrou/rtl8821CU #140: Problem with `usb_modeswitch` on RPI 4B

zhiyuanzhai, 2021-07-07:
Here's the problem I've got
```log
Take all parameters from the command line

 * usb_modeswitch: handle USB devices with multiple modes
 * Version 2.5.2 (C) Josua Dietze 2017
 * Based on libusb1/libusbx

 ! PLEASE REPORT NEW CONFIGURATIONS !

DefaultVendor=  0x0bda
DefaultProduct= 0x1a2b

StandardEject=1

Look for default devices ...
  found USB ID 174c:55aa
  found USB ID 1d6b:0003
  found USB ID 0bda:1a2b
   vendor ID matched
   product ID matched
  found USB ID 2109:3431
  found USB ID 1d6b:0002
 Found devices in default mode (1)
Access device 003 on bus 001
Get the current device configuration ...
Current configuration number is 1
Use interface number 0
 with class 8
Use endpoints 0x0b (out) and 0x8a (in)

USB description data (for identification)
-------------------------
Manufacturer: Realtek
     Product: DISK
  Serial No.: not provided
-------------------------
Sending standard EJECT sequence
Looking for active drivers ...
Set up interface 0
Use endpoint 0x0b for message sending ...
Trying to send message 1 to endpoint 0x0b ...
 Sending the message returned error -7. Try to continue
Read the response to message 1 (CSW) ...
 Response reading failed (error -8)
 Device is gone, skip any further commands
-> Run lsusb to note any changes. Bye!
```
More importantly, it seems that the USB-WiFi unit RTL8821CU is never regarded as a WiFi adapter...
```
pi@raspberrypi:~ $ lsusb
Bus 002 Device 002: ID 174c:55aa ASMedia Technology Inc. Name: ASM1051E SATA 6Gb/s bridge, ASM1053E SATA 6Gb/s bridge, ASM1153 SATA 3Gb/s bridge, ASM1153E SATA 6Gb/s bridge
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0bda:1a2b Realtek Semiconductor Corp.
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

ganioc, 2021-12-24:
Today I'm facing the same problem. The solution I found from CSDN is :
(1) /usr/share/usb_modeswitch/configPack.tar.gz
    Add a file named 0bda:1a2b into the compressed file.
(2) Put 0bda:1a2b file into directory /etc/usb_modeswitch.d

The content of 0dba:1a2b is:

```
# RTL8821cu
TargetVendor=0x0bda
TargetProduct=0xc820
StandardEject=1
```

Then it's done.

EacoChen, 2022-10-24:
@ganioc hello there, I do as you told. But it did not work. Can I see the CSDN?