brektrou/rtl8821CU #146: Not working for 4.18.0-338.el8 (Centos 8 Stream/Rocky linux 8.4)

alexisfrjp, 2021-10-03:
I modified the code to support Centos 8:
```#if (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 8, 0))```
to 
```#if (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 8, 0) || RHEL_RELEASE_CODE >= RHEL_RELEASE_VERSION(8, 0))```

USB dongle: ***Bus 001 Device 011: ID 0bda:c811 Realtek Semiconductor Corp. ***

I installed the driver but once I plugged the usb dongle in, dmesg shows:
```
[ 4955.644162] usb 1-3: new high-speed USB device number 11 using xhci_hcd
[ 4955.770564] usb 1-3: New USB device found, idVendor=0bda, idProduct=c811, bcdDevice= 2.00
[ 4955.770565] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 4955.770565] usb 1-3: Product: 802.11ac NIC
[ 4955.770566] usb 1-3: Manufacturer: Realtek
[ 4955.770566] usb 1-3: SerialNumber: 123456
[ 4955.772535] RTW: [HALMAC]11692M
               HALMAC_MAJOR_VER = 1
               HALMAC_PROTOTYPE_VER = 6
               HALMAC_MINOR_VER = 5
               HALMAC_PATCH_VER = 5
[ 4955.773015] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[ 4955.855259] RTW: HW EFUSE
[ 4955.855260] RTW: 0x000: 29 81 00 BC  09 00 21 00  6E 04 A4 10  10 00 30 0B  
[ 4955.855262] RTW: 0x010: 20 20 20 20  20 20 26 26  26 26 26 04  FF FF FF FF  
[ 4955.855264] RTW: 0x020: FF FF 26 26  26 24 21 22  23 23 24 25  25 27 27 27  
[ 4955.855265] RTW: 0x030: 24 FF FF FF  FF FF D0 FF  FF FF 20 20  20 20 20 20  
[ 4955.855267] RTW: 0x040: 26 26 26 26  26 04 FF FF  FF FF FF FF  FF FF FF FF  
[ 4955.855269] RTW: 0x050: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 4955.855270] RTW: 0x060: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 4955.855272] RTW: 0x070: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 4955.855273] RTW: 0x080: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 4955.855275] RTW: 0x090: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 4955.855276] RTW: 0x0A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 4955.855278] RTW: 0x0B0: FF FF FF FF  FF FF FF FF  7F 37 1E 00  FF FF FF FF  
[ 4955.855279] RTW: 0x0C0: FF 22 00 41  00 00 00 00  00 FF 22 FF  FF FF FF FF  
[ 4955.855281] RTW: 0x0D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 4955.855282] RTW: 0x0E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 4955.855284] RTW: 0x0F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 4955.855285] RTW: 0x100: DA 0B 11 C8  E3 46 02 20  0D B0 47 43  CA 09 03 52  
[ 4955.855287] RTW: 0x110: 65 61 6C 74  65 6B 0E 03  38 30 32 2E  31 31 61 63  
[ 4955.855289] RTW: 0x120: 20 4E 49 43  08 03 31 32  33 34 35 36  FF FF FF FF  
[ 4955.855290] RTW: 0x130: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 4955.855292] RTW: 0x140: FF FF FF FF  00 00 31 0F  FF FF FF FF  FF FF FF FF  
[ 4955.855293] RTW: 0x150: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 4955.855295] RTW: 0x160: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 4955.855296] RTW: 0x170: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 4955.855298] RTW: 0x180: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 4955.855299] RTW: 0x190: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 4955.855301] RTW: 0x1A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 4955.855302] RTW: 0x1B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 4955.855304] RTW: 0x1C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 4955.855305] RTW: 0x1D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 4955.855307] RTW: 0x1E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 4955.855308] RTW: 0x1F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 4955.855366] RTW: hal_com_config_channel_plan chplan:0x7F
[ 4955.860511] RTW: [HALMAC][ALWAYS]shall R reg twice!!
[ 4955.905128] RTW: WARN [HALMAC][WARN]the H2C ver. does not match halmac
[ 4955.909009] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[ 4955.909087] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[ 4955.909189] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[ 4955.909246] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[ 4955.909326] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[ 4955.909406] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[ 4955.909486] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[ 4955.909572] RTW: [RF_PATH] IC's RF PATH:RF_1T1R, max_tx_cnt:1
[ 4955.909572] RTW: [RF_PATH] PG's trx_path_bmp:0x00, max_tx_cnt:0
[ 4955.909572] RTW: [RF_PATH] Registry's RF PATH:UNKNOWN
[ 4955.909573] RTW: [RF_PATH] HALDATA's trx_path_bmp:0x11, max_tx_cnt:1
[ 4955.909573] RTW: [RF_PATH] HALDATA's RF PATH:RF_1T1R
[ 4955.909573] RTW: [RF_PATH] NumTotalRFPath:1
[... 147 lines cut ...]
[ 4955.931337]  ? kernfs_activate+0x63/0x80
[ 4955.931338]  usb_generic_driver_probe+0x48/0x70
[ 4955.931339]  usb_probe_device+0x3a/0xc0
[ 4955.931339]  really_probe+0x255/0x4a0
[ 4955.931340]  ? __driver_attach_async_helper+0x90/0x90
[ 4955.931340]  driver_probe_device+0x49/0xc0
[ 4955.931341]  bus_for_each_drv+0x79/0xc0
[ 4955.931342]  __device_attach+0xdc/0x160
[ 4955.931343]  bus_probe_device+0x9d/0xb0
[ 4955.931344]  device_add+0x418/0x780
[ 4955.931345]  usb_new_device+0x27b/0x4a0
[ 4955.931345]  hub_port_connect+0x617/0xa50
[ 4955.931346]  port_event+0x57a/0x880
[ 4955.931347]  hub_event+0x14f/0x3c0
[ 4955.931349]  process_one_work+0x1a7/0x360
[ 4955.931350]  worker_thread+0x30/0x390
[ 4955.931351]  ? create_worker+0x1a0/0x1a0
[ 4955.931351]  kthread+0x116/0x130
[ 4955.931352]  ? kthread_flush_work_fn+0x10/0x10
[ 4955.931353]  ret_from_fork+0x35/0x40
[ 4955.931354] ---[ end trace 4cdc9db6c841b3cf ]---
[ 4955.931412] RTW: WARN free_recv_skb_queue not empty, 8
```

* How can I fix?

luckylittle, 2021-11-25:
@brektrou I have a very similar problem - recently upgraded to Red Hat Enterprise Linux release `8.5` (Ootpa), kernel `4.18.0-348.2.1.el8_5.x86_64`.

* `make`:

```text
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.18.0-348.2.1.el8_5.x86_64/build M=/home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1  modules
make[1]: Entering directory '/usr/src/kernels/4.18.0-348.2.1.el8_5.x86_64'
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/core/rtw_cmd.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/core/rtw_security.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/core/rtw_debug.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/core/rtw_io.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/core/rtw_ioctl_query.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/core/rtw_ioctl_set.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/core/rtw_ieee80211.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/core/rtw_mlme.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/core/rtw_mlme_ext.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/core/rtw_mi.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/core/rtw_wlan_util.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/core/rtw_vht.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/core/rtw_pwrctrl.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/core/rtw_rf.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/core/rtw_chplan.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/core/rtw_recv.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/core/rtw_sta_mgt.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/core/rtw_ap.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/core/mesh/rtw_mesh.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/core/rtw_xmit.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/core/rtw_p2p.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/core/rtw_rson.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/core/rtw_tdls.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/core/rtw_br_ext.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/core/rtw_iol.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/core/rtw_sreset.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/core/rtw_btcoex_wifionly.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/core/rtw_btcoex.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/core/rtw_beamforming.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/core/rtw_odm.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/core/rtw_rm.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/core/rtw_rm_fsm.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/core/rtw_rm_util.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/core/efuse/rtw_efuse.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/os_dep/osdep_service.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/os_dep/linux/os_intfs.o
/home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/os_dep/linux/os_intfs.c:1558:12: warning: ‘rtw_select_queue’ defined but not used [-Wunused-function]
 static u16 rtw_select_queue(struct net_device *dev, struct sk_buff *skb
            ^~~~~~~~~~~~~~~~
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/os_dep/linux/usb_intf.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/os_dep/linux/usb_ops_linux.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/os_dep/linux/ioctl_linux.o
/home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/os_dep/linux/ioctl_linux.c:12332:30: warning: ‘rtw_get_wireless_stats’ defined but not used [-Wunused-function]
 static struct iw_statistics *rtw_get_wireless_stats(struct net_device *dev)
                              ^~~~~~~~~~~~~~~~~~~~~~
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/os_dep/linux/xmit_linux.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/os_dep/linux/mlme_linux.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/os_dep/linux/recv_linux.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/os_dep/linux/wifi_regd.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/os_dep/linux/rtw_android.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/os_dep/linux/rtw_proc.o
  CC [M]  /home/lmaly/Projects/WiFi_drivers/brektrou-rtl8821CU-5.8.1/os_dep/linux/rtw_rhashtable.o
[... cut ...]
[ 3831.759401] RDX: 000055a3560c4630 RSI: 00000000004dcbe0 RDI: 000055a3568d26a0
[ 3831.759402] RBP: 000055a3560c4630 R08: 000055a3560c301a R09: 0000000000000001
[ 3831.759402] R10: 000055a3560c3010 R11: 0000000000000246 R12: 000055a3568d26a0
[ 3831.759403] R13: 000055a3560c4270 R14: 0000000000060000 R15: 0000000000000000
[ 3831.759404] ---[ end trace 06a65904d16f61e3 ]---
[ 3831.759465] RTW: WARN free_recv_skb_queue not empty, 8
[ 3831.759509] usbcore: registered new interface driver rtl8821cu
[ 3831.759509] RTW: module init ret=0
```

ssayouri, 2022-02-23:
Hello @luckylittle and @alexisfrjp , 

Have a similar issue here when trying to build the driver for CentOS Stream 8. 

did you managed to workaround this issue ? 

Many thanks in advance

alexisfrjp, 2022-03-19:
No, it has never worked, I gave up my 500yens wifi.

yatindrav, 2022-08-07:
I was able to make it working with https://github.com/brektrou/rtl8821CU/commit/db47217e31b95b4a760a58d6c4790631d1da4896
fix by adding:
#if (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 3, 0)) || (LINUX_VERSION_CODE == KERNEL_VERSION(4, 18, 0))
instead of just 
#if (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 3, 0))
in whole patch
though you need to create wifi interface using nmtui once module is loaded.