brektrou/rtl8821CU #151: Fails to build DKMS for Linux 5.15.1-1 on Manjaro

fishears, 2021-11-08:
Just installed the Linux 5.15.1-1 kernel and the DKMS build of this driver failed with the following log output:

`DKMS make.log for rtl8821cu-5.4.11202e43634 for kernel 5.15.1-1-MANJARO (x86_64)
Mon  8 Nov 09:04:36 GMT 2021
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.15.1-1-MANJARO/build M=/var/lib/dkms/rtl8821cu/5.4.11202e43634/build  modules
make[1]: Entering directory '/usr/lib/modules/5.15.1-1-MANJARO/build'
  CC [M]  /var/lib/dkms/rtl8821cu/5.4.11202e43634/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.4.11202e43634/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.4.11202e43634/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.4.11202e43634/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.4.11202e43634/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.4.11202e43634/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.4.11202e43634/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.4.11202e43634/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.4.11202e43634/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.4.11202e43634/build/core/rtw_mi.o
/var/lib/dkms/rtl8821cu/5.4.11202e43634/build/core/rtw_mlme.c: In function ‘rtw_drv_scan_by_self’:
/var/lib/dkms/rtl8821cu/5.4.11202e43634/build/core/rtw_mlme.c:3122:17: warning: this ‘else’ clause does not guard... [-Wmisleading-indentation]
 3122 |                 else
      |                 ^~~~
/var/lib/dkms/rtl8821cu/5.4.11202e43634/build/core/rtw_mlme.c:3125:25: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘else’
 3125 |                         goto exit;
      |                         ^~~~
  CC [M]  /var/lib/dkms/rtl8821cu/5.4.11202e43634/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.4.11202e43634/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.4.11202e43634/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.4.11202e43634/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.4.11202e43634/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.4.11202e43634/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.4.11202e43634/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.4.11202e43634/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.4.11202e43634/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.4.11202e43634/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.4.11202e43634/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.4.11202e43634/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.4.11202e43634/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.4.11202e43634/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.4.11202e43634/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.4.11202e43634/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.4.11202e43634/build/core/rtw_sreset.o
/var/lib/dkms/rtl8821cu/5.4.11202e43634/build/core/rtw_br_ext.c:20:18: fatal error: net/ipx.h: No such file or directory
   20 |         #include <net/ipx.h>
      |                  ^~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:277: /var/lib/dkms/rtl8821cu/5.4.11202e43634/build/core/rtw_br_ext.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:1868: /var/lib/dkms/rtl8821cu/5.4.11202e43634/build] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.15.1-1-MANJARO/build'
make: *** [Makefile:2212: modules] Error 2`

kelebek333, 2021-11-08:
Try this https://github.com/brektrou/rtl8821CU/pull/152/commits/3e5240c3d4d01a756bf43f048291eee6793fa20a

gaoyanglion, 2021-11-12:
it works , 3q

closed 2021-11-16 by brektrou