brektrou/rtl8821CU #156: Won't connect to an AP after reboot.

Shravansingh, 2021-12-14:
After I installed the module. I was able to get connected to an AP. But after restart it won't connect to the AP anymore. I need to unplug and plug the Wifi Module again.

Shravansingh, 2021-12-17:
Some log files

```
System:    Kernel: 5.11.0-41-generic x86_64 bits: 64 compiler: N/A Desktop: Gnome 3.36.9 wm: gnome-shell 
           dm: GDM3 3.36.3 Distro: Ubuntu 20.04.3 LTS (Focal Fossa) 
Machine:   Type: Desktop Mobo: N/A model: EMB-BT4 serial: <filter> UEFI: American Megatrends v: 0125 
           date: 06/03/2021 
Battery:   Device-1: hidpp_battery_0 model: Logitech Wireless Touch Keyboard K400 Plus serial: <filter> 
           charge: 100% (should be ignored) rechargeable: yes status: Discharging 
CPU:       Topology: Quad Core model: Intel Celeron J1900 bits: 64 type: MCP arch: Silvermont rev: 9 
           L2 cache: 1024 KiB 
           flags: lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx bogomips: 16000 
           Speed: 1334 MHz min/max: 1333/2416 MHz Core speeds (MHz): 1: 1334 2: 1334 3: 1333 4: 1333 
Graphics:  Device-1: Intel Atom Processor Z36xxx/Z37xxx Series Graphics & Display driver: i915 v: kernel 
           bus ID: 00:02.0 chip ID: 8086:0f31 
           Display: x11 server: X.Org 1.20.11 driver: i915 compositor: gnome-shell 
           resolution: 1920x1080~60Hz 
           OpenGL: renderer: Mesa DRI Intel HD Graphics (BYT) v: 4.2 Mesa 21.0.3 compat-v: 3.0 
           direct render: Yes 
Audio:     Device-1: Intel Atom Processor Z36xxx/Z37xxx Series High Definition Audio driver: snd_hda_intel 
           v: kernel bus ID: 00:1b.0 chip ID: 8086:0f04 
           Sound Server: ALSA v: k5.11.0-41-generic 
Network:   Device-1: Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet driver: r8169 v: kernel 
           port: c000 bus ID: 03:00.0 chip ID: 10ec:8168 
           IF: eth0 state: up speed: 1000 Mbps duplex: full mac: <filter> 
Drives:    Local Storage: total: 29.82 GiB used: 21.84 GiB (73.2%) 
           ID-1: /dev/sda vendor: Transcend model: TS32GSSD420K size: 29.82 GiB speed: 6.0 Gb/s 
           serial: <filter> rev: 8B temp: 39 C scheme: GPT 
Partition: ID-1: / size: 28.73 GiB used: 21.83 GiB (76.0%) fs: ext4 dev: /dev/sda2 
Sensors:   System Temperatures: cpu: 49.0 C mobo: N/A 
           Fan Speeds (RPM): N/A 
Repos:     Active apt repos in: /etc/apt/sources.list 
           1: deb http://archive.ubuntu.com/ubuntu focal main restricted
           2: deb http://archive.ubuntu.com/ubuntu focal-updates main restricted
           3: deb http://archive.ubuntu.com/ubuntu focal universe
           4: deb http://archive.ubuntu.com/ubuntu focal-updates universe
           5: deb http://archive.ubuntu.com/ubuntu focal multiverse
           6: deb http://archive.ubuntu.com/ubuntu focal-updates multiverse
           7: deb http://archive.ubuntu.com/ubuntu focal-backports main restricted universe multiverse
           8: deb http://archive.ubuntu.com/ubuntu focal-security main restricted
           9: deb http://archive.ubuntu.com/ubuntu focal-security universe
           10: deb http://archive.ubuntu.com/ubuntu focal-security multiverse
Info:      Processes: 202 Uptime: 21m Memory: 3.73 GiB used: 677.8 MiB (17.7%) Init: systemd v: 245 
           runlevel: 5 Compilers: gcc: 9.3.0 alt: 9 Shell: bash v: 5.0.17 running in: gnome-terminal 
           inxi: 3.0.38
```

```
blue@blue:~$ lsusb 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 14e1:3508 Dialogue Technology Corp. 
Bus 001 Device 005: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 003: ID 0bda:c820 Realtek Semiconductor Corp. 802.11ac NIC
Bus 001 Device 004: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
blue@blue:~$ 
```

```
blue@blue:~$ lsmod | grep 8821cu
8821cu               2207744  0
cfg80211              888832  1 8821cu
blue@blue:~$ 
```
Changed the GRUB file

```
GRUB_DEFAULT=0
GRUB_TIMEOUT_STYLE=hidden
[... 1967 lines cut ...]
[    8.586219] kernel: input: HDA Intel PCH HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1b.0/sound/card0/input26
[    8.822104] kernel: audit: type=1400 audit(1639436974.252:2): apparmor="STATUS" operation="profile_load" profile="unconfined" name="nvidia_modprobe" pid=534 comm="apparmor_parser"
[    8.822122] kernel: audit: type=1400 audit(1639436974.252:3): apparmor="STATUS" operation="profile_load" profile="unconfined" name="nvidia_modprobe//kmod" pid=534 comm="apparmor_parser"
[    8.822749] kernel: audit: type=1400 audit(1639436974.252:4): apparmor="STATUS" operation="profile_load" profile="unconfined" name="libreoffice-oopslash" pid=533 comm="apparmor_parser"
[    8.832125] kernel: audit: type=1400 audit(1639436974.260:5): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/usr/sbin/tcpdump" pid=535 comm="apparmor_parser"
[    8.839024] kernel: audit: type=1400 audit(1639436974.268:6): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/usr/lib/cups/backend/cups-pdf" pid=532 comm="apparmor_parser"
[    8.839041] kernel: audit: type=1400 audit(1639436974.268:7): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/usr/sbin/cupsd" pid=532 comm="apparmor_parser"
[    8.839049] kernel: audit: type=1400 audit(1639436974.268:8): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/usr/sbin/cupsd//third_party" pid=532 comm="apparmor_parser"
[    8.840259] kernel: audit: type=1400 audit(1639436974.268:9): apparmor="STATUS" operation="profile_load" profile="unconfined" name="ippusbxd" pid=537 comm="apparmor_parser"
[... cut ...]
[   10.465719] kernel: r8169 0000:03:00.0 eth0: Link is Down
[   10.481426] kernel: RTW: [HALMAC][ALWAYS]shall R reg twice!!
[   10.482550] kernel: RTW: WARN [HALMAC][WARN]H2C/C2H ver is compatible!!
[   11.129114] kernel: RTW: wlan0- hw port(0) mac_addr =e8:4e:06:96:e3:5f
[   11.528986] kernel: loop14: detected capacity change from 0 to 8
[   11.845807] kernel: Bluetooth: RFCOMM TTY layer initialized
[   11.845841] kernel: Bluetooth: RFCOMM socket layer initialized
[   11.845858] kernel: Bluetooth: RFCOMM ver 1.11
```

Shravansingh, 2021-12-17:
I am not a driver expert but anything that would guide to a solution would help me

wwgoncalves, 2023-09-07:
Hi! The solution described here https://askubuntu.com/a/1082418 worked for me.
> I've found the way to do it. I've modified the file /lib/udev/rules.d/40-usb_modeswitch.rules appending before the line LABEL="modeswitch_rules_end" the following:
> 
> ```
> # Realtek 8211CU Wifi AC USB
> ATTR{idVendor}=="0bda", ATTR{idProduct}=="1a2b", RUN+="/usr/sbin/usb_modeswitch -K -v 0bda -p 1a2b"
> ```

For some reason (no idea why) the dongle sometimes is identified as a storage device with that `1a2b` product code (e.g. `lsusb` lists it as `Bus 001 Device 005: ID 0bda:1a2b Realtek Semiconductor Corp. DISK`) and the rule above will "eject" it every time the dongle is identified as such.
Confused? Me too. :sweat_smile: