brektrou/rtl8821CU #157: Realtek  0bda:c811   usb_modeswitch  could't switch

licc, 2021-12-24:
Linux version 5.4.0-91-generic (buildd@lcy01-amd64-017) (gcc version 9.3.0 (Ubuntu 9.3.0-17ubuntu1~20.04)) #102-Ubuntu SMP Fri Nov 5 16:31:28 UTC 2021

[image]

root@leecc:~# sudo usb_modeswitch -KW -v 0bda -p c811
Take all parameters from the command line

 * usb_modeswitch: handle USB devices with multiple modes
 * Version 2.6.0 (C) Josua Dietze 2017
 * Based on libusb1/libusbx

 ! PLEASE REPORT NEW CONFIGURATIONS !

DefaultVendor=  0x0bda
DefaultProduct= 0xc811

StandardEject=1

Look for default devices ...
  found USB ID 1d6b:0003
  found USB ID 0bda:c811
   vendor ID matched
   product ID matched
  found USB ID 05e3:0610
  found USB ID 1d6b:0002
 Found devices in default mode (1)
Access device 003 on bus 001
Get the current device configuration ...
Current configuration number is 1
Use interface number 0
 with class 255
Error: can't use storage command in MessageContent with interface 0; interface class is 255, expected 8. Abort

adamyap204, 2021-12-31:
same problem as you. I also use ubuntu 20.04 and same wifi adapter. after reboot due to some software update. the wifi adapter stop working and the moment i tried to install back then this problem hit me. I kinda new to this linux os. Hope anyone can help please

Pendrag00n, 2022-03-14:
Same problem here. It used to work on an older Kali installation

PawelZ82, 2022-03-24:
For me, it didn't even install 
`Error! Your kernel headers for kernel 5.15.30-v8+ cannot be found.
Please install the linux-headers-5.15.30-v8+ package,
or use the --kernelsourcedir option to tell DKMS where it's located
Error! Your kernel headers for kernel 5.15.30-v8+ cannot be found.
Please install the linux-headers-5.15.30-v8+ package,
or use the --kernelsourcedir option to tell DKMS where it's located
Finished running dkms install steps.`
The output of the modeswitch is the same

multikirin, 2022-03-24:
Install your kernel headers or attempt to do:apt upgrade and then apt update (debian based)pacman -Syu linux-headers-$(uname -r)