brektrou/rtl8821CU #160: Doesn't work after updating to kernel version 5.13.0-30

Artafinde, 2022-03-01:
After updating my kernel, I re-followed the instructions from the README.md in this repository. I followed the _Build and install without DKMS_ instructions.

I can confirm the file `8821cu.ko` is at `/lib/modules/5.13.0-30-generic/kernel/drivers/net/wireless/realtek/rtl8821cu/`

But the wifi adapter does not work, even after a reboot. The flashing light that would indicate data transfer is permanently dark. 

It worked fine with kernel `5.13.0-27-generic`

Artafinde, 2022-03-01:
The output of `cat /etc/os-release`

PRETTY_NAME="Ubuntu 21.10"
NAME="Ubuntu"
VERSION_ID="21.10"
VERSION="21.10 (Impish Indri)"
VERSION_CODENAME=impish
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=impish

multikirin, 2022-03-01:
Does building with DKMS work? For me, with dkms it erros but upon update && upgrade it magically works--Sent from my Android phone with mail.com Mail. Please excuse my brevity.

multikirin, 2022-03-01:
What the hell mail.com