brektrou/rtl8821CU #162: Why do we need to check the interface name?

eyalroz, 2022-03-20:
The installation instructions in the README require us to determine the name of the wifi adapter interface. 

1. Why?
2. How would we have this name before the driver is already installed and working?

multikirin, 2022-03-20:
I dont see that, where is it?

Message ID: ***@***.***>

eyalroz, 2022-03-20:
@multikirin : It's right there in the README file...

> ### Check the name of the interface

multikirin, 2022-03-21:
If you arent using a obscure distro, skip that step

eyalroz, 2022-03-21:
> If you arent using a obscure distro, skip that step

I see. Please consider clarifying this in the README itself, so that other people don't get confused and try to do something with the interface name even if they don't need to.

Otherwise - thank you so much for writing this driver!

closed 2022-03-21 by eyalroz