brektrou/rtl8821CU #163: Building with DKMS doesn't work for 5.17.0-051700

v-stikhilyas, 2022-03-21:
After upgrading ubuntu kernel to 5.17 dkms-install.sh gets failed with compilation error:

```
$ cat /var/lib/dkms/rtl8821CU/5.4.1/build/make.log
DKMS make.log for rtl8821CU-5.4.1 for kernel 5.17.0-051700-generic (x86_64)
Пн 21 мар 2022 12:40:36 +07
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.17.0-051700-generic/build M=/var/lib/dkms/rtl8821CU/5.4.1/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.17.0-051700-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (Ubuntu 11.2.0-18ubuntu1) 11.2.0
  You are using:           gcc (Ubuntu 11.2.0-7ubuntu2) 11.2.0
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_mlme.o
/var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_mlme.c: In function ‘rtw_drv_scan_by_self’:
/var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_mlme.c:3122:17: warning: this ‘else’ clause does not guard... [-Wmisleading-indentation]
 3122 |                 else
      |                 ^~~~
/var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_mlme.c:3125:25: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘else’
 3125 |                         goto exit;
      |                         ^~~~
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_rm.o
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/rtw_rm_fsm.o
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/core/efuse/rtw_efuse.o
/var/lib/dkms/rtl8821CU/5.4.1/build/core/efuse/rtw_efuse.c: In function ‘rtw_efuse_analyze’:
/var/lib/dkms/rtl8821CU/5.4.1/build/core/efuse/rtw_efuse.c:781:17: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
  781 |                 if (i % 16 == 0)
      |                 ^~
In file included from /var/lib/dkms/rtl8821CU/5.4.1/build/include/drv_types.h:60,
                 from /var/lib/dkms/rtl8821CU/5.4.1/build/core/efuse/rtw_efuse.c:17:
/var/lib/dkms/rtl8821CU/5.4.1/build/include/rtw_debug.h:231:9: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
  231 |         do {\
      |         ^~
/var/lib/dkms/rtl8821CU/5.4.1/build/core/efuse/rtw_efuse.c:783:25: note: in expansion of macro ‘_RTW_PRINT_SEL’
  783 |                         _RTW_PRINT_SEL(RTW_DBGDUMP, "%02X%s"
      |                         ^~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/osdep_service.o
/var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/osdep_service.c: In function ‘thread_exit’:
/var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/osdep_service.c:1297:9: error: implicit declaration of function ‘complete_and_exit’ [-Werror=implicit-function-declaration]
 1297 |         complete_and_exit(comp, 0);
      |         ^~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/osdep_service.c: In function ‘rtw_change_ifname’:
/var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/osdep_service.c:2490:28: warning: passing argument 1 of ‘_rtw_memcpy’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 2490 |         _rtw_memcpy(pnetdev->dev_addr, adapter_mac_addr(padapter), ETH_ALEN);
      |                     ~~~~~~~^~~~~~~~~~
/var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/osdep_service.c:925:24: note: expected ‘void *’ but argument is of type ‘const unsigned char *’
  925 | void _rtw_memcpy(void *dst, const void *src, u32 sz)
      |                  ~~~~~~^~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:288: /var/lib/dkms/rtl8821CU/5.4.1/build/os_dep/osdep_service.o] Error 1
make[1]: *** [Makefile:1837: /var/lib/dkms/rtl8821CU/5.4.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.17.0-051700-generic'
make: *** [Makefile:2222: modules] Error 2

```

It worked fine with kernel 5.16.generic

```
cat /etc/os-release
PRETTY_NAME="Ubuntu 21.10"
NAME="Ubuntu"
VERSION_ID="21.10"
VERSION="21.10 (Impish Indri)"
VERSION_CODENAME=impish
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=impish

```

kelebek333, 2022-03-21:
You can try to install same driver from my wireless PPA, if you wish. I will send same buildfix to this source tomorrow.

```
sudo add-apt-repository ppa:kelebek333/kablosuz
sudo apt-get update
sudo apt install rtl8821cu-dkms
```

> Changelog
> 
> rtl8821cu-dkms (5.4.1~impish8) impish; urgency=medium
> 
>   * Added Linux 5.17 build fix.
>   * Removed obselete REMAKE_INITRD from dkms.conf

castelan-jesus, 2022-03-22:
> You can try to install same driver from my wireless PPA, if you wish. I will send same buildfix to this source tomorrow.
>
> ```
> sudo add-apt-repository ppa:kelebek333/kablosuz
> sudo apt-get update
> sudo [...]

Thank you!