brektrou/rtl8821CU #172: BT+WLAN not working after reboot (BT works, WLAN not)

Dephrilibrium, 2022-05-06:
Heyho,

Since a few month I'm having the problem, that the driver won't work after a reboot of my Ubuntu 20.04 LTE (switched from Gnome to KDE). I'm using a BT+AC600 Bluetooth donlge (https://www.amazon.de/Bluetooth-Adapter-Wireless-Netzwerk-Dongles/dp/B08YK3GRN6/ref=sr_1_3?keywords=bluetooth+wlan+dongle+linux&link_code=qs&qid=1651870421&sourceid=Mozilla-search&sr=8-3). The Bluetooth works everytime correct, but not the WLAN-module.

After I bought the dongle it worked perfectly but it seems that there was an update that caused problems on my dongle (its not correctly modeswitched). After some checks I noticed, that after each update of "wireless-regdb" the error occured and I had to rebuild and reinstall the driver (note that only reinstalling not worked! I had also complete rebuild the driver and then reinstalling it to get the dongle back to work). Re-plugging has also no effect -> The BT works correctly, but no WLAN.

After that there happened something else which causes the same error again. So now I have to rebuild and reinstall the driver each time after a reboot. Sleep-Mode is ok and the Dongle works correctly after wake. 

I'm not sure what to do now, may you now how to fix it.
Btw: May it is similar to the issue https://github.com/brektrou/rtl8821CU/issues/156

Kind regards,
Deph

wwgoncalves, 2023-09-07:
Hi! The solution described here https://askubuntu.com/a/1082418 worked for me.
> I've found the way to do it. I've modified the file /lib/udev/rules.d/40-usb_modeswitch.rules appending before the line LABEL="modeswitch_rules_end" the following:
> 
> ```
> # Realtek 8211CU Wifi AC USB
> ATTR{idVendor}=="0bda", ATTR{idProduct}=="1a2b", RUN+="/usr/sbin/usb_modeswitch -K -v 0bda -p 1a2b"
> ```

For some reason (no idea why) the dongle sometimes is identified as a storage device with that `1a2b` product code (e.g. `lsusb` lists it as `Bus 001 Device 005: ID 0bda:1a2b Realtek Semiconductor Corp. DISK`) and the rule above will "eject" it every time the dongle is identified as such.
Confused? Me too. :sweat_smile:

Dephrilibrium, 2023-09-10:
Hi [wwgoncalves](https://github.com/wwgoncalves),

thanks for the response, but I knew and configured that already. It is described under "Make it permanent" in [brektrou's README.md](https://github.com/brektrou/rtl8821CU/blob/master/README.md).

The problem still happens after the PC was turned off (completely disconnected by a switchable net-power strip).

Kind regards,
Deph